esp8266/Arduino #3333: i2c issue

battosai30, 2017-06-06:
### Basic Infos

#### Hardware
Hardware:			Node Mcu
Core Version:      	2.3.0

### Description

Hi,

Trying to play with a CCS811 detector + HC1080 T°C/H% sensor, I met troubles with i2c. After some tests, I discovered that put some strategic delay() allows everything to work good. For the moment I don't know why, but it allows me to get everything working so ....

My modifications only concern Wire.cpp : 

```
size_t TwoWire::requestFrom(uint8_t address, size_t size, bool sendStop){
	delay(1);
  if(size > BUFFER_LENGTH){
    size = BUFFER_LENGTH;
  }
  size_t read = (twi_readFrom(address, rxBuffer, size, sendStop) == 0)?size:0;
  rxBufferIndex = 0;
  rxBufferLength = read;
  delay(1);
  return read;
}

void TwoWire::beginTransmission(uint8_t address){
  delay(1);
  transmitting = 1;
  txAddress = address;
  txBufferIndex = 0;
  txBufferLength = 0;
}

uint8_t TwoWire::endTransmission(uint8_t sendStop){
  int8_t ret = twi_writeTo(txAddress, txBuffer, txBufferLength, sendStop);
  txBufferIndex = 0;
  txBufferLength = 0;
  transmitting = 0;
  delay(1);
  return ret;
}
```
It seems to have solved another issue that I got : sometimes the i2c bus seems to get stuck (no response at all) and the only way to unlock it was to completly turn power off and on.

### Settings in IDE

Module:  			NodeMcu
Flash Size: 		??
CPU Frequency:		80Mhz
Flash Mode:			??
Flash Frequency:	??
Upload Using:		SERIAL
Reset Method:		??

### Sketch

```
#include <Wire.h>
#include <SparkFunCCS811.h>
#include <ClosedCube_HDC1080.h>

#define CCS811_ADDR 0x5A 

//Global sensor objects
CCS811 myCCS811(CCS811_ADDR);
ClosedCube_HDC1080 hdc1080;

void setup() {
  
 Serial.begin(9600);
  Serial.println();
  Serial.println("Apply BME280 data to CCS811 for compensation.");

  //This begins the CCS811 sensor and prints error status of .begin()
  CCS811Core::status returnCode = myCCS811.begin();
  Serial.print("CCS811 begin exited with: ");
  //Pass the error code to a function to print the results
  printDriverError( returnCode );
  Serial.println();

    hdc1080.begin(0x40);

      Serial.print("Manufacturer ID=0x");
  Serial.println(hdc1080.readManufacturerId(), HEX); // 0x5449 ID of Texas Instruments
  Serial.print("Device ID=0x");
  Serial.println(hdc1080.readDeviceId(), HEX); // 0x1050 ID of the device
 
}

void loop() {
  
//Check to see if data is available
  if (myCCS811.dataAvailable())
  {
    //Calling this function updates the global tVOC and eCO2 variables
    myCCS811.readAlgorithmResults();
    //printInfoSerial fetches the values of tVOC and eCO2
    printInfoSerial();

    float BMEtempC = hdc1080.readTemperature();// myBME280.readTempC();
    float BMEhumid = hdc1080.readHumidity();//myBME280.readFloatHumidity();

    Serial.print("Applying new values (deg C, %): ");
    Serial.print(BMEtempC);
    Serial.print(",");
    Serial.println(BMEhumid);
    Serial.println();

    //This sends the temperature data to the CCS811
    myCCS811.setEnvironmentalData(BMEhumid, BMEtempC);
  }
  else if (myCCS811.checkForStatusError())
  {
    //If the CCS811 found an internal error, print it.
    printSensorError();
  }

  delay(2000); //Wait for next reading
  
}

//---------------------------------------------------------------
void printInfoSerial()
{
  //getCO2() gets the previously read data from the library
  Serial.println("CCS811 data:");
  Serial.print(" CO2 concentration : ");
  Serial.print(myCCS811.getCO2());
  Serial.println(" ppm");

  //getTVOC() gets the previously read data from the library
  Serial.print(" TVOC concentration : ");
  Serial.print(myCCS811.getTVOC());
  Serial.println(" ppb");

  Serial.println("BME280 data:");
  Serial.print(" Temperature: ");
  Serial.print(hdc1080.readTemperature(), 2);
  Serial.println(" degrees C");

  Serial.print(" %RH: ");
  Serial.print(hdc1080.readHumidity(), 2);
  Serial.println(" %");

  Serial.println();

}

//printDriverError decodes the CCS811Core::status type and prints the
//type of error to the serial terminal.
//
//Save the return value of any function of type CCS811Core::status, then pass
//to this function to see what the output was.
void printDriverError( CCS811Core::status errorCode )
{
  switch ( errorCode )
  {
    case CCS811Core::SENSOR_SUCCESS:
      Serial.print("SUCCESS");
      break;
    case CCS811Core::SENSOR_ID_ERROR:
      Serial.print("ID_ERROR");
      break;
    case CCS811Core::SENSOR_I2C_ERROR:
      Serial.print("I2C_ERROR");
      break;
    case CCS811Core::SENSOR_INTERNAL_ERROR:
      Serial.print("INTERNAL_ERROR");
      break;
    case CCS811Core::SENSOR_GENERIC_ERROR:
      Serial.print("GENERIC_ERROR");
      break;
    default:
      Serial.print("Unspecified error.");
  }
}

//printSensorError gets, clears, then prints the errors
//saved within the error register.
void printSensorError()
{
  uint8_t error = myCCS811.getErrorRegister();

  if ( error == 0xFF ) //comm error
  {
    Serial.println("Failed to get ERROR_ID register.");
  }
  else
  {
    Serial.print("Error: ");
    if (error & 1 << 5) Serial.print("HeaterSupply");
    if (error & 1 << 4) Serial.print("HeaterFault");
    if (error & 1 << 3) Serial.print("MaxResistance");
    if (error & 1 << 2) Serial.print("MeasModeInvalid");
    if (error & 1 << 1) Serial.print("ReadRegInvalid");
    if (error & 1 << 0) Serial.print("MsgInvalid");
    Serial.println();
  }
}
```

### Debug Messages

bhumindesai, 2018-02-09:
THANK YOU A LOT.
 add "delay(1);" works for me too.. 
 i cannot describe my happiness. 
you have solved my prob which was on halt since last 2 yrs.

battosai30, 2018-02-09:
Very happy that it helped you :)

queenbey91, 2018-02-11:
#4331

queenbey91, 2018-02-11:
I found the explanation ;) #4331

devyte, 2018-02-15:
In such cases, the wdt triggers when the 3rd party lib doesn't feed it between api calls. The delays do not belong inside the wire calls themselves.
Usually this happens when a 3rd party lib hasn't been tested/ported to work with the esp.
Closing.

closed 2018-02-15 by devyte