esp8266/Arduino #3334: I2C Bit Banging

artemalek, 2017-06-07:
#  Basic Infos

#### Hardware
Hardware:        	ESP-12E
Core Version:      	2.3

### Description
We are trying to connect 4 identical BMP280 to the same ESP8266 AI Thinker or NodeMCU V3. The problem is that we have only 2 pins designated for that matter (GPIO4 and GPIO5)... The ESP8266 allocates the same address for all of the 4 bmps hence we can't get a reading from 4 devices.
Right now we are searching for solutions for this matter:
1) The first is implementing I2C bit banging software and define there the GPIO we want top work with.
2) Using TCA9548A i2c multiplexer 1 -8 board which gives us additional I2C ports.

Which is better? And how is it possible to define different GPIO ports to work with I2C by using Bit Banging?

P.S.
Our goal in this project is to use the lowest amount of devices as possible..

Best Regards and thanks in advance,
Artem

### Settings in IDE

Module:  			nodeMCU V3/esp8266 ai thinker
Flash Size: 		?4MB/1MB?
CPU Frequency:	?80Mhz?
Flash Mode:		?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

igrr, 2017-06-07:
How about reinitializing the library to work with new pins each time?
E.g.
```c++
Wire.begin(SDA1, SCL1);
bmp.begin();
bmp.readPressure();

Wire.begin(SDA2, SCL2);
bmp.begin();
bmp.readPressure();
```

artemalek, 2017-06-07:
Hi igrr,
Thank you for the quick response. Our problem lays in the fact that we want to connect 4 device to the esp8266. With your advice we were able to connect 2 devices on different GPIOs. However we are short of GPIO pins.
So we thought maybe to connect two device on the same bus (same GPIO) and then to communicate with both of them by giving them turns to "speak". That way we will be able to connect 4 devices on 2 different BUSes.

Our code so far: for two device using two GPIOs:
`
Adafruit_BMP280 bmp1; // I2C
Adafruit_BMP280 bmp2; // I2C

void setup() {
  Serial.begin(115200);
  Serial.println(F("BMP280 test"));
  
  Wire.begin(4,5);

  if (!bmp1.begin()) {  
    Serial.println(F("Could not find a valid bmp1 sensor, check wiring!"));
    while (1);
  }

  Wire.begin(2,0);

  if (!bmp2.begin(0x77)) {  
    Serial.println(F("Could not find a valid bmp2 sensor, check wiring!"));
    while (1);
  }
 
}

void loop() {
  Wire.begin(4,5);
  
    Serial.print(F("Temperature of bmp1 = "));
    Serial.print(bmp1.readTemperature());
    Serial.println(" *C");
    
    Serial.print(F("Pressure of bmp1 = "));
    Serial.print(bmp1.readPressure());
    Serial.println(" Pa");

    Serial.print(F("Approx altitude of bmp1 = "));
    Serial.print(bmp1.readAltitude(1013.25)); // this should be adjusted to your local forcase
    Serial.println(" m");
    
    Serial.println();
	  delay(3000);

  Wire.begin(2,0);
  delay(1000);
	Serial.print(F("Temperature of bmp2 = "));
    Serial.print(bmp2.readTemperature());
    Serial.println(" *C");
    
    Serial.print(F("Pressure of bmp2 = "));
    Serial.print(bmp2.readPressure());
    Serial.println(" Pa");

    Serial.print(F("Approx altitude of bmp2 = "));
    Serial.print(bmp2.readAltitude(1013.25)); // this should be adjusted to your local forcase
    Serial.println(" m");
    
    Serial.println();
    
    delay(2000);
}`

Do you have any clue how we can do so?
Best Regards, and thanks in advance, Artem

Humancell, 2017-06-08:
Hello,

You are half-way there!  Here is what you can do now.

Referring to the BMP280 data sheet:  [https://cdn-shop.adafruit.com/datasheets/BST-BMP280-DS001-11.pdf](https://cdn-shop.adafruit.com/datasheets/BST-BMP280-DS001-11.pdf)

Section 5.2 states:  "_The last bit is changeable by SDO value and can be changed during operation. Connecting SDO to GND results in slave address 1110110 (0x76); connection it to VDDIO results in slave address 1110111 (0x77), which is the same as BMP180’s I²C address. The SDO pin cannot be left floating; if left floating, the I²C address will be undefined._"

So on each of the I2C busses, you need one of the BMP280 to have address 0x76, and the other to have address 0x77.  On the Adafruit PCB, this should be the SDO pin that you pull high or low to set this address. 

In the Adafruit BMP280 library, the default address that is expected is 0x77.

https://github.com/adafruit/Adafruit_BMP280_Library/blob/master/Adafruit_BMP280.h#L37

But in the bmp{x}.begin() that you are doing, you can pass in the I2C address that you want to use:

https://github.com/adafruit/Adafruit_BMP280_Library/blob/master/Adafruit_BMP280.h#L130

With all of that, you would then have to double up your current process, doing something like:

```
Adafruit_BMP280 bmp1; // I2C
Adafruit_BMP280 bmp2; // I2C
Adafruit_BMP280 bmp3; // I2C
Adafruit_BMP280 bmp4; // I2C

void setup() {
  Serial.begin(115200);
  Serial.println(F("BMP280 test"));
  
  Wire.begin(4,5);

  if (!bmp1.begin(0x77)) {  
    Serial.println(F("Could not find a valid bmp1 sensor, check wiring!"));
    while (1);
  }

  if (!bmp2.begin(0x76)) {  
    Serial.println(F("Could not find a valid bmp2 sensor, check wiring!"));
    while (1);
  }

  Wire.begin(2,0);

  if (!bmp3.begin(0x77)) {  
    Serial.println(F("Could not find a valid bmp3 sensor, check wiring!"));
    while (1);
  }
 
  if (!bmp4.begin(0x76)) {  
    Serial.println(F("Could not find a valid bmp4 sensor, check wiring!"));
    while (1);
  }

}

void loop() {
  Wire.begin(4,5);
  
    Serial.print(F("Temperature of bmp1 = "));
    Serial.print(bmp1.readTemperature());
    Serial.println(" *C");
    
    Serial.print(F("Pressure of bmp1 = "));
    Serial.print(bmp1.readPressure());
    Serial.println(" Pa");

    Serial.print(F("Approx altitude of bmp1 = "));
    Serial.print(bmp1.readAltitude(1013.25)); // this should be adjusted to your local forcase
    Serial.println(" m");
    
    Serial.println();

   Serial.print(F("Temperature of bmp2 = "));
    Serial.print(bmp2.readTemperature());
    Serial.println(" *C");
    
    Serial.print(F("Pressure of bmp2 = "));
    Serial.print(bmp2.readPressure());
    Serial.println(" Pa");

    Serial.print(F("Approx altitude of bmp2 = "));
    Serial.print(bmp2.readAltitude(1013.25)); // this should be adjusted to your local forcase
    Serial.println(" m");
    
    Serial.println();
	  delay(3000);

  Wire.begin(2,0);
  delay(1000);
    Serial.print(F("Temperature of bmp3 = "));
    Serial.print(bmp3.readTemperature());
    Serial.println(" *C");
    
    Serial.print(F("Pressure of bmp3 = "));
    Serial.print(bmp3.readPressure());
    Serial.println(" Pa");

    Serial.print(F("Approx altitude of bmp3 = "));
    Serial.print(bmp3.readAltitude(1013.25)); // this should be adjusted to your local forcase
    Serial.println(" m");
    
    Serial.println();
    
    Serial.print(F("Temperature of bmp4 = "));
    Serial.print(bmp4.readTemperature());
    Serial.println(" *C");
    
    Serial.print(F("Pressure of bmp4 = "));
    Serial.print(bmp4.readPressure());
    Serial.println(" Pa");

    Serial.print(F("Approx altitude of bmp4 = "));
    Serial.print(bmp4.readAltitude(1013.25)); // this should be adjusted to your local forcase
    Serial.println(" m");
    
    Serial.println();

    delay(2000);
}
```

DISCLAIMER:  I did not TEST this code ... so it might not work exactly ... but hopefully you get the idea.  This *should* allow you the four sensors - two per I2C bus - that you are looking for.

artemalek, 2017-06-12:
@Humancell  Thanks a lot I will check the code you have suggested. Do you think that it is possible to add more than two devices on a single bus? Or is it better to open a new bus? (from the aspect of efficiency)

pasko-zh, 2017-06-15:
In general you can add more than two i2c slaves on a i2c bus, you can even have multiple i2c masters. In the case of the BMP280 (an i2c slave) you cannot, since as @Humancell pointed out, there is only one bit to set i2c address of it. 

btw, if i2c speed should be critical for your application, you might want to have a look at my i2c library for the esp8266 [brzo_i2c](https://github.com/pasko-zh/brzo_i2c). Could be an option, since according to the BMP2880 datasheet "All modes (standard, fast, high speed) are supported".

earlephilhower, 2019-01-17:
Closing as a "How do I...?" question, not a bug or issue.

closed 2019-01-17 by earlephilhower