esp8266/Arduino #3335: Exception 3 when trying to connect to api.coinmarketcap.com

witnessmenow, 2017-06-07:
I am getting a Exception 3 when I try connect to coinmarketcap.com's api using WiFiClientSecure.  I tried one other library that I have that connects to a HTTPS endpoint using WiFiClientSecure and that worked, so not sure what is different between the two.

## Info
Device: Wemos D1 mini
Version of core: latest (updated from master yesterday)

## Sketch:
https://github.com/witnessmenow/arduino-coinmarketcap-api/blob/master/examples/ESP8266/GetTickerInfo/GetTickerInfo.ino

Last part of code of mine before the exception is:
https://github.com/witnessmenow/arduino-coinmarketcap-api/blob/master/src/CoinMarketCapApi.cpp#L15
(the serial print below this line does not get hit if its not commented out)

## Serial Output
```
 1384, room 16 
tail 8
chksum scandone
Connecting Wifi: Mikrotik
.....scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 8
cnt 

connected with Mikrotik, channel 1
dhcp client start...
.ip:192.168.1.35,mask:255.255.255.0,gw:192.168.1.1

WiFi connected
IP address: 
192.168.1.35
Getting ticker data for bitcoin
:ref 1
please start sntp first !
State:	sending Client Hello (1)
:wr 108 108 0
:wrc 108 108 0
:sent 108
:rn 1440
:rd 5, 1440, 0
:rdi 1440, 5
:rd 80, 1440, 5
:rdi 1435, 80
State:	receiving Server Hello (2)
:rd 5, 1440, 85
:rdi 1355, 5
:rd 1350, 1440, 90
:rdi 1350, 1350
:c0 1350, 1440
:rn 1440
:rd 1440, 1440, 0
:rdi 1440, 1440
:c0 1440, 1440
:rn 1440
:rd 1440, 1440, 0
:rdi 1440, 1440
:c0 1440, 1440
:rn 1252
:rd 1243, 1252, 0
:rdi 1252, 1243
State:	receiving Certificate (11)
Fatal exception 3(LoadStoreErrorCause):
epc1=0x4020ce03, epc2=0x00000000, epc3=0x00000000, excvaddr=0x40244450, depc=0x00000000

Exception (3):
epc1=0x4020ce03 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40244450 depc=0x00000000

[... 156 lines cut ...]
0x40216dcc: ssl_read at ssl/tls1.c line 2005
0x40204a21: SSLContext::connect(ClientContext*, char const*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 615
:  (inlined by) WiFiClientSecure::_connectSSL(char const*) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 327
0x40204116: WiFiClient::connect(IPAddress, unsigned short) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp line 324
0x40204ba5: WiFiClientSecure::connect(char const*, unsigned short) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 319
0x402022a5: CoinMarketCapApi::SendGetToCoinMarketCap(String) at /Users/brian/Documents/Arduino/libraries/arduino-coinmarketcap-api/src/CoinMarketCapApi.cpp line 15
0x402052b3: String::changeBuffer(unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x40205331: String::copy(char const*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x402053f0: String::operator=(String const&) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x40203538: CoinMarketCapApi::GetTickerInfo(String, String) at /Users/brian/Documents/Arduino/libraries/arduino-coinmarketcap-api/src/CoinMarketCapApi.cpp line 71
0x402052b3: String::changeBuffer(unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x40205331: String::copy(char const*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x4020537e: String::String(char const*) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x4020209a: printTickerData(String) at /var/folders/sl/1b3tnsqj3t9301xzfkh9g1900000gn/T/arduino_modified_sketch_875138/GetTickerInfo.ino line 51
0x4010053d: _umm_realloc at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1491
:  (inlined by) realloc at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1709
0x40204f9e: Print::print(unsigned long, int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 99
0x40201900: uart_start_isr at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/uart.c line 461
0x402052b3: String::changeBuffer(unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x402052ff: String::reserve(unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x40205331: String::copy(char const*, unsigned int) at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/WString.cpp line 439 (discriminator 4)
0x40202161: loop at /var/folders/sl/1b3tnsqj3t9301xzfkh9g1900000gn/T/arduino_modified_sketch_875138/GetTickerInfo.ino line 94
0x402056e8: loop_wrapper at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x4010070c: cont_norm at /Users/brian/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 109
```

witnessmenow, 2017-06-07:
Some more info:
On my old windows machine that is running old core (some version of 2.3 from github) the same code works fine.

witnessmenow, 2017-09-18:
This is still an issue for me, tested today using 2.4.0-rc1

Here is a basic sketch that reproduces the error. This sketch works fine with the 2.3.0 release

Is there a special cert or something that this site uses that causes issues?

`#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

WiFiClientSecure client;

char ssid[] = "SSID";       // your network SSID (name)
char password[] = "pass";  // your network key

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was Previously
  // connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  // Attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  IPAddress ip = WiFi.localIP();
  Serial.println(ip);

}

void connectToCoinMarket(){
  if (client.connect("api.coinmarketcap.com", 443)) {
    Serial.println(".... connected to server");
    char c;
    int ch_count=0;
    client.println("GET /v1/ticker/bitcoin HTTP/1.1");
    client.println("Host: api.coinmarketcap.com");
    client.println(F("User-Agent: arduino/1.0.0"));
    client.println();
    client.setTimeout(1200);
    char endOfHeaders[] = "\r\n\r\n";
    bool ok = client.find(endOfHeaders);

    if (!ok) {
      Serial.println("No response or invalid response!");
    }
  }
}

void loop() {
  Serial.println("Connecting To coinMarketCap");
  connectToCoinMarket();
  delay(2000);

}`

witnessmenow, 2017-09-19:
@igrr sorry to reference you directly but would you be able to advice on the next steps with this?

It's a pretty old issue at this stage so I'm not sure if me adding a new comment would get noticed, but I am also conscious of creating a duplicate to just bump it to the top

igrr, 2017-09-20:
This seems to be the same issue as reported in this comment: https://github.com/esp8266/Arduino/issues/3140#issuecomment-304667491, it was fixed in https://github.com/esp8266/Arduino/commit/c52b0da11cfd9666e45d576d3fdd2533415c4ab6 after 2.4.0-rc1 was released. Can you try the latest git version? (installation instructions are on the front page)

igrr, 2017-09-20:
Fixed in master, please give a try.

witnessmenow, 2017-09-20:
Thanks @igrr I'll try the git version after work this evening.

d-a-v, 2017-12-13:
@witnessmenow any feedback on this ?

closed 2018-01-02 by igrr

witnessmenow, 2018-01-05:
@igrr I know you fixed this a long time ago and I forgot to get back to you and then didn't get time to test since davs reminder, but just wanted to say that i tested this out on 2.4 today and it's working fine. Thanks!

commit c8947953 (2017-09-20): axtls: update to 66d530a, fixes #3335

SHA512 module stored padding in PROGMEM, but would then pass the pointer
to padding into Update function which would do a memcpy. Use the same
approach as with SHA256, that is to copy padding to the stack first, and
then pass it to Update.