esp8266/Arduino #3336: curl: (52) Empty reply from server

RebelDesigner, 2017-06-08:
I have a problem that sometimes my Sonoff relay module does receive requests, but fails to respond correctly. I have written a small sketch for a simple http server that handles three different requests. All works fine, for about 50% of the time. The other 50% of the time I get the following error when doing a request using `curl` requests from commandline.

```
$ curl http://192.168.178.220/off
curl: (52) Empty reply from server
```

The request is received OK (it seems); the ESP does indeed turn off the LED and the relay. The problem is that I should receive a http response with the body just containing the current state (in this case `0`).

No I have no idea if this is a problem in general with the ESP module, the Arduino library or my code. I do not tend towards the latter, since I have used similar code in the past on other chips. One thing I did notice is that the signal strength reported by my WiFi access point is between -31 and -40dB. That seems rather strong, but very well possible. The module is basically next to the access point (30cm). After moving it a bit, the signal is pretty much steady at -41dB. Still way too high for WiFi to function normally (this is what I have always thought, correct me if I am wrong). Could this be the case for the failure to transmit anything? Clearly the http response headers are not even received by the client.

I know this might not be related to the library, but I do think this is a reasonable place to assert this issue. On the web, there was no mention of such issues; while I do think this might be a common thing for some users. One suggestion might be to wrap the thing in some tin foil. Can anyone with experience comment on this?

### Sketch

```cpp

#define RELAY_PIN  12
#define LED_PIN    13

#include <ESP8266WiFi.h>

#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>

bool state = false;
uint32_t last_on = 0;

WiFiServer server(80);

void set_state(bool _state) {
  state = _state;
  digitalWrite(RELAY_PIN, state);
  digitalWrite(LED_PIN, !state);

  if (state)
    last_on = millis();
}

void reply_state(WiFiClient client) {
  client.flush();

  String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\nConnnection: close\r\n\r\n";
  s += state ? "1" : "0";
  s += "\n";

  client.print(s);
  delay(1);
}

void setup() {
  Serial.begin(115200);
  delay(10);

  pinMode(RELAY_PIN, OUTPUT);
  pinMode(LED_PIN, OUTPUT);

  set_state(LOW);

  WiFiManager wifiManager;
  wifiManager.autoConnect("AutoConnectAP");
  Serial.println("Connected");

  server.begin();
  Serial.println("Server started");
}

void loop() {
  if (millis() - last_on > 1000 * 60 * 15)
    set_state(LOW);

  WiFiClient client = server.available();
  if (!client)
    return;

  Serial.println("new client");
  while (!client.available())
    delay(1);

  String req = client.readStringUntil('\r');
  Serial.println(req);
  client.flush();

  if (req.indexOf("/off") != -1)
    set_state(LOW);
  else if (req.indexOf("/on") != -1)
    set_state(HIGH);
  else if (req.indexOf("/state") != -1) {
  } else {
    Serial.println("invalid request");
    client.stop();
    return;
  }

  reply_state(client);

  Serial.println("Client disonnected");
}
```

devyte, 2017-09-06:
In reply_state(), shouldn't client.flush() be below client.print(s)?

devyte, 2018-02-28:
Closing due to lack of feedback.

closed 2018-02-28 by devyte