esp8266/Arduino #3340: SSL Validation fails for some root certs

salqadri, 2017-06-10:
The following commit started causing a certain root cert to fail validation: 1d4185923805347ee3f010232006c77c4547dee1

To reproduce the issue, change the HTTPSRequestCACert.ino sample to use the following domain:
"securetoken.googleapis.com"

In the Cert.ino file, grab the cert data from the following file: https://github.com/salqadri/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/SSLCertificateVerification/SSLCertificateVerification.ino

Make sure to set the length to 804.

Run the sample, and notice the following errors:
WiFi connected
IP address: 
192.168.1.23
Setting time using SNTP.
Current time: Sun Jun 11 02:05:15 2017
connecting to securetoken.googleapis.com
ERROR: certificate verification failed!
connecting to securetoken.googleapis.com
connection failed

igrr, 2017-06-10:
Thanks for the report. Unfortunately I won't be able to reproduce the issue with googleapis.com because google is blocked here in China. I'll try a few other services to see if same thing happens... This may have something to do with key usage constraints checking which was merged from upstream axTLS, and similar issue was mentioned here: https://github.com/esp8266/Arduino/issues/1851#issuecomment-307495375 (also with a google domain).

salqadri, 2017-06-11:
That's unfortunate. The thing to note is that I isolated the problem specifically to this commit, so something happened between this axTLS pull and the last one: https://github.com/esp8266/Arduino/commit/1d4185923805347ee3f010232006c77c4547dee1

Defozo, 2017-06-28:
Any update on this?

muchtall, 2017-07-20:
@igrr , I'm having similar issues with certs from LetsEncrypt. Are you able to use those certs in China?

igrr, 2017-07-20:
Some of my automated tests of WiFiClientSecure use dl.espressif.com to test connection and download stuff. It uses Letsencrypt as its CA, and I haven't noticed issues with these certificates.

salqadri, 2017-07-29:
I just tried this again, and the issue still persists :(.

salqadri, 2017-07-30:
Based on my similar issue on the ESP32 (https://github.com/espressif/arduino-esp32/issues/545), I discovered there that the correct root cert to use for Google is the Geotrust Global CA, instead of the Equifax one I was using here. So, excitedly, I tried the Geotrust one with the latest axtls library hoping that it solves this problem. Unfortunately, I still get the error, so it seems something else is wrong :(.

denouche, 2017-09-08:
@salqadri in your fork you added the possibility to use ESP8266HTTPClient with RootCA, that's very interesting. But that's not in the main repo. 
It is possible to ask to merge your branch into the main repository, so everyone can enjoy your `HTTPClient::setRootCA` function and examples?
Thank you

Defozo, 2017-09-29:
Can we merge it as @denouche said?

salqadri, 2017-10-01:
Well the reason I can't really ask @igrr to merge it in, is because my fork uses an older version of the axtls binary. So we need to wait for this particular issue to be fixed with axtls not working with domains like securetoken.googleapis.com.

slurdge, 2017-10-26:
@igrr , I can't make the latest version work with LetsEncrypt at all. I can setup a test server for you if you need it. I tried both the immediate parent (LetsEncrypt) and the root and it doesn't validate, the error is -514.

igrr, 2017-10-26:
Could you please paste the debug output first? Enable debugging in Tools menu (core+TLS) and add Serial.setDebugOutput(true); to the setup. You should get a printout of all the certificates among other things.
514 is X509_VFY_ERROR_NO_TRUSTED_CERT, which means that you are providing wrong root certificate, or not loading the certificate at all, according to axTLS.

slurdge, 2017-10-26:
I could do that, what's the IDE where I would do it ? I don't find it in default Arduino IDE. I also figured out that -514 is X509_VFY_ERROR_NO_TRUSTED_CERT by digging the source. However, since axTLS does _not_ validate CACert upon loading, it's not easy to know if it's a wrong root certificate or not loading :smile:

slurdge, 2017-10-26:
Ok I made a stupid mistake in my DER conversion routine, and I could fix it. 
It still didn't work, but now I could investigate why: I was loading the following https://letsencrypt.org/certs/isrgrootx1.pem.txt as root, whereas for let's encrypt need this one: https://www.identrust.com/certificates/trustid/root-download-x3.html

After putting it, everything works! @igrr , thank you ! Please disregard my previous comment.

As a side note, it works well validating alt-names. Now, I don't know if SNI will work or not (but this is irrelevant to this ticker).

whyameye, 2017-12-02:
Based on [letencrypt's description](https://letsencrypt.org/certificates/) of its root certificates it seems both ISRG ROOT X1 and DST Root CA X3 should work. However, as @slurdge mentioned in the previous comment, only DST Root CA X3 works. I would much prefer to use ISRG ROOT X1 as that root expires in 2035 and DST Root CA X3 expires in 2021. What does ISRG ROOT X1 fail?

From https://letsencrypt.org/certificates/ :  

> There are two certificates available that both represent our intermediate. One is signed by DST Root CA X3, and the other is signed by ISRG Root X1.

[image]

closed 2018-01-02 by igrr