esp8266/Arduino #3342: Exception 28 on connect in WifiClient using 2.4.0-rc1

PhracturedBlue, 2017-06-11:
I am getting a crash in WifiClient::connect() (called by PubSubClient::connect()) usually within 100 failed attempts to connect to a non-existant MQTT broker.

```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 28 results
0x402079f1: ClientContext::connect(ip_addr*, unsigned short) at /home/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/libraries/ESP8266WiFi/src/WiFiClient.cpp line 155
:  (inlined by) WiFiClient::connect(IPAddress, unsigned short) at /home/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/libraries/ESP8266WiFi/src/WiFiClient.cpp line 135 
0x402079ea: WiFiClient::connect(IPAddress, unsigned short) at /home/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/libraries/ESP8266WiFi/src/WiFiClient.cpp line 155
0x40101d30: trc_NeedRTS at ?? line ? 
0x4020765c: WiFiClient::connect(char const*, unsigned short) at /home/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/libraries/ESP8266WiFi/src/WiFiClient.cpp line 155
0x4020d82d: PubSubClient::connected() at /home/Arduino/libraries/PubSubClient/src/PubSubClient.cpp line 586
0x40208a9f: PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, unsigned char, char const*) at /home/Arduino/libraries/PubSubClient/src/PubSubClient.cpp line 586
0x4020c76f: String::changeBuffer(unsigned int) at /home/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/WString.cpp line 589
0x40208bf4: PubSubClient::connect(char const*) at /home/Arduino/libraries/PubSubClient/src/PubSubClient.cpp line 586
0x4020c884: String::String(String const&) at /home/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/WString.cpp line 589
0x40209978: ESP8266MQTTMesh::connect_mqtt() at /home/Arduino/libraries/ESP8266MQTTMesh/ESP8266MQTTMesh.cpp line 448
0x4020d826: PubSubClient::connected() at /home/Arduino/libraries/PubSubClient/src/PubSubClient.cpp line 586
0x402092a1: ESP8266MQTTMesh::connected() at /home/Arduino/libraries/ESP8266MQTTMesh/ESP8266MQTTMesh.cpp line 274 (discriminator 4)
0x4023437a: wifi_get_opmode at ?? line ?
0x40235497: wifi_station_get_connect_status at ?? line ?
0x40206b67: loop_wrapper at /home/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/core_esp8266_main.cpp line 57 
0x4020b603: ESP8266MQTTMesh::loop() at /home/Arduino/libraries/ESP8266MQTTMesh/ESP8266MQTTMesh.cpp line 179
0x4023437a: wifi_get_opmode at ?? line ?
0x40235497: wifi_station_get_connect_status at ?? line ?
0x4020d6d0: std::__shared_count(__gnu_cxx::_Lock_policy)0>::~__shared_count() at /home/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/FS.cpp line 188
0x40207f84: ESP8266WiFiSTAClass::status() at /home/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp line 459
0x4020a005: ESP8266MQTTMesh::begin() at /home/Arduino/libraries/ESP8266MQTTMesh/ESP8266MQTTMesh.cpp line 119
0x40208fb4: _M_manager at /home/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 1931
0x4020d82d: PubSubClient::connected() at /home/Arduino/libraries/PubSubClient/src/PubSubClient.cpp line 586
0x40206e81: loop at /home/git/ESP8266MQTTMesh/examples/ESP8266MeshHelloWorld/ESP8266MeshHelloWorld.ino line 66 
0x402013b4: __pinMode at /home/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/core_esp8266_wiring_digital.c line 34 
0x40206e44: setup at /home/git/ESP8266MQTTMesh/examples/ESP8266MeshHelloWorld/ESP8266MeshHelloWorld.ino line 59 
0x40206b64: loop_wrapper at /home/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/core_esp8266_main.cpp line 57 
0x40100108: cont_norm at /home/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/cont.S line 109
```
### Basic Infos
#### Hardware
Hardware:		ESP-12
Core Version:      	2.4.0-rc1

### Description

Trying to test my custom  Mesh Network code:
https://github.com/PhracturedBlue/ESP8266MQTTMesh.git
examples/ESP8266MeshHelloWorld

The code in question connects to the WiFi network, gets an IP address, and tries to connect to the MQTT broker.  If the connection fails, it waits 5 seconds and tries again.  Eventually (usually within 100 tries) if it cannot create the connection, I see this stack trace.  I never experienced his crash when using the same code with the 2.3 release.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB/256kB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

andysc, 2018-03-09:
Did this ever get fixed?

I'm seeing same problem (Exception 28) quite reproducibly, by using WiFi hotspot on phone:

With mobile data on and wifi on, connect ESP8266 to an MQTT broker.
Turn off mobile data, to force the socket connection to break.
Wait for MQTT to notice the keep alive time has been exceeded so it goes into a reconnect phase.

Turn off WiFi on phone. 
Next time the MQTT connect happens in the retry loop, the Exception 28 occurs.

It's happening in the TCP connect call in PubSubClient.cpp

```
        if (domain != NULL) {
            result = _client->connect(this->domain, this->port);
        } else {
            result = _client->connect(this->ip, this->port);
        }
```
it doesn't get past that connect call when this happens.

```
Attempting MQTT connection...now
in connect - _client OK

Exception (28):
epc1=0x40203e25 epc2=0x00000000 epc3=0x00000000 excvaddr=0x01a7027b depc=0x00000000

ctx: cont 
sp: 3fff1100 end: 3fff14c0 offset: 01a0

>>>stack>>>
3fff12a0:  3fff00f0 00000000 3fff2864 40203e1e  
...
```

devyte, 2018-07-13:
Several fixes have been merged since this was reported related to WiFiClient. WiFiClient has since been stress tested.
Closing due to age. If there is still a problem, please open a new issue and fill out the required info.

closed 2018-07-13 by devyte