esp8266/Arduino #3345: esp8266 httpclient Image file posting

GreenturnLabs, 2017-06-12:
I am trying to post image files of size around 15Kb, form my esp8266 device to a non-sql database. The database support image attachment as base64 encoded data. I could post the json files of size less than 1Kb, But the http post fails for size grater than 1460 bytes.

`HTTPClient http;
if( http.begin(myurl) ){
   delay(1000);
   char *auth = authString;
   http.setAuthorization(auth);

   http.addHeader("Content-Type","application/json",  false, true);
   int err = http.POST(largejsondata);
   Serial1.print("HTTP Post reutrns : ");Serial1.println(err);
}`

d-a-v, 2017-06-12:
Dos it fail with an error, or simply http.POST() does not return ?

GreenturnLabs, 2017-06-12:
Returns "-3" . That means "HTTPC_ERROR_SEND_PAYLOAD_FAILED".
In ESP8266HTTPClient.h:
`#define HTTPC_ERROR_SEND_PAYLOAD_FAILED (-3)`

Hardware : NodeMCU1.0
Arduino Version : 2.3.0

d-a-v, 2017-06-19:
You could first try using the latest git version of esp8266/arduino which is the pre-2.4.0 version and contains a number of improvements.

devyte, 2017-09-06:
@GreenturnLabs any progress with this?

closed 2018-01-02 by igrr