esp8266/Arduino #3347: SPI ESP8266 slave not working correct.

newtrace, 2017-06-13:
Hi,
I have problems using the ESP8266 as SPI slave.
I had modified the demo code as follow:
    SPISlave.onData([](uint8_t * data, size_t len) {
      SPISlave.setData("OK");
    });
but the master can't correct to recieve,it can be found SPI timing no correct,"OK"(0x4f,0x4b) transfer 0x9f,0x96,it seem be shift a bit.

tonton81, 2017-11-05:
to add to this im testing the slave ino on esp8266 and master on teensy3.6

heres an interesting issue
running at 2mhz i get garbage
1mhz garbage
250khz i get intermittant replies (with/without garbage)
200khz, garbage

ameliaamelia, 2018-02-14:
I experienced the same problem.
The SPI Slave , on ESP8266, seems it is sending the reply data bytes with the bits left-shifted by 1.

korish, 2018-02-19:
@ameliaamelia I'm also facing the same problem.

I've connected two esp8266 via SPI and uploaded SPISlave_Master and SPISlave_Test from examples on them. The Slave receives data from Master correctly but Master received data left-shifted by 1.

Any ideas how to fix that?

ameliaamelia, 2018-02-19:
@korish 
I am using the **DUE** as master and the **ESP8266** as slave.
I do not know the environment of esp8266; I am amazed to see the **SPISlave_Master** and **SPISlave_Test** published when they are not working.
But since in my system all the messages are only ASCII 0x20<->0x7f, no bit 0x80 present, the bytes coming from the Slave I shift them >> 1 !!!
My project is to replace the boards with W5100 with a (my own) board with ESP8266.
I'm sorry for not helping

prateek027, 2018-11-20:
@korish,
for sending from master you can add 0x02 0x00 before sending 32 bytes, and when you send any data from slave to master then you can add 0x03 before sending data from slave to master. there is no additional setting required.

wuxx, 2019-10-04:
here is a solution, it seems that there is a bug in library.
https://cmheong.blogspot.com/2018/03/spi-master-slave-programs-with-esp8266.html