esp8266/Arduino #3348: esp8266 can't connect to server

warrior0, 2017-06-13:
i'm trying to connect esp8266(nodemcu v1.0) to my socket server(programming with node.js) the esp8266 can connect to my wifi but it can't connect to the server.
here is my esp8266 arduino ide code:
#include <ESP8266WiFi.h>
#include <SocketIOClient.h>
 
SocketIOClient client;
const char* ssid = "Family";          
const char* password = "bat3glendi";                            
 
char host[] = "192.168.1.2";   
int port = 3848;               
 
extern String RID;
extern String Rfull;

unsigned long previousMillis = 0;
long interval = 2000;
 
void setup(){

    Serial.begin(115200);
    delay(10);
    Serial.print("connect to: ");
    Serial.println(ssid);
 
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print('.');
    }
 
    Serial.println();
    Serial.println(F("connected"));
    Serial.print(F("ESP8266's ip: "));
    Serial.println(WiFi.localIP());
 
    if (client.connect(host, port)) {
        Serial.println(F("successful connect to socket server"));
        return;
    }
 
    if (client.connected()) {
        client.send("connection", "message", "Connected !!!!");
    }
    Serial.println(WiFi.localIP());
}
 
void loop()
{
   
    if (millis() - previousMillis > interval) {
      
        previousMillis = millis();
        client.send("atime", "message", "Time please?");
    }
 
    if (client.monitor()) {
        Serial.println(RID);
        Serial.println(Rfull);
    }
 
  
    if (!client.connected()) {
      client.reconnect(host, port);
    }
}
here is my socket server code:

var http = require('http');
var ip = require('ip');

var app = http.createServer(function(req,res){
	res.write('<h1> server good!</h1>');
	res.end();
});
var io = require('socket.io')(app);

app.listen(3848);
console.log('server ip: ' + ip.address() + ':' + 3848);

io.on('connection', function(socket){
	console.log('connected');

	socket.on('connection', function(message){
		console.log(message);
		socket.emit('welcome',{
			message: 'connect!!'
		});
	});
	socket.on('atime', function(data){
		sendTime();
		console.log(data);
	});
});

function sendTime() {
	
	//Đây là một chuỗi JSON
	var json = {
        ESP8266: 12,		
		soPi: 3.14,						
		time: new Date()						
    }
    io.sockets.emit('atime', json);
}

tanaydeshmukh1, 2017-07-07:
please provide me link for the <ESP8266WiFi.h> library........

devyte, 2017-09-05:
This is not the right place to ask for assistance on your particular project. This is an issue tracker for the core libs.
Please refer to esp8266.com or stackoverflow or similar for discussion/assistance.
Closing.

closed 2017-09-05 by devyte