esp8266/Arduino #3351: PROGMEM/F: section type conflict with __c

bertmelis, 2017-06-13:
I haven't got enough understanding of the inner working to assert if this is a coding error, or a bug, but I've got the following:

sketch:
```C++
#include "TestClass.h"

TestClass testClass;
const uint32_t INTERVAL = 5000;

void setup() {
  Serial.begin(115200);
  testClass.setPrinter(&Serial);
}

void loop() {
  static uint32_t previousMillis = 0;
  if ((millis() - previousMillis) > INTERVAL) {
    previousMillis = millis();
    testClass.doSomething();
  }
}
```
TestClass.h:
```C++
#pragma once
#include <Arduino.h>

class TestClass {
  public:
    void setPrinter(HardwareSerial* serial);
    void doSomething();
  private:
    static const char* _testString;
    HardwareSerial* _serial;
};
```
TestClass.cpp:
```C++
#include "TestClass.h"

const char* TestClass::_testString PROGMEM = "Test String";

void TestClass::setPrinter(HardwareSerial* serial){
  _serial = serial;
}

void TestClass::doSomething() {
  //do something
  _serial->println(F("testing"));
}
```

Compiling results in an error: `TestClass.cpp:3: error: TestClass::_testString causes a section type conflict with __c`

Removing the `F` macro solved the error as does removing the PROGMEM modifier. However, I'd like to use both. eg: list of error messages and printing small debug messages, both from progmem.

bertmelis, 2017-06-13:
OK, it obviously had nothing to do with being is a class (and so being in a seperate file). You just cannot use `PROGMEM` and `F()` in the same file.

Makuna, 2017-06-23:
Please see https://github.com/esp8266/Arduino/issues/3369 for a possible solution.

devyte, 2017-09-06:
@bertmelis did you test the solution found in #3369?

bertmelis, 2017-09-14:
Or I just don't understand the solution, or it doesn't solve my "error".

I'm trying to put debug messages in PROGMEM using an array of char pointers, as in [Nick Gammon's blogpost](http://www.gammon.com.au/progmem). These debug messages are used in a class-library, for now without any namespace defined. I had all the declarations and initialisations in a header file.

Makuna, 2017-09-14:
While the declaration can be, the initializations can't be in the header; as mentioned in #3369.

bertmelis, 2017-09-15:
But then also: I've got a class method with `Serial.println(F("debug text"));`. This works untill I `inline` the method. Despite the proposed solution, it is quite unhandy.

But I'll ask it in another way: are there any thoughts on improving the issue? Or shall I live with it?

(in case of the latter, I'll modify my code to deal with it and close the issue)

Makuna, 2017-09-18:
You can move the actual string into a const variable that is declared in the header, leaving the Sertal.println in place. 

This is really what you would want anyhow.  

Currently, anything in a F("") will be a duplicated string, even if the text is the same.  For any work other than simple examples and sketches, you really want a separate .h/.cpp that contain all your strings.

devyte, 2018-02-28:
Closing due to user error.

closed 2018-02-28 by devyte