esp8266/Arduino #3352: Multiline raw Strings not parsed in Macro

romanseidl, 2017-06-13:
### Basic Infos

The esp8266 toolkit uses gcc 4.8.2 an thus it fails to parse a multiline raw String when it is passed as an argument to a macro. The same sketch as below will build without errors when using the original Arduino toolkit (when switching to AVR / Arduino Hardware in the Arduino IDE).

This was fixed in gcc 4.9 : 
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=57824

### Sketch

```cpp

#include <Arduino.h>

#define S(s) s

const char y[] = S(R"(
abc
)");

void setup() {
}

void loop() {
}

```

### Error Message

```
sketch_jun13a:3: error: unterminated raw string
 const char y[] = S(R"(
                    ^
exit status 1
unterminated raw string
```

Is there a reason on why you are using the 4.8.2 version of gcc? Max Filippov seems to provide newer versions on https://github.com/jcmvbkbc/linux-xtensa

devyte, 2017-09-05:
The link you point to seems to be a linux port for xtensa architecture, which means running linux on xtensa. What we are using is what is called a cross-compiler: a compiler that runs on x64 architecure, but that produces a binary meant to be run on xtensa.
Is there a cross-compiler in that repo somewhere? At first glance I didn't see one (I didn't look thoroughly).

The reason we are using 4.8.2 is that the toolchain comes mostly from espressif, and that compiler version is the official one for the espressif SDK. Keep in mind that these core libs are built on top of the espressif SDK, and that SDK has elements that are distributed as precompiled binary blobs (they are proprietary code). In theory, there could be binary incompatibilities between outputs (I have already encountered this on x64 compilers).

On the other hand, I wouldn't mind moving to a new compiler. I invite you to:
- pull the compiler out of the repo, if it is there
- test it by temporarily replacing the current official compiler
- load a simple test sketch compiled with it into an ESP

If you are willing to pick up the challenge, I can provide some guidance, or can look for someone else to provide it.

devyte, 2018-01-08:
@igrr I just found out that with gcc 5 there is a change in abi, which could impact us when we move to a newer gcc, e.g.: 5.2. I don't know details, though.

igrr, 2018-01-08:
As long as you don't link pre-5 code which uses libstdc++ with new libstdc++, all should be fine.

devyte, 2018-01-15:
Changing milestone due to high impact.

PR #6294 (2020-07-07): Upgrade to GCC 10.1 toolchain
All examples I've tested run fine.  GDB is working.  No warnings.

Fixes #3352
closed 2020-07-07 by earlephilhower