esp8266/Arduino #3355: ConfigFile example

nguyenhunga5, 2017-06-14:
Hi,

I have problem with esp-12f module, after try ConfigFile, OTA update examples with latest Arduino IDE, my module cannot boot. I don't known why.

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12F
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		4MB
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

ConfigFile example sketch

### Debug Messages
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 441 results
0x40101ca3: trc_NeedRTS at ?? line ?
0x401021a6: wDev_ProcessFiq at ?? line ?
0x40101f40: wDev_ProcessFiq at ?? line ?
0x40201569: uart1_write_char_d at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_postmortem.c line 197
:  (inlined by) uart_write_char_d at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_postmortem.c line 161
0x40207e9c: esp_yield at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x402010a7: delay at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.c line 51
0x402017c4: uart0_write_char at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/uart.c line 461
0x4020170d: print_stack at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_postmortem.c line 197
:  (inlined by) __wrap_system_restart_local at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_postmortem.c line 119
0x401044c4: wdt_feed at ?? line ?
0x4022c788: user_uart_wait_tx_fifo_empty at ?? line ?
0x4023ad80: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4022c788: user_uart_wait_tx_fifo_empty at ?? line ?
0x4022c86b: user_uart_wait_tx_fifo_empty at ?? line ?
0x4022c788: user_uart_wait_tx_fifo_empty at ?? line ?
0x401044e3: wdt_feed at ?? line ?
0x401044c4: wdt_feed at ?? line ?
0x4022c788: user_uart_wait_tx_fifo_empty at ?? line ?
0x4022c788: user_uart_wait_tx_fifo_empty at ?? line ?
0x401044e3: wdt_feed at ?? line ?
0x401044c4: wdt_feed at ?? line ?
0x40102243: wDev_ProcessFiq at ?? line ?
0x401044dd: wdt_feed at ?? line ?
0x401044da: wdt_feed at ?? line ?
0x40101f40: wDev_ProcessFiq at ?? line ?
0x4022c788: user_uart_wait_tx_fifo_empty at ?? line ?
0x401044e3: wdt_feed at ?? line ?
0x401044c4: wdt_feed at ?? line ?
0x401044dd: wdt_feed at ?? line ?
0x401044da: wdt_feed at ?? line ?
0x4022c788: user_uart_wait_tx_fifo_empty at ?? line ?
0x401044e3: wdt_feed at ?? line ?
0x401044c4: wdt_feed at ?? line ?
0x401044dd: wdt_feed at ?? line ?
0x401044da: wdt_feed at ?? line ?
0x4022c788: user_uart_wait_tx_fifo_empty at ?? line ?
[... 307 lines cut ...]
0x40208dac: EspClass::flashEraseSector(unsigned int) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Esp.cpp line 507
0x40207f54: optimistic_yield at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x402034dc: spiffs_erase_block at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/spiffs/spiffs_nucleus.c line 941
0x40208b3c: spiffs_hal_read(unsigned int, unsigned int, unsigned char*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/spiffs_hal.cpp line 45
0x40208bf8: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/spiffs_hal.cpp line 102
0x40208d34: spiffs_hal_erase(unsigned int, unsigned int) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/spiffs_hal.cpp line 168
0x4020206c: SPIFFS_format at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/spiffs/spiffs_hydrogen.c line 981
0x40208a10: SPIFFSImpl::begin() at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/spiffs_api.h line 128
0x402010b2: delay at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.c line 54
0x40209185: fs::FS::begin() at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/FS.cpp line 251
0x40207808: setup at /Users/hungnguyenthanh/Documents/untitled folder/ConfigFile/ConfigFile.ino line 78
0x40207ee0: loop_wrapper at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x40100108: cont_norm at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 109
0x4022c7a0: user_uart_wait_tx_fifo_empty at ?? line ?
0x402106c4: dns_recv at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/dns.c line 300
0x40212980: arp_timer at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/timers.c line 372
0x40212960: dhcp_timer_coarse at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/timers.c line 372
0x40212940: dhcp_timer_fine at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/timers.c line 372
0x40212920: igmp_timer at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/timers.c line 372
0x40212900: dns_timer at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/timers.c line 372
0x4021283c: tcpip_tcp_timer at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/timers.c line 372
0x40224cfc: cnx_connect_timeout at ?? line ?
```
messages here
```

devyte, 2017-09-05:
Are you working from Windows? Are you programming trying to flash from the Arduino IDE? Specifically which sketch is giving you trouble? Which core version? Which board? or is it a standalone ESP-12?
I can confirm that both 2.3.0 and the current git works fine on a all of Nodemcu v3 lolin, Wittycloud, and Wemos D1 mini v2, so I suspect user error of some kind.

nguyenhunga5, 2017-09-06:
hi @devyte My fault, my power not enough.

closed 2017-09-06 by nguyenhunga5