esp8266/Arduino #3358: section `.text' will not fit in regi on `iram1_0_seg' when using std::ostringstream

TimWoosh, 2017-06-16:
### Basic Infos
I hope I'm providing sufficient information. I tried my best to collect as much as possible

#### Hardware
Hardware:			ESP-8266MOD
Core Version:      	3.4.0b12

### Description

Problem description

Compiler error when  std::ostringstream is used.
When commenting out all depenend lines and replacing ```+ convert.str() + ``` with a 1 (inside the quotes) in the line below, all compiles fine.:
```  
std::string msgStr = "{\"message\" : \"hello\", \"msgNum\": \"" + convert.str() + "\"}"; 
```
### Settings in IDE

Module:  			d1_mini
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		n/a
Reset Method:		n/a

### Sketch

```cpp

#include <Arduino.h>
#include <PubSubClient.h>
#include <ESP8266WiFi.h>
#include <string>
#include <sstream>

const char* ssid = "ssid";
const char* password = "blabla";
const char* mqtt_server = "192.168.0.100:1883";

WiFiClient espClient;
PubSubClient client(espClient);

const byte ledPin = 15;
int counter = 0;
const char *msg;

void callback(char* topic, byte* payload, unsigned int length) {
 Serial.print("Message arrived [");
 Serial.print(topic);
 Serial.print("] ");
 for (int i=0;i<length;i++) {
  char receivedChar = (char)payload[i];
  Serial.print(receivedChar);
  if (receivedChar == '0')
  // ESP8266 Huzzah outputs are "reversed"
  digitalWrite(ledPin, HIGH);
  if (receivedChar == '1')
   digitalWrite(ledPin, LOW);
  }
  Serial.println();
}

void reconnect() {
 // Loop until we're reconnected
 while (!client.connected()) {
 Serial.print("Attempting MQTT connection...");
 // Attempt to connect
 if (client.connect("ESP8266 Client")) {
  Serial.println("connected");
  // ... and subscribe to topic
  client.subscribe("ledStatus");
 } else {
  Serial.print("failed, rc=");
  Serial.print(client.state());
  Serial.println(" try again in 5 seconds");
  // Wait 5 seconds before retrying
  delay(5000);
  }
 }
}

void setup() {
  Serial.begin(9600);
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);

  pinMode(ledPin, OUTPUT);
}

void loop()
{
 if (!client.connected()) {
  reconnect();
 }
 counter++;
 std::ostringstream convert;
 convert << counter;
 std::string msgStr = "{\"message\" : \"hello\", \"msgNum\": \"" + convert.str() + "\"}";

 msg = msgStr.c_str();
 client.publish("test/hello", msg);
 client.loop();
 delay(5000);
}

```
### Compiler output

```
[06/16/17 23:39:21] Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
 
Verbose mode can be enabled via `-v, --verbose` option
Collected 24 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <PubSubClient> v2.6

|-- <ESP8266WiFi> v1.0
Linking .pioenvs\d1_mini\firmware.elf
c:/users/gebruiker/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pioenvs\d1_mini\firmware.elf section `.text' will not fit in regi
on `iram1_0_seg'
C:\users\gebruiker\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib\libstdc++.a(ostream-inst.o):(.literal._ZNSo9_M_insertIlEERSoT_[_ZNSo9_M_insertIlEERSoT_]+0x0): undefined reference to
 `std::__throw_bad_cast()'
C:\users\gebruiker\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib\libstdc++.a(ostream-inst.o): In function `std::ostream& std::ostream::_M_insert<long>(long)':
/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/ostream:544: undefined reference to `std::__throw_bad_cast()'
C:\users\gebruiker\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib\libstdc++.a(wlocale-inst.o):(.literal._ZNKSt7collateIwE10do_compareEPKwS2_S2_S2_[_ZNKSt7collateIwE10do_compareEPKwS2_
S2_S2_]+0x0): undefined reference to `wcslen'
C:\users\gebruiker\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib\libstdc++.a(wlocale-inst.o):(.literal._ZSt16__convert_from_vRKPiPciPKcz[_ZSt16__convert_from_vRKPiPciPKcz]+0x4): unde
fined reference to `setlocale'
C:\users\gebruiker\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib\libstdc++.a(wlocale-inst.o):(.literal._ZSt16__convert_from_vRKPiPciPKcz[_ZSt16__convert_from_vRKPiPciPKcz]+0x10): und
efined reference to `vsprintf'
C:\users\gebruiker\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib\libstdc++.a(wlocale-inst.o):(.literal._ZNKSt7num_putIwSt19ostreambuf_iteratorIwSt11char_traitsIwEEE14_M_group_floatEP
KcjwPKwPwS9_Ri[_ZNKSt7num_putIwSt19ostreambuf_iteratorIwSt11char_traitsIwEEE14_M_group_floatEPKcjwPKwPwS9_Ri]+0x0): undefined reference to `wmemcpy'
C:\users\gebruiker\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib\libstdc++.a(wlocale-inst.o):(.literal._ZNSt5__padIwSt11char_traitsIwEE6_S_padERSt8ios_basewPwPKwii[_ZNSt5__padIwSt11c
har_traitsIwEE6_S_padERSt8ios_basewPwPKwii]+0x0): undefined reference to `wmemset'
C:\users\gebruiker\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib\libstdc++.a(wlocale-inst.o):(.literal._ZNKSt7num_putIwSt19ostreambuf_iteratorIwSt11char_traitsIwEEE15_M_insert_floatI
dEES3_S3_RSt8ios_basewcT_[_ZNKSt7num_putIwSt19ostreambuf_iteratorIwSt11char_traitsIwEEE15_M_insert_floatIdEES3_S3_RSt8ios_basewcT_]+0x0): undefined reference to `memchr'
C:\users\gebruiker\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib\libstdc++.a(wlocale-inst.o):(.literal._ZNKSt9money_getIwSt19istreambuf_iteratorIwSt11char_traitsIwEEE10_M_extractILb0
EEES3_S3_S3_RSt8ios_baseRSt12_Ios_IostateRSs[_ZNKSt9money_getIwSt19istreambuf_iteratorIwSt11char_traitsIwEEE10_M_extractILb0EEES3_S3_S3_RSt8ios_baseRSt12_Ios_IostateRSs]+0x4): undefined reference to `wm
emchr'
```
Followed by more "undefined reference" messages. I think the first line after "Linking .pioenvs\d1_mini\firmware.elf" is the most important.

### Debug Messages

```
messages here
```

JohnGlassmyer, 2017-08-05:
I am also getting errors of undefined reference to std::__throw_bad_cast() when trying to use istringstream, on 2.4.0-rc1.

e.g.:
/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/locale_facets_nonio.tcc:1173: undefined reference to `std::__throw_bad_cast()'

I also get "multiple definition of `_putc_r'" when I try to use istringstream:
/Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/putc.c:97: multiple definition of `_putc_r'
C:\Users\stil\AppData\Local\Temp\arduino_build_872957/arduino.ar(libc_replacements.c.o):C:\Users\stil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/libc_replacements.c:96: first defined here

Removing the use of istringstream makes both errors (__throw_bad_cast and _putc_r) go away.

DanielOgorchock, 2018-06-01:
I ran into a similar issue with multiple definitions of _putc_r and am using this workaround:

A _putc_r definition is found on line 96 of packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/libc_replacements.c

Simply commenting it out allowed me to compile.

shineangelic, 2018-07-11:
I was able to "solve" this adding stubs to my sketch:

```
namespace std {

void __throw_bad_cast(void)
{
    // print error and stop execution
}
void __throw_ios_failure(char const*){}
void __throw_runtime_error(char const*){}

} // namespace std
```

Idea comes from https://github.com/nlohmann/json/issues/1111 above

closed 2018-08-26 by devyte