esp8266/Arduino #3360: WiFiClientSecure wdt reset caused by time out to high?

Jeroen88, 2017-06-18:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	?2.1.0-rc2?

### Description

Problem description. While using WiFiClientSecure I get a watch dog reset:
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

I searched through the code, and found that in WiFiClientSecure.cpp at line 327 _ssl->connect(_client, hostName, 5000); is called. The 5000 is a time out that is passed through the connect function at line 91. In this function at line 109 is a loop that may run for 5000 milliseconds. 

May this cause the wdt reset?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

cptang2007, 2017-06-28:
Thanks and I can also repeat the same issue while using ESP-12F @80MHz.
I am using the WiFiClientSecure to connect to the Mosquitto MQTT server with SSL.

Some testing results:
1. ESP-12F@80MHz > can't connect, my PubSubClient report a MQTT_CONNECT_FAILED
2. ESP-12F@160MHz with debug on "core+SSL" > usually can connect with small number of connection failures
3. ESP-12F@160MHz without debug > All connection trials success

Probably 160Mz is marginally enough to complete the connection within 5000ms
This problem is gone @80MHz if the timeout value is changed to a larger value [e.g. 20000ms]  in the WiFiClientSecure.cpp

ESP.wdtDisable() seems no effect

earlephilhower, 2019-01-16:
Closing as related to axtls (deprecated).  If still around w/BearSSL, please open a new issue.

closed 2019-01-16 by earlephilhower