esp8266/Arduino #3361: Arduino Uploads OK, but ESP run old program

Shumatic, 2017-06-19:
Hi

I've been using Arduino 1.8.3 for a while, without problems. but now i cannot upload new programs, everything seems to be alright, and the IDE show progress bar and ends with DONE as normal, but the ESP still runs the old program.
There's no change as to what speed I use to upload (115200 512000 etc.)
I have tried 2 new ESP12E and also just the BLINK sketch, but without luck.
I have reinstalled my PC, but it still dosn't work

I have an old ESP, that is uploading fine.
EDIT
If i upload the Espressif firmware whit there tolle i get AT command , so this upload works.
after i uploaded my own code, and now it worked too? any suggestions??

### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	Dont know.

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module?
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL?
Reset Method:		ck

### Sketch

Dosn't matter, not even blink Sketch work

### Debug Messages

```
No errors
```

mape1991, 2017-06-19:
It happened for me several times due to HW issues. I have used the DIO flash mode with 115200 baudrate. If the pins of the ESP are not in the correct state (0 or 3.3V) the Arduino flash is processing but the ESP is not writting anything in the bootloader. Don't know why this happens, probably it is the ESP that doesn't detect a bad boot state.

Check the pins VIN_3V3, GPIO_0, GPIO_15 and GPIO_2 of your board

Shumatic, 2017-06-19:
I'll do that later today.
Though it is funny that it wotks in Espressifs programmer....

git-tiger, 2017-06-21:
I have seen this in Platformio too. I had re-install the tools on a new computer. 
Uploads 100%, but after boot, no output on the serial.
By some strange luck, it finally booted with serial output, and unit behaves as before.

Shumatic, 2017-06-21:
Hi mape1991
The error disappeared, that is very annoying.
I don't understand your comment about GPIO_2, normally i set
GPIO0 and CH_PD High through 10K
GPIO15 LOW through 10K
on ESP12E the Reset already have a 10K pull up
VCC are powered from a separate supply

To git-tiger
Yes, it is pretty frustrating that we can't isolate to cause.

beaglesparks, 2017-06-24:
I have a similar problem which I've half solved but i'm looking for a complete solution. Hardware:	NodeMCU 1.0 ESP-12E Module

my circuit is simple.  I'm running an IR remote sketch which works.  At pin D3 in series with an IR led. At D7 I have a visible LED.  Both are in series with a resister connected to ground.

If i power up with the IR LED wired in, both LEDs are on dimly and stay on (they should be off).  I can flash my sketch and the Arduino IDE displays as expected but the sketch never runs.  

If i power up with the IR LED not wired in, the visible LED will be off.  If i then wire in the IR LED they both stay off as they should.  The sketch then runs as it should.

I've tried replacing the LED

So my suggestion to you is try taking the load of the board during power up.  Does anyone have a suggestion for me so i don't have to un-wire and the re-wire the IR LED?  A power capacitor maybe?

devyte, 2017-06-24:
Isn't D3 used for the flash? I suggest double checking which pins are
really "free", and make sure you're not confusing the D numbering with the
GPIO numbering.

beaglesparks, 2017-06-24:
thanks devyte.  Exactly.  I switched to D2 and it works fine!

rolpal, 2017-08-06:
I have a similar problem with the 1MB versions of ESP8266-01.
In the HW setup I have the 512KB version runns perfect, flashes and executes just fine.
When replaced by a 1MB version it uploads but it never executes, not even "blink"...
Running esptool.py i can communicate with the 1MB version and even verify the flash with the .bin file from the IDE (1.6.8 with board 2.3.0).
I know there is a different wiring of the 1MB where reset is no longer reset but deep_sleep_wakeup.

closed 2018-01-02 by igrr