esp8266/Arduino #3367: Exception (3) thrown during clear() or when destroying vectors

Defozo, 2017-06-22:
### Description

Sometimes I get exception 3 when trying to clear() a std::vector<double>.

### Hardware

 - Hardware: ESP-07S
 - ESP8266-Arduino library core version: 2.4.0-rc1

### Settings in IDE

 - Module: Generic ESP8266 Module
 - Flash Size: 4MB
 - CPU Frequency:	160Mhz
 - Flash Mode:	qio
 - Flash Frequency:	40Mhz
 - Upload Using:	OTA / SERIAL
 - Reset Method:	ck

### Stacktrace

This is stacktrace from Arduino's Exception Decoder:

    Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
    Decoding 30 results
    0x401000b7: umm_disconnect_from_free_list at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\umm_malloc/umm_malloc.c line 1151
    0x4021e9bc: memmove at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/string/../../../.././newlib/libc/string/memmove.c line 69 (discriminator 1)
    0x40217464: Print::write(unsigned char const*, unsigned int) at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/Print.cpp line 38
    0x4010020c: _umm_free at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\umm_malloc/umm_malloc.c line 1287
    0x4010068c: free at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\umm_malloc/umm_malloc.c line 1733
    0x40215b04: operator delete(void*) at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/abi.cpp line 57
    0x40207940: std::_Vector_base   >::_M_deallocate(double*, unsigned int) at C:\Users\Defozo\Documents\Arduino\jeszcze nowsza wersja laser_na_stacje_v2\laser_na_stacje_v2/laser_na_stacje_v2.ino line 565
    0x402088ac: std::_Vector_base   >::~_Vector_base() at c:\users\defozo\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits/stl_vector.h line 161
    0x40209748: std::vector   >::_M_erase_at_end(double*) at c:\users\defozo\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits/stl_vector.h line 1353
    :  (inlined by) std::vector   >::clear() at c:\users\defozo\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits/stl_vector.h line 1126
    :  (inlined by) loop at C:\Users\Defozo\Documents\Arduino\jeszcze nowsza wersja laser_na_stacje_v2\laser_na_stacje_v2/laser_na_stacje_v2.ino line 412
    0x40217082: std::default_delete ::operator()(TransportTraits*) const at c:\users\defozo\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits/unique_ptr.h line 68
    0x4010020c: _umm_free at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\umm_malloc/umm_malloc.c line 1287
    0x401020ea: wDev_ProcessFiq at ?? line ?
    0x40215d84: loop_wrapper at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_main.cpp line 124
    0x4010070c: cont_norm at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/cont.S line 109

This is reset cause:

    Hardware Watchdog
    Fatal exception:4 flag:1 (WDT) epc1:0x40000f83 epc2:0x00000000 epc3:0x4000dd3b excvaddr:0x40023c5c depc:0x00000000

### Code

I'm initializing vectors like that:

    std::vector<double> dhtTempVector, dhtHumidityVector;
    std::vector<double> dhtTempMedianVector, dhtHumidityMedianVector;
    std::vector<double> bmpTempVector, bmpPressureVector;
    std::vector<double> bmpTempMedianVector, bmpPressureMedianVector;
    dhtTempVector.reserve(5);
    dhtHumidityVector.reserve(5);
    bmpTempVector.reserve(5);
    bmpPressureVector.reserve(5);
    dhtTempMedianVector.reserve(6);
    dhtHumidityMedianVector.reserve(6);
    bmpPressureMedianVector.reserve(6);
    bmpTempMedianVector.reserve(6);

Inserting new values like that:

    if (!isnan(temp) && !isnan(hum)) {
        dhtTempVector.push_back(temp);
        dhtHumidityVector.push_back(hum);
    }

and:

	if (!dhtTempVector.empty()) {
		dhtTempMedianVector.push_back(median(dhtTempVector));
		dhtTempVector.clear();
	}
	if (!dhtHumidityVector.empty()) {
		dhtHumidityMedianVector.push_back(median(dhtHumidityVector));
		dhtHumidityVector.clear(); //line number 412
	}

This is the median function:

    double median(std::vector<double> v)
    {
        size_t n = v.size() / 2;
        std::nth_element(v.begin(), v.begin()+n, v.end());
        double vn = v[n];
        if(v.size()%2 == 1) {
            return vn;
        } else  {
            std::nth_element(v.begin(), v.begin()+n-1, v.end());
            return 0.5*(vn+v[n-1]);
        }
    }

### Specific lines from stacktrace:

Here is **umm_malloc.c**:
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/umm_malloc/umm_malloc.c

**Print.cpp**:
https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/cores/esp8266/Print.cpp

**cont.S**:
https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/cores/esp8266/cont.S

Specific **lines from umm_malloc.c**:

1151:

    UMM_NFREE(UMM_PFREE(c)) = UMM_NFREE(c);

https://github.com/esp8266/Arduino/blob/master/cores/esp8266/umm_malloc/umm_malloc.c#L1151

1287:

    if( UMM_NBLOCK(UMM_PBLOCK(c)) & UMM_FREELIST_MASK ) {

https://github.com/esp8266/Arduino/blob/master/cores/esp8266/umm_malloc/umm_malloc.c#L1287

1733:

    void umm_free( void *ptr ) {
    
      ptr = GET_UNPOISONED(ptr);
    
      /* check poison of each blocks, if poisoning is enabled */
      if (!CHECK_POISON_ALL_BLOCKS()) {
        return;
      }
    
      /* check full integrity of the heap, if this check is enabled */
      if (!INTEGRITY_CHECK()) {
        return;
      }
    
      _umm_free( ptr );
    } //1733

https://github.com/esp8266/Arduino/blob/master/cores/esp8266/umm_malloc/umm_malloc.c#L1733

Specific **line from Print.cpp**:

    size_t Print::write(const uint8_t *buffer, size_t size) {
        size_t n = 0;
        while(size--) {
            n += write(*buffer++); //line number 38
        }
        return n;
    }

https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/cores/esp8266/Print.cpp#L38

Specific **line from core_esp8266_main.cpp**:

    static void loop_wrapper() {
        static bool setup_done = false;
        preloop_update_frequency();
        if(!setup_done) {
            setup();
    #ifdef DEBUG_ESP_PORT
            DEBUG_ESP_PORT.setDebugOutput(true);
    #endif
            setup_done = true;
        }
        loop();
        run_scheduled_functions(); //line number 124
        esp_schedule();
    }

Line number 109 from cont.S:

https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/cores/esp8266/cont.S#L109

### Additional debug information

This is a result:

    dhtTempVector size: 4
    dhtTempMedianVector size: 5
    dhtHumidityVector size: 4
    dhtHumidityMedianVector size: 5
    29.00 29.00 29.00 29.00 --
    28.70 28.75 28.80 28.90 28.90 --
    41.20 41.30 41.80 41.40 --
    42.65 42.55 41.80 42.30 41.75 --

of these lines of code:

    Serial.print("dhtTempVector size: ");
	Serial.println(dhtTempVector.size());
	Serial.print("dhtTempMedianVector size: ");
	Serial.println(dhtTempMedianVector.size());
	Serial.print("dhtHumidityVector size: ");
	Serial.println(dhtHumidityVector.size());
	Serial.print("dhtHumidityMedianVector size: ");
	Serial.println(dhtHumidityMedianVector.size());
	for (int k = 0; k < dhtTempVector.size(); ++k) {
		Serial.print(dhtTempVector.at(k));
		Serial.print(" ");
	}
	Serial.println("--");
	for (int k = 0; k < dhtTempMedianVector.size(); ++k) {
		Serial.print(dhtTempMedianVector.at(k));
		Serial.print(" ");
	}
	Serial.println("--");
	for (int k = 0; k < dhtHumidityVector.size(); ++k) {
		Serial.print(dhtHumidityVector.at(k));
		Serial.print(" ");
	}
	Serial.println("--");
	for (int k = 0; k < dhtHumidityMedianVector.size(); ++k) {
		Serial.print(dhtHumidityMedianVector.at(k));
		Serial.print(" ");
	}
	Serial.println("--");

just before the exception is thrown.

### Summary

It looks like the exception is thrown during clearing the vector. I think I experienced exceptions thrown at the end of the loop() function too - so at the time when destructors of the vectors are fired. I'd think it's a problem with destroying the elements of the vectors but they're just double numbers.

What could be the problem?

devyte, 2017-09-05:
You ahve posted pieces of code, which by themselves are difficult to understand. Please post a complete minimal sketch that reproduces the issue, so that complete context can be understood.
I am using std::vector<double> myself, and it works fine, including destruction. I therefore suspect memory corruption, such as writing past the boundaries of some buffer.

devyte, 2018-01-08:
I can't reproduce any of the issues described. I use std::vector, std::map, std::list in my own app, and they are also used in several places in our core libs.
Closing due to can't reproduce, and lack of feedback in several months.

closed 2018-01-08 by devyte