esp8266/Arduino #3368: Switching between STA and AP fails

DynamoBen, 2017-06-22:
### Basic Info

#### Hardware
Hardware:			ESP-12
Core Version:      	2.3.0

### Description

I can enter AP mode shortly after connecting as client, however if I wait 1 minute or more and try to enter AP mode it stays in STA. Can't tell if I'm doing something wrong or if this is a bug.

Repro Steps:
1. Allow ESP to connect as client.
2. Wait 1+ minutes
3. Press and hold config switch until AP mode is entered
NOTICE: Mode = STA not AP as it should

Below is example code to demonstrate what I'm seeing.

### Settings in IDE

Module:  			Adafruit Huzzah
Flash Size: 		4MB/3MB
CPU Frequency:       80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp

/* 
ESP8266: STA vs AP
Hardware: Adafruit Huzzah
*/

#include <ESP8266WiFi.h>          // ESP8266 Core WiFi Library
#include <WiFiClient.h>

const char* ssid = "Linksys";  // <----------------- UPDATE for client connection
const char* passphrase  = "admin1234";

// Constants     
const int WifiTimeout   = 30,       // Wait time for WIFI connect (seconds)
          CnfgBtn     = 12;         // Config pin depressed = AP enabled     
          
long  switchDuration  = 0;          // Duration the config switch is held                 

// ----- WIFI Client Connect -----
void wifiConnect() {
  int timeout = WifiTimeout;                          // Set WIFI connect retried

  Serial.println();
  Serial.println("WIFI Connect");
  Serial.println("------------");
  Serial.print("SSID: ");                            // Display SSID, Password, and Hostname
  Serial.println(ssid);
  Serial.print("Passphrase: "); 
  Serial.println(passphrase); 
  
  WiFi.mode(WIFI_STA);                                 
  WiFi.begin(ssid, passphrase);                       // Try connecting to WIFI network as client 
  
  Serial.print("Waiting for connection.");            // Wait for connection
  while (WiFi.status() != WL_CONNECTED and timeout > 0) {  
    delay(1000);             
    timeout--;                                        // Decrease timeout counter
    Serial.print(".");
  }

  if (WiFi.status() == WL_CONNECTED) {                // Connected, Success!
    Serial.println("Success!"); 
    Serial.print("IP address: ");                     // Print out IP address
    Serial.println(WiFi.localIP());
  } else {                                            // Didn't connect, Failure!
    Serial.println("Failed!"); 
  }

  Serial.println("");
  WiFi.printDiag(Serial);
}

// ----- AP Start up -----
void apStart() {   
  Serial.println();
  Serial.println("AP Start");
  Serial.println("--------");

  WiFi.mode(WIFI_AP); 
  WiFi.softAP(ssid);
          
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("IP address: ");
  Serial.println(myIP); 

  Serial.println("");
  WiFi.printDiag(Serial);
}

// ===== SETUP =====
void setup(void) {
  // Switch
  pinMode(CnfgBtn, INPUT_PULLUP);         // configuration pin as input with pull-up
  
  // Serial Debug
  Serial.begin(115200);                   // Serial debug      
  Serial.println("");

  // WIFI
  WiFi.persistent(false);                 // Turn off persistent to fix flash crashing issue.   
  WiFi.mode(WIFI_OFF);                    // https://github.com/esp8266/Arduino/issues/3100 
  wifiConnect();                          // Connect to AP as client   

}
// ========= MAIN LOOP =========
void loop(void) {
  // Configuartion Switch Polling  
  if(digitalRead(CnfgBtn) == LOW) {
    long now = millis();
    if (now - switchDuration > 3000) {
      switchDuration = now;
      apStart(); 
    }
  } else {
    switchDuration = millis();
  }

}   

```

### Debug Messages

```
None
```

devyte, 2017-09-05:
I am changing modes between station, softap, and stationap at runtime, without any reboot, and it works. However, I am not using persistent(false), nor am I setting mode to OFF at any point in the process.
Either there is an interaction with those settings, or there is a usage error. 
Please pursue this further.

devyte, 2018-02-28:
Closing due to lack of feedback.

closed 2018-02-28 by devyte

vijaybande, 2020-01-21:
> runtime

can I get a code