esp8266/Arduino #3371: Error for missing AVR libraries

KartavyaBhatt, 2017-06-23:
### Basic Infos
I have properly installed Arduino IDE and so I have all the AVR libraries under **/home/joker/arduino-1.8.2/hardware/tools/avr/avr/include** (My username is joker)

### Description

Error message : 
/home/joker/Arduino/libraries/Arduino-IRremote-master/IRremote.cpp:28:27: fatal error: avr/interrupt.h: No such file or directory
 #include <avr/interrupt.h>

### Settings in IDE

Module:  			SparkFun ESP8266 Thing
CPU Frequency:		80Mhz

### Sketch

```cpp
#include <IRremote.h>

void setup() {
}

void loop() {
}
```

Makuna, 2017-06-23:
The AVR directory should be your hint; including a platforms specific header (everything under AVR is specific to the AVR chipset) will not work.
What methods are you using from that header?  You maybe able to just wrap the include with a #ifdef ARDUINO_ARCH_AVR

devyte, 2017-09-05:
This looks like a user error. Usage of the ESP has nothing to do with AVR.
Please research how to install and use the core libs properly, e.g.: readthedocs, install via board manager or use git version.
Closing this.

closed 2017-09-05 by devyte