esp8266/Arduino #3374: Modem sleep not working

SimCordy, 2017-06-25:
### Basic Infos

#### Hardware
Hardware:			?ESP-8266EX?

### Description

I'm trying to run my wemos d1, in modem sleep mode, in order to reduce the power consumption, when I run my script I use the WiFi.forceSleepBegin(), and this instruction works because, the module disconnect from Wifi, in fact if I try to ping it, the module doesn't respond. The problem is that when I execute the WiFi.forceSleepWake(), the module don't wake up, and remain disconnected from wifi.
I suspect that the modem sleep is not set properly, how I can set it?

Module:  			?Wemos D1?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?

### Sketch

```
void setup() {
   WiFi.begin(ssid,pass);
  while(WiFi.status() != WL_CONNECTED){
    delay(500);
  }
}

void loop() {
WiFi.forceSleepBegin();
delay(10000);
WiFi.forceSleepWake()
}
```

Tech-TX, 2020-01-27:
The **WiFi.forceSleepWake()** can take a long time to reconnect.  I've seen it connect in _as little as_ 7 seconds, and anywhere up to 35 seconds.  It attempts reconnection with the previous WiFi settings.  The way your code is written, it's questionable whether it finishes the reconnect before that 10 second delay() times out, and it goes right back into Forced Modem Sleep.  You should change the loop after WiFi.forceSleepWake() to check WiFi.status() and wait 30 or 40 seconds maximum, and re-initialize WiFi if it fails that timeout.  That, or change your second delay from 10 seconds to 30 seconds for the quick test.

In reality, I'd just re-initialize WiFi from scratch, as doing a full WiFi.begin() is frequently quicker than reconnecting using WiFi.forceSleepWake().

devyte, 2020-02-02:
Closing per previous comment.

closed 2020-02-02 by devyte