esp8266/Arduino #3375: 2.4.0-rc1  ESP8266WebServer and webServer.setContentLength(CONTENT_LENGTH_UNKNOWN)

microentropie, 2017-06-25:
### Basic Infos
Web server has some problems: the returned html pages have something wrong.
Get "Connection timeout" under Android.

#### Hardware
Hardware:			ESP-07
Core Version:      	2.4.0-rc1

### Description
The example makes use of ESP8266HTTPUpdateServer and
webServer.setContentLength(CONTENT_LENGTH_UNKNOWN) ended by webServer.client().stop();
Compiling with Arduino 2.3.0 everything is ok.
Compiling with Arduino 2.4.0-rc1:
- Under Windows environment: 
  - all the pages are ok
  - the /update page shows in browser a \n at end
- Under Android environment: 
  - the pages are not completed: browser (Chrome and Mozilla) returns "Connection timeout"

Problem description
Change the .ino file by changing your wifi username/password and IP address, Flash the firmware with Arduino 2.4.0-rc1.
Connect to the ESP8266 web page (192.168.0.170/ and 192.168.0.170/info in my .ino) from Android or IOS smartphone:
you get a partial page or nothing and later a "Connection Timeout". Provided 3 different pages.
If you do the same actions from Windows, all the pages are displayed correctly without problems.

If you flash the same identical .ino with 2.3.0 version, the ESP web server works fine.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB (NO SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch
.ino included

### Debug Messages

[bug_240rc1.zip](https://github.com/esp8266/Arduino/files/1100172/bug_240rc1.zip)

flavianstef, 2017-06-28:
I also have the same problem... Anyone has any idea how to resolve it?

devyte, 2017-06-28:
What happens with Chrome under Windows?

flavianstef, 2017-06-28:
I forwarded the ESP ip/port on my router and with 2.4.0.0 rc1 I can't access the ESP over the internet. I tried with 2.3.0.0 and the forward works good. 

If I call locally the ESP I receive the response (2.4.0.0 rc1).

Over the internet, the call is received by ESP and the ESP send the response (code - server.send ( 200, "text/html", "<html> .......... </html>") but the client (Safari/Chrome/IE) didn't receive it. The status of the request is CONTENT_LENGTH_UNKNOWN or timeout.

This problem is available only in 2.4.0.0 rc1

bbx10, 2017-07-27:
@microentropie  Try adding `sendContent("");` at the end of the chunked transfer before calling stop(). A zero length chunk is required to terminate the chuck sesssion.

https://github.com/esp8266/Arduino/issues/2481

microentropie, 2017-07-30:
Your suggestion works 👍  !
Now I complete the html page send as you described: it works and web browser rendering is quicker ! Thanks a lot.
From my point of view the issue can be considered closed.
Maybe adding some hints in the _examples_ may be useful.
I changed my code as follows:
```
webServer.sendHeader("Cache-Control",` "no-cache, no-store, must-revalidate");
webServer.sendHeader("Pragma", "no-cache");
webServer.sendHeader("Expires", "-1");
webServer.setContentLength(CONTENT_LENGTH_UNKNOWN); // *** BEGIN ***
webServer.send(200, "text/html", "");
webServer.sendContent("<html><head>");
// ...
webServer.sendContent("</body></html>");
webServer.sendContent(""); // *** END 1/2 ***
webServer.client().stop(); // *** END 2/2 ***
```

closed 2017-07-30 by microentropie