esp8266/Arduino #3378: Exception (0) with GPIO interrupts (during WiFi.begin())

an-erd, 2017-06-26:
### Basic Infos

#### Hardware
Hardware:			Feather Huzzah esp8266
Core Version:      	2.4.0-rc1

### Description

Exception (0) when using WiFIMulti.run() and GPIO interrupts.
In my code, the ISR (put in RAM) essentially does a comparison of last time called (for debouncing) using millis() and reading with digitalRead() and some comparisons. If the code is close to WiFi.begin(), the problem can be reproduced pretty easy by turning the attached rotary encoder.

The same problem occurs with just using WiFi.begin(), but than is harder to reproduce.

```cpp
void ICACHE_RAM_ATTR ROTENC::isrInt0() {
	if (millis() - int0_time_ < THRESHOLD)
		return;
	int0_history_ = int0_signal_;
	int0_signal_ = digitalRead(ENCODER_PIN_A);
	if (int0_history_ == int0_signal_)
		return;
	int0_time_ = millis();
	if (int0_signal_ == int1_signal_) {
		rotary_half_steps_--;
	}
	else {
		rotary_half_steps_++;
	}

	changed_halfSteps_ = true;
}
```

### Debug Messages

```
Exception (0):
epc1=0x40214624 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1ec0 end: 3fff2450 offset: 01a0

>>>stack>>>
3fff2060:  40106556 40100823 00000000 3ffeed50  
3fff2070:  20017035 00000000 00000002 40100ec2  
3fff2080:  00000000 00000000 0000001f 00000022  
3fff2090:  3fffc200 4010651c 3fffc258 4000050c  
3fff20a0:  400043e6 00000030 00000016 ffffffff  
3fff20b0:  400044ab 3fffc718 3fff21a0 08000000  
3fff20c0:  60000200 08000000 08000000 00000000  
3fff20d0:  0000ffff 00042035 00002035 003fe000  
3fff20e0:  4024dfe0 00000494 003fe000 00000030  
3fff20f0:  00000000 3fff21b1 3fff2120 401049ed  
3fff2100:  4000050c 4010651c 3fffc258 4010664e  
3fff2110:  4021c0fc 3fff2210 3fff0c70 40106343  
3fff2120:  00000075 3fff122c 00000015 60000318  
3fff2130:  3ffe82f8 00000006 00000000 40216c47  
3fff2140:  3fff21b3 3ffe9591 80000000 40214636  
3fff2150:  401065c1 3ffe8778 60000314 401065a8  
3fff2160:  00000002 4000444e 3fff2210 4021c3a4  
3fff2170:  00000000 4000422e 60000200 3fff2220  
3fff2180:  00000000 4000444e 00000100 003ff000  
3fff2190:  00000002 401052a3 00000001 60000200  
3fff21a0:  00000002 4000410f 00001001 00000205  
3fff21b0:  3fffc718 40004a3c 000003fe 4024dfe0  
3fff21c0:  3fffc718 40105514 000003fe 401068dc  
3fff21d0:  000003fe 402410a0 000003ff 40241089  
3fff21e0:  3fff475c 40241173 3fff42c4 00000494  
3fff21f0:  000003fd 00000004 3fff42c4 40241156  
3fff2200:  ffffff01 55aa55aa 0000000f 00000020  
[... 62 lines cut ...]
0x40241089: wifi_param_save_protect_with_check at ?? line ?
0x40241173: system_param_save_with_protect at ?? line ?
0x40241156: system_param_save_with_protect at ?? line ?
0x402414a5: wifi_station_ap_number_set at ?? line ?
0x40241458: wifi_station_ap_number_set at ?? line ?
0x40207b6d: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at C:\Users\AKAEM\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 516
0x40207aa3: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at C:\Users\AKAEM\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 516
0x402101bc: TwoWire::requestFrom(unsigned char, unsigned int, bool) at C:\Users\AKAEM\Documents\Arduino\hardware\esp8266com\esp8266\libraries\Wire/Wire.cpp line 252
0x402100e9: TwoWire::available() at C:\Users\AKAEM\Documents\Arduino\hardware\esp8266com\esp8266\libraries\Wire/Wire.cpp line 252
0x4020f428: I2Cdev::readBytes(unsigned char, unsigned char, unsigned char, unsigned char*, unsigned short) at C:\Users\AKAEM\Documents\Arduino\libraries\I2Cdev/I2Cdev.cpp line 148
0x40240dc2: wifi_get_opmode at ?? line ?
0x40214b3a: Print::printf(char const*, ...) at C:\Users\AKAEM\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.cpp line 211
0x40207c08: ESP8266WiFiSTAClass::begin(char*, char*, int, unsigned char const*, bool) at C:\Users\AKAEM\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 516
0x40207423: ESP8266WiFiMulti::run(WifiAPProfile_t) at C:\Users\AKAEM\Documents\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/ESP8266WiFiMulti.cpp line 124
0x40210274: TwoWire::endTransmission() at C:\Users\AKAEM\Documents\Arduino\hardware\esp8266com\esp8266\libraries\Wire/Wire.cpp line 252
0x4021103a: SPBOX_COM::enableWlan() at C:\Users\AKAEM\AppData\Local\Temp\VMBuilds\spBox\esp8266com_huzzah\Release/spbox_com.cpp line 234
0x402110b1: SPBOX_COM::checkWlan() at C:\Users\AKAEM\AppData\Local\Temp\VMBuilds\spBox\esp8266com_huzzah\Release/spbox_com.cpp line 234
0x4020512a: loop at C:\Users\AKAEM\AppData\Local\Temp\VMBuilds\spBox\esp8266com_huzzah\Release/spBox.cpp line 1419
0x402142bc: loop_wrapper at C:\Users\AKAEM\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 57
0x4010070c: cont_norm at C:\Users\AKAEM\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 109
```

I'm lost in where to continue debugging my code and any hints are appreciated!

Thanks Andreas

igrr, 2017-06-26:
Can you see what is the output of `xtensa-lx106-elf-addr2line -pfia -e path/to/build/directory/your-sketch.elf 0x40214624`?

an-erd, 2017-06-26:
Since I'm not at home I needed to recompile on a different PC, and checked with a couple of addresses. They correspond with the stack trace above.
 
Result of the command above for the given address 0x40214624:

```
C:\Users\andrerdm\Source\Repos\spBox\spBox\spBox\Release>C:\Users\andrerdm\Docum
ents\Arduino\hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\bin\xtensa-lx106
-elf-addr2line.exe -pfia -e C:\Users\andrerdm\Source\Repos\spBox\spBox\spBox\Rel
ease\spBox.ino.elf 0x40214624
0x40214624: _ZN10MD5Builder8getCharsEPc at C:\Users\andrerdm\Documents\Arduino\h
ardware\esp8266com\esp8266\cores\esp8266/MD5Builder.cpp:84 (discriminator 2)
```

igrr, 2017-06-26:
Unless you are actually calling MD5Builder functions somewhere from an interrupt, this is likely caused by the fact that your ELF file compiled on another PC is slightly different from the one which was used to obtain the stack trace.

The ELF file needs to be exactly the same one as was used for programming the board. Alternatively, recompile and flash the new one. When exception happens, take `epc1` value reported in the exception output and feed it into addr2line.

an-erd, 2017-06-26:
Thanks for the excellent hint to use xtensa-lx106-elf-addr2line.exe . I was not aware of the tool. It would be great to add a link to this tool to the documentation, FAQ, etc. Finally it then took me 5 minutes to find the reason.

Actually the problem was caused by using the functional interrupt (pull request Functional Interrupts #2745). If it is of interest, I will spend some time with my setup to better understand the issue.

Thanks Andreas

closed 2017-06-30 by an-erd