esp8266/Arduino #3379: Getting ets Jan  8 2013,rst cause:2, boot mode:(3,6) while using wifi.config() method.

akashash, 2017-06-28:
Here is my code. Can anyone help please?

/*********
  Rui Santos
  Complete project details at http://randomnerdtutorials.com  
*********/

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

MDNSResponder mdns;

// Replace with your network credentials
const char* ssid = "Vertos";
const char* password ="aNonymous#1797";

ESP8266WebServer server(80);

String webPage = "";

int gpio0_pin = 0;
int gpio2_pin = 2;

IPAddress ip(192, 168, 1, 24); // where xx is the desired IP Address
IPAddress gateway(192, 168, 1, 1); // set gateway to match your network

IPAddress subnet(255, 255, 255, 0); // set subnet mask to match your network

void setup(void){

   webPage += "<h1>ESP8266 Web Server</h1><p>Socket #1 <a href=\"socket1On\"><button>ON</button></a>&nbsp;<a href=\"socket1Off\"><button>OFF</button></a></p>";
  webPage += "<p>Socket #2 <a href=\"socket2On\"><button>ON</button></a>&nbsp;<a href=\"socket2Off\"><button>OFF</button></a></p>";
  
  // preparing GPIOs
  pinMode(gpio0_pin, OUTPUT);
  digitalWrite(gpio0_pin, LOW);
  pinMode(gpio2_pin, OUTPUT);
  digitalWrite(gpio2_pin, LOW);
  
  delay(1000);

  Serial.begin(115200);
  Serial.println("Setting static ip to : ");
  Serial.println(ip);
  Serial.println("Connecting to Vertos...");
  WiFi.config(ip, gateway, subnet);
  delay(5000);
  //checking for SHIELD
  if (WiFi.status() != WL_NO_SHIELD) {
    Serial.println("WiFi shield is present"); 
    while(true);  // don't continue
  }

  WiFi.begin(ssid, password);
  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  
  if (mdns.begin("esp8266", WiFi.localIP())) {
    Serial.println("MDNS responder started");
  }
  
  
  server.on("/", [](){
    server.send(200, "text/html", webPage);
  });
  server.on("/socket1On", [](){
    server.send(200, "text/html", webPage);
    digitalWrite(gpio0_pin, HIGH);
    delay(1000);
  });
  server.on("/socket1Off", [](){
    server.send(200, "text/html", webPage);
    digitalWrite(gpio0_pin, LOW);
    delay(1000); 
  });
  server.on("/socket2On", [](){
    server.send(200, "text/html", webPage);
    digitalWrite(gpio2_pin, HIGH);
    delay(1000);
  });
  server.on("/socket2Off", [](){
    server.send(200, "text/html", webPage);
    digitalWrite(gpio2_pin, LOW);
    delay(1000); 
  });
  server.begin();
  Serial.println("HTTP server started");
}
 
void loop(void){
  server.handleClient();
}

lrmoreno007, 2017-06-29:
You must set **"WiFi.mode(WIFI_STA);"** after WiFi.config and before WiFi.begin
```
WiFi.config(ip, gateway, subnet);
delay(100);
//checking for SHIELD
if (WiFi.status() != WL_NO_SHIELD) {
Serial.println("WiFi shield is present");
while(true); // don't continue
}
WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
}
```

lrmoreno007, 2017-06-30:
Please close it, if solved.

closed 2018-01-02 by igrr