esp8266/Arduino #3382: Exception(0)/(9)/(28) during HTTP request

sdarquea, 2017-06-29:
### Basic Infos

OS: Windows 8.1
IDE: Arduino 1.8.3

#### Hardware
Hardware:			ESP-12
Core Version:      	2.3.0

### Description

Hello, I have a doozie of a bug that I've been trying track down for a few weeks. The project is a timer. There's a total of 6 ESPs, all communicating by UDP. Via a web interface hosted by one of the modules, I arm the timer and as each sensor is passed in a specified order, it sends a UDP packet to the "master" ESP, which logs the time. After the last sensor is passed, which is the master and the code I'm including, it triggers an interrupt which sets a flag. When this flag is true, "loop()" then calls a few functions to process the times held in an array and build the string to send back to the browser to be displayed. When I run the timers everything works fine. It processes the data just fine and then sits there. However, when I come along with any HTTP request, the system crashes. I've narrowed down that the issue (or one of them) lies in when I build the string in "publishResults" or "publishSpeed," as when I comment these out it doesn't crash. The weird thing is that this all works at first, and then becomes intermittent, and then eventually doesn't work at all. Also, sometimes it will crash after about a second of running the timer even if I comment out these parts and simply let the timer run. My inclination is that when I build these char arrays, it corrupts something that has to do with the WebServer/WiFi, so that when I send a request from my browser, it asks the ESP to go look for something that's now garbage. But I have been unable to pinpoint this. The exceptions are always either 0, 9, or 28. Lastly, I am aware that there should be a ICACHE_RAM_ATTR in front of my interrupts, however when I add this #define to my ISRs, it thinks that it's just part of the ISR name. Any thoughts? Please let me know if you need any more info. Thanks for your time. Cheers!
-Shep  

### Settings in IDE

Module:  			WeMos D1(retired)
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Flash Frequency:	921600baud
Upload Using:		SERIAL

### Sketch
`#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <Ticker.h>
#include <DHT.h>
#include <Adafruit_Sensor.h>
extern "C" {
#include "user_interface.h"
}

#define WiFiConnectedPin    BUILTIN_LED
#define interruptPin        14
#define timerArmedPin       12
#define buzzerPin           4
#define DHTPIN              13
#define DHTTYPE             DHT22
#define armTimer            "91"
#define ACK                 "93"

ESP8266WebServer server(80);
WiFiUDP Udp;      
DHT dht(DHTPIN, DHTTYPE, 11);   
Ticker secondTick;                               

char* ssid = "Battle Born Sports Tech";                     
char* password = "password";

IPAddress V1IP(192, 168, 0, 1);
IPAddress V2IP(192, 168, 0, 2);
IPAddress V3IP(192, 168, 0, 3);
IPAddress V4IP(192, 168, 0, 4);
IPAddress VBIP(192, 168, 0, 5);

IPAddress L1IP(192, 168, 0, 10);
IPAddress L2IP(192, 168, 0, 11);
IPAddress L3IP(192, 168, 0, 12);
IPAddress L4IP(192, 168, 0, 13);

IPAddress mCastIP(192, 168, 0, 255);

unsigned int localUdpPort = 4210;
volatile unsigned int grabTime = 0;
volatile unsigned int totalTime = 0;
unsigned int timerVal[255];

float humidity, temp_f;

[... 601 lines cut ...]
Sometimes this one:

Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 21 results
0x40206df3: String::move(String&) at C:\Users\Shepherd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x40206d9e: String::String(char const*) at C:\Users\Shepherd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x40206c70: String::~String() at C:\Users\Shepherd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x40205c38: ESP8266WebServer::_parseRequest(WiFiClient&) at C:\Users\Shepherd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/Parsing.cpp line 72 (discriminator 2)
0x40104e0a: lmacTxFrame at ?? line ?
0x40101776: ppEnqueueRxq at ?? line ?
0x40104a0f: lmacMSDUAged at ?? line ?
0x402036dc: WiFiClient::~WiFiClient() at C:\Users\Shepherd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x40104ebf: lmacRxDone at ?? line ?
0x40203bd8: WiFiServer::available(unsigned char*) at C:\Users\Shepherd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiServer.cpp line 105
0x40204e8b: ESP8266WebServer::handleClient() at C:\Users\Shepherd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x40207460: __yield at C:\Users\Shepherd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40202b24: loop at C:\Users\Shepherd\Desktop\Documents\Arduino\Speed Cones\V1-2/V1-2.ino line 137
0x40102c5c: wDev_ProcessFiq at ?? line ?
0x40202ca9: loop at C:\Users\Shepherd\Desktop\Documents\Arduino\Speed Cones\V1-2/V1-2.ino line 188
0x40207434: loop_wrapper at C:\Users\Shepherd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x401006d9: cont_continue at C:\Users\Shepherd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 51
0x40207409: esp_schedule at C:\Users\Shepherd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40207434: loop_wrapper at C:\Users\Shepherd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\Shepherd\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109
`

lrmoreno007, 2017-06-30:
Please, you can try to insert all the code in a frame with "Insert code" so that it is readable and can be used to try to help you ..

sdarquea, 2017-06-30:
Hi Irmoreno007, 
I hope that looks better. I pulled out the yields(). I had put those in there to see if maybe all the strcats weren't giving the background WiFi operations enough time to do their stuff.  Is that something I need to consider or is one yield at the end of loop() enough? Thanks again!

lrmoreno007, 2017-07-03:
You have many yield(). Try to remove them all or leave only those that are within big "for" or "while".

On the other hand, I see the following:
- You've created your own watchdog ... Why?
- It is good practice to declare the functions before setup () and loop ().
- The variable armTimer defines it with #define and then converts it to '* char' in "handShake" function. Should be:
Boolean handShake (const char * _opCode, int _dest)
- You should try to locate the problem a little more, eliminating possibilities, because the sketch is very big and with many things.

sdarquea, 2017-07-10:
I put the WD in there because I watched some video telling me to do it for reliability. But now that I've thought about it, it doesn't make sense and have taken it out. Thanks for all your input, I have made the changes you recommend.

devyte, 2017-09-05:
To me it looks like you have a massive memory leak. In the functions you mention, you are using calloc to allocate a variable-length array of variable-length strings, but then you are not free()ing them anywhere. Because you are calling publish again and again, you are allocating again and again, and of course you will eventually run out of heap.
Rather than using C-style strings with fragile mem handling, I suggest using a std::vector<String> or std::vector<std::string> to hold your strings. Be aware that std::vector currently has a compiler bug, related to the #defines min and max in Arudino.h. Until it is fixed, you should:
#ifdef min
#undef min
#endif
#ifdef max
#undef max
#endif
#include <vector>

This looks like a user error to me, and not an issue in the core. In any case, this is not the correct place to discuss what is wrong with your particular project. If you need further discussion, please ask for assistance at esp8266.com or stackoverflow.
Closing for now.

closed 2017-09-05 by devyte