esp8266/Arduino #3383: NodeMcu 8266 Clone not detecting on PC

TrapFather, 2017-06-30:
I have being trying to plug in my NodeMcu 8266 Clone from China but it's not being detected by my PC
I have tried numerous different cords (including data transfer cables) and installed all the CH340G Drivers, ESP8266 Drivers and had no luck!

If anyone has a solution please help me.
I appreciate your help.

Regards,
TrapFather

davisonja, 2017-06-30:
What OS?
Is it showing up at all on the PC?
Any lights on the module to indicate it's being powered?

TrapFather, 2017-06-30:
@davisonja 
Windows 10 / Just did a fresh install
No Indications that It is being detected on the PC, I plug it in and the computer doesn't react or anything
The module isn't showing any power from a LED Light but it is getting very hot. 

Thank you

lrmoreno007, 2017-06-30:
This is not the best place to help you with your spoiled Chinese NodeMCU.

This site is for bugs in the ESP8266 / Arduino Framework code.

The best site is www.esp8266.com

WereCatf, 2017-07-01:
@TrapFather If it gets very hot, there's most likely a short on the board. The ESP8266 should barely get lukewarm to the hand in normal operation.

You should check the 3.3V-voltage; if it's higher than 3.3V, your ESP-module itself is almost definitely burned and there's no rescuing your board. If it's 3.3V or less, you need to find out if there's a solder-bridge somewhere or if some of the support-components are broken.

devyte, 2017-09-05:
@TrapFather  Please refer to esp8266.com or stackoverflow for assistance. @lrmoreno007 is correct, this is not the place to ask for help, as this is an issue tracker for the core libs.
Closing.

closed 2017-09-05 by devyte