esp8266/Arduino #3384: while(1) at end of loop() causes a compilation error

tofrnr, 2017-06-30:
running the following code on an ESP8266 nodeMCU, the  while(1) at end of loop causes a runtime error, reset (on Uno ok though)

as meanwhile, noticed, this issue is a "bug by ESP8266 IDE design", nonetheless it should be left to the programmer if he wants to stop the loop by a final while(1) or not.
As usual as always for C:
**_The programmer decides, what he want to program, what ever is possible in C/C++, not the IDE decides it._**

```
Soft WDT reset

ctx: cont 
sp: 3ffef2a0 end: 3ffef4a0 offset: 01b0

>>>stack>>>
3ffef450:  3ffe8438 00000001 3ffee448 402020a1  
3ffef460:  0000007f feefeffe 3ffee448 40202504  
3ffef470:  3fffdad0 00000000 3ffee448 3ffee46c  
3ffef480:  3fffdad0 00000000 3ffee464 402027b4  
3ffef490:  feefeffe feefeffe 3ffee480 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)
```

Uno: no error, ok;
instead delay(5000) for ESP8266: also no error, ok.

 

```
    // --------------------------------------
    // i2c_scanner
    //    
    // This sketch tests the standard 7-bit addresses
    // Devices with higher bit address might not be seen properly.
    //
     
    #include <Wire.h>

 
    void setup()
    {
      Wire.begin(D4,D5);  // ESP8266
      //Wire.begin();       // Uno
 
      Serial.begin(115200);
      while (!Serial);             // Leonardo: wait for serial monitor
      Serial.println("\nI2C Scanner");
    }
     
     
    void loop()
    {
      byte error, address;
      int nDevices;
     
      Serial.println("Scanning...");
     
      nDevices = 0;
      for(address = 0; address < 127; address++ )
      {

        if (address%16 == 0)  {   
          Serial.println();  
          Serial.print( (address+1)/16);
          Serial.print("  ");
        }

        
        if(address==0) {
           Serial.print("** ");
           continue;
        }
        
        Wire.beginTransmission(address);
        error = Wire.endTransmission();
     
        if (error == 0)
        {          
          if (address<16) Serial.print("0"); 
          Serial.print(address,HEX); Serial.print(" ");         
          nDevices++;
        }
        else if (error==4)
        {
          Serial.print("?? ");   
        }    
        else
        {      
           Serial.print("-- ");   
        }          
      }
      
      Serial.println();
      Serial.print("found: "); Serial.print(nDevices); Serial.print(" devices \n");
     
      //delay(5000);  /// <<< fine on all boards
      while(1);    // <<<<<<<<<< runtime error: reset on ESP8266 // Uno: ok
    }

```

igrr, 2017-06-30:
Aside from running this on top of a preemptive scheduler (i.e. ESP8266-RTOS-SDK instead of the non-OS SDK), there is not much that can be done to "fix" this. Switching to RTOS would reduce the amount of available memory, which isn't plenty. Having to `yield` inside busy loops is not nice either, it's not compatible with code written for Uno and other boards, but it's the best compromise we have.

devyte, 2017-09-05:
This is a usage error due to lack of understanding of the ESP's use semantics. You can't do a while(1) in the loop, because there are wdt in place to avoid precisely such things. This is because, the wifi stack must be serviced every so often.
Please research usage for the ESP, and look at examples. If you need further assistance, please ask at esp8266.com or stackoverflow, as this is not the right place for that.
Closing.

closed 2017-09-05 by devyte

tofrnr, 2017-09-07:
Arduino uses 
setup()
and
loop()
instead of 
int main(),

and for the Arduino API function loop() there is no restrictions to perpetual loops, so there also must not be either ones for esp8266.
WiFi functionality has to be provided in the background, obscured to the end user, e.g. by interrupts or multithreading.

tofrnr, 2017-09-07:
more RAM would be no issue, because the nodeMCU has 80kB, whilst a M0 Zero has just about the half and an Uno or Nano has just 1/30 of that.