esp8266/Arduino #3385: improvement request: please enable sprintf()  for float/double, like it's possible in Arduino ARM core

tofrnr, 2017-06-30:
improvement request: 
please enable sprintf()  for float/double, like it's possible in Arduino ARM core
```
char* str[100];
// not possible for AVR and nodeMCU but possible for Arduino ARM core
sprintf(s, "floats: %4.2f %+.0e %E \n", 3.1416, 3.1416, 3.1416); 
```

igrr, 2017-06-30:
See https://github.com/esp8266/Arduino/issues/1179, this feature is available in 2.4.0-rc1.

closed 2017-06-30 by igrr

tofrnr, 2017-06-30:
not possible in Arduino IDE 1.8.2 feat. esp8266 board manager
when is it automatically available via Arduino IDE?

tofrnr, 2018-01-03:
this program still doesn't work as expected - when will it be fixed?

```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  char sbuf[100];
  static int n;
  
  sprintf(sbuf, "%3d %10.8f", n++, PI);
  Serial.println(sbuf);
  delay(1000);
}
```