esp8266/Arduino #3386: "$T0b#e6" during HTTP request

mchetan133, 2017-06-30:
A client ESP8266 continuously making a HTTP request to a server ESP8266. Server responding for some duration after it turning of hotspot and poping "$T0b#e6" this on serial monitor. 

Please Help me for solving this.........

devyte, 2017-09-05:
This description makes no sense. Please gather your thoughts, do some research and look at examples, and if necessary, open a new issue while making use of the issue template.
Closing due to lack of info.

closed 2017-09-05 by devyte

d-a-v, 2017-09-05:
$T0b#e6 is the message sent on serial line by corelibs for gdb debugging.
Hence <gdbstub.h> is included in your server's sources and gdb is needed to further debug.
check [doc](https://github.com/esp8266/Arduino/tree/master/libraries/GDBStub).