esp8266/Arduino #3391: how to install/include/assign compatible SD+SPI lib?

shiftleftplusone, 2017-07-01:
hello,
as I observed, there is an ESP8266-specific SD lib
https://github.com/esp8266/Arduino/tree/master/libraries/SD
it has the same lib name as the prefetched built-in Arduino lib 
C:\Program Files (x86)\Arduino\libraries\SD

same about SPI:
https://github.com/esp8266/Arduino/tree/master/libraries/SPI
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\

how to install/include/assign ESP8266-compatible SPI+SD lib without deleting or overwriting the original Arduino and to avoid ambiguous #include <...>  Libs?

WereCatf, 2017-07-01:
As long as you have the correct board selected it'll use the appropriate header-file. If you e.g. do `#include<SPI.h>` it'll use the ESP8266-specific one as long as you have an ESP8266-board selected, and it'll use the AVR-specific one if you've got an AVR-based Arduino-board selected. There is no need for deleting or overwriting anything.

shiftleftplusone, 2017-07-02:
ok, thank you, that this lib selection is performed by the IDE automatically behind the curtons  without an explicite lib to be installed in the sketch folder was not clear to me.

closed 2017-07-02 by shiftleftplusone