esp8266/Arduino #3392: Crashes doing a GET request on port 443 with a sha1 fingerprint

Jeroen88, 2017-07-01:
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 35 results
0x40218242: memset at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S line 133
0x402259a6: more_comps at crypto/bigint.c line 672
0x40225b38: alloc at crypto/bigint.c line 672
0x40225db9: regular_multiply at crypto/bigint.c line 672
0x4021031c: esp_yield at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/core_esp8266_main.cpp line 57
0x40225b38: alloc at crypto/bigint.c line 672
0x4022686f: bi_barrett at crypto/bigint.c line 1289
0x40226ac4: bi_mod_power at crypto/bigint.c line 1414
0x40225a08: trim at crypto/bigint.c line 672
0x40227bf5: RSA_public at crypto/rsa.c line 252
:  (inlined by) RSA_encrypt at crypto/rsa.c line 286
0x4022366c: send_client_key_xchg at ssl/tls1_clnt.c line 409
0x40223b01: do_clnt_handshake at ssl/tls1_clnt.c line 123
0x402075e0: ClientContext::_consume(unsigned int) at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 609
:  (inlined by) ClientContext::read(char*, unsigned int) at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/libraries/ESP8266WiFi/src/include/ClientContext.h line 253
0x40210394: __yield at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/core_esp8266_main.cpp line 57
0x40223474: do_handshake at ssl/tls1.c line 2005
:  (inlined by) basic_read at ssl/tls1.c line 1481
0x40223850: do_client_connect at ssl/tls1_clnt.c line 168
0x401004d8: malloc at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/umm_malloc/umm_malloc.c line 1664
0x402235dc: ssl_read at ssl/tls1.c line 2005
0x40207aa1: SSLContext::connect(ClientContext*, char const*, unsigned int) at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 609
:  (inlined by) WiFiClientSecure::_connectSSL(char const*) at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 321
0x40207036: WiFiClient::connect(IPAddress, unsigned short) at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/libraries/ESP8266WiFi/src/WiFiClient.cpp line 324
0x40207e29: WiFiClientSecure::connect(char const*, unsigned short) at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp line 313
0x4020e684: HTTPClient::connect() at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 235
0x4020f791: String::copy(char const*, unsigned int) at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/WString.cpp line 720
0x4020eb64: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 235
0x4020f12d: Print::write(char const*) at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/Print.cpp line 87
0x4020f12d: Print::write(char const*) at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/Print.cpp line 87
0x4020ec12: HTTPClient::GET() at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp line 235
0x40204b3d: postTelegramData(DLMSDevice&) at /run/user/1000/gvfs/smb-share:server=touch_p5-share,share=volume-a1/Arduino sketches/Arduino/IoT_Embedded_Software_004/IoT_Embedded_Software_004.ino line 972
0x4020f158: Print::println() at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/Print.cpp line 87
0x40205540: loop at /run/user/1000/gvfs/smb-share:server=touch_p5-share,share=volume-a1/Arduino sketches/Arduino/IoT_Embedded_Software_004/IoT_Embedded_Software_004.ino line 1274
0x4020398c: handleSetID() at /run/user/1000/gvfs/smb-share:server=touch_p5-share,share=volume-a1/Arduino sketches/Arduino/IoT_Embedded_Software_004/IoT_Embedded_Software_004.ino line 972
0x40210620: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /home/jeroen/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 1934
0x4021060c: std::_Function_handler ::_M_invoke(std::_Any_data const&) at /home/jeroen/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2069
0x40210368: loop_wrapper at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/core_esp8266_main.cpp line 57
0x4010070c: cont_norm at /home/jeroen/.arduino15/packages/esp8266/hardware/esp8266/2.4.0-rc1/cores/esp8266/cont.S line 109

### Basic Infos

#### Hardware
Hardware:			NodeMCU (ESP-12E module)
Core Version:      	2.4.0-rc1

### Description

Crash after the second time a connection is created using port 443 and a finger print

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:			?qio?
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch
void postTelegramData(DLMSDevice &DSMRMeter) {
  HTTPClient httpClient;
  if(configurationParameters.postServer != "") {
    Serial.println("BEGIN posting telegram data...");
    httpClient.setTimeout(1000);

    Serial.print("Server: "); Serial.println(configurationParameters.postServer);
    Serial.print("Port: "); Serial.println(configurationParameters.postPort);

    String uriToPost;
    if(configurationParameters.postUri != "") {
      uriToPost = replaceVariables(configurationParameters.postUri, DSMRMeter);
    } else {
      uriToPost = "/";
    }
    Serial.print("Uri: "); Serial.println(uriToPost);

    if(configurationParameters.postFingerprint == "") 
      httpClient.begin(configurationParameters.postServer, configurationParameters.postPort, uriToPost);
    else {
      Serial.print("Fingerprint = '") ;Serial.print(configurationParameters.postFingerprint); Serial.println("'");
      httpClient.begin(configurationParameters.postServer, configurationParameters.postPort, uriToPost, configurationParameters.postFingerprint);
    }
    
    httpClient.setUserAgent("Metriot");
    httpClient.setReuse(false);

    t_http_codes httpCode;
    if(configurationParameters.postMethod == "POST") {
      Serial.println("Method = POST");
      httpClient.addHeader("Content-Type", "application/x-www-form-urlencoded");
      String payload = replaceVariables(configurationParameters.postBody, DSMRMeter);
      Serial.print("payload = '"); Serial.print(payload); Serial.println("'");
      httpCode = (t_http_codes) httpClient.POST(payload);
    } else if(configurationParameters.postMethod == "GET") {
      Serial.println("Method = GET");
      httpCode = (t_http_codes) httpClient.GET();
    } else {
      Serial.print("Method '"); Serial.print(configurationParameters.postMethod); Serial.println("' not supported");
    }

    Serial.print("HTTP Client returned: '"); Serial.print(httpCodeToString(httpCode)); Serial.println("'");
    
  } else Serial.println("NO SERVER CONFIGURED");
  Serial.println("... END posting telegram data");
}

### Debug Messages

Now = 1498930713
BEGIN posting telegram data...
Server: api.thingspeak.com
Port: 443
Uri: /update?api_key=6JIYT3U9YW6BBC1J&field1=000013.178&field2=000000.000&field3=00.000&field5=000&field6=00.000&field7=00000.271&field8=-67 
Fingerprint = '78:60:18:44:81:35:BF:DF:77:84:D4:0A:22:0D:9B:4E:6C:DC:57:2C'
Method = GET

Exception (29):
epc1=0x40218242 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

Defozo, 2017-07-07:
I'm having the same problem:
https://github.com/esp8266/Arduino/issues/3402

marcelowajnsztok, 2017-07-09:
I'm having the same problem here after updating.

Jeroen88, 2017-07-14:
This issue is mentioned in [https://github.com/esp8266/Arduino/issues/2075](url), the problem is that the ESP is running out of memory. I tidied up my big local class instances (create them with new just before I need them and destroy them with delete right after using them). I was also using a client and a server at the same time. I tested to connect using wificlientsecure without EPS8266Webserver, and the crash dissapeared. So @Defozo and @marcelowajnsztok, check your memory usage!
Closing the issue for now.

closed 2017-07-14 by Jeroen88