esp8266/Arduino #3396: setDebugOutput(false) doesn't stop messages in version 2.3.0

tokitherobot, 2017-07-03:
### Basic Infos

#### Hardware
Hardware:		ESP-01
Core Version:      	2.3.0

### Description

After calling "Serial.setDebugOutput(false)", the module still sends messages to the serial port after my PC connects to the access point.  I'm receiving the following messages:

```
add 1
aid 1
station: c0:56:27:e0:47:e3 join, AID = 1
```   

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB (512K SPIFFS)
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

### Debug Messages

```
add 1
aid 1
station: c0:56:27:e0:47:e3 join, AID = 1
```

lrmoreno007, 2017-07-03:
Follow this instructions and select "None"

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html#informations

tokitherobot, 2017-07-03:
Thank-you!  I've googled around and have had trouble finding good documentation on this project.

closed 2017-07-03 by tokitherobot