esp8266/Arduino #3402: Exception 29 when using SSL

Defozo, 2017-07-04:
### Basic Infos

#### Hardware
Hardware:			ESP-07S
Core Version:      	2.4.0-rc1

### Description

Sometimes (really rare) I get exception 29 when I'm trying to send something over HTTPS.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB (1MB SPIFFS)
CPU Frequency:		160Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		OTA / SERIAL
Reset Method:		ck

### Debug Messages

```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 95 results
0x4021d8c6: memmove at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/string/../../../.././newlib/libc/string/memmove.c line 74
0x4022ab11: regular_square at crypto/bigint.c line 969
:  (inlined by) bi_square at crypto/bigint.c line 1014
0x4022a146: AES_cbc_decrypt at crypto/aes.c line 345
0x4022a2d8: exp_bit_is_one at crypto/bigint.c line 672
0x4022a559: bi_int_multiply at crypto/bigint.c line 672
0x40214bb4: loop_task at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_main.cpp line 131
0x4022a2d8: exp_bit_is_one at crypto/bigint.c line 672
0x4022b00f: bi_set_mod at ?? line ?
0x4022b264: bi_mod_power at crypto/bigint.c line 1387
0x4022a1a8: more_comps at crypto/bigint.c line 672
0x4022c395: RSA_encrypt at crypto/rsa.c line 285
0x40227e0c: ssl_read at ssl/tls1.c line 2005
0x402282a1: send_cert_verify at ssl/tls1_clnt.c line 500
:  (inlined by) do_clnt_handshake at ssl/tls1_clnt.c line 118
0x4020af40: ClientContext::read(char*, unsigned int) at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 609
0x40214c2c: esp_yield at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_main.cpp line 87
0x40227c14: do_handshake at ssl/tls1.c line 2005
:  (inlined by) basic_read at ssl/tls1.c line 1481
0x40227ff0: send_client_hello at ssl/tls1_clnt.c line 273
:  (inlined by) do_client_connect at ssl/tls1_clnt.c line 167
0x401004d8: malloc at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x40227d7c: basic_read at ssl/tls1.c line 2005
0x4020b401: SSLContext::connect(ClientContext*, char const*, unsigned int) at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 609
:  (inlined by) WiFiClientSecure::_connectSSL(char const*) at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 321
0x4020a942: ClientContext at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClient.cpp line 324
:  (inlined by) WiFiClient::connect(IPAddress, unsigned short) at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClient.cpp line 132
0x4020b789: WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 304
0x4020cb4c: std::unique_ptr   >::get() const at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 196
:  (inlined by) std::unique_ptr   >::operator bool() const at c:\users\defozo\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits/unique_ptr.h line 245
:  (inlined by) HTTPClient::connect() at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 820
0x4010053d: _umm_realloc at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\umm_malloc/umm_malloc.c line 1491
:  (inlined by) realloc at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x4020d02c: std::unique_ptr   >::get() const at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 196
:  (inlined by) std::unique_ptr   >::operator->() const at c:\users\defozo\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits/unique_ptr.h line 229
:  (inlined by) HTTPClient::returnError(int) at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 1114
0x40213f2b: String::~String() at ?? line ?
0x40213f77: String::changeBuffer(unsigned int) at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/WString.cpp line 519
0x4020d0fc: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at ?? line ?
0x4020d11a: HTTPClient::GET() at ?? line ?
0x40207aca: sendMeasurementsToServer(String) at C:\Users\Defozo\AppData\Local\Temp\arduino_modified_sketch_857861/laser_na_stacje_v2.ino line 426
0x402140d8: String::move(String&) at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/WString.cpp line 519
0x402087f5: loop at C:\Users\Defozo\AppData\Local\Temp\arduino_modified_sketch_857861/laser_na_stacje_v2.ino line 398
0x40240000: ieee80211_send_action_vendor_spec at ?? line ?
0x40240000: ieee80211_send_action_vendor_spec at ?? line ?
[... 40 lines cut ...]
0x40250000: wps_registrar_probe_req_rx at ?? line ?
0x40240000: ieee80211_send_action_vendor_spec at ?? line ?
0x40210000: ESP8266WebServer::handleClient() at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 56
0x40220000: dns_enqueue at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/dns.c line 925
:  (inlined by) dns_gethostbyname at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/dns.c line 985
0x40220000: dns_enqueue at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/dns.c line 925
:  (inlined by) dns_gethostbyname at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/dns.c line 985
0x40220000: dns_enqueue at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/dns.c line 925
:  (inlined by) dns_gethostbyname at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/dns.c line 985
0x40210000: ESP8266WebServer::handleClient() at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 56
0x40100ec2: pp_post at ?? line ?
0x40240000: ieee80211_send_action_vendor_spec at ?? line ?
0x40210000: ESP8266WebServer::handleClient() at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 56
0x40240000: ieee80211_send_action_vendor_spec at ?? line ?
0x40240000: ieee80211_send_action_vendor_spec at ?? line ?
0x40240000: ieee80211_send_action_vendor_spec at ?? line ?
0x40240000: ieee80211_send_action_vendor_spec at ?? line ?
0x40103109: lmacProcessTXStartData at ?? line ?
0x40238000: wifi_softap_start at ?? line ?
0x40210000: ESP8266WebServer::handleClient() at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 56
0x40245555: rijndaelEncrypt at ?? line ?
0x40214c00: preloop_update_frequency() at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_main.cpp line 73
0x4010070c: cont_norm at C:\Users\Defozo\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/cont.S line 109

```

marcelowajnsztok, 2017-07-10:
I changed the Method for POST and solved my problem.

igrr, 2017-12-28:
The underlying issue is tracked in https://github.com/igrr/axtls-8266/issues/32, but recent axTLS versions have lower memory footprint, so you may be able to work with 26kB of available heap before the start of TLS connection.

devyte, 2018-07-12:
Currently bearssl is integrated into master. The axtls-based sockets are planned for deprecation and then retirement.
Closing this.

closed 2018-07-12 by devyte