esp8266/Arduino #3404: Fatal exception 0(IllegalInstructionCause): epc1=0x40106cf6, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

aatronics, 2017-07-06:
when i add "#include <ESP8266mDNS.h>" this file with "  if (!MDNS.begin("lighting")) {
    Serial.println("Error setting up MDNS responder!");
    while(1) { 
      delay(1000);
    }"
  MDNS.addService("http", "tcp", 80); "

I got the 
"Fatal exception 0(IllegalInstructionCause): epc1=0x40106cf6, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000 ................"
This message in serial monitor..

Plzzzzzzzzzz help

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

lrmoreno007, 2017-07-06:
You are making a infinite loop with
```
while(1) {}
```
And ESP8266 protects itself firing the WDT.

aatronics, 2017-07-07:
Thanks a lot, I have used same example given in Arduino IDE

devyte, 2017-09-05:
Please post your full minimal sketch that reproduces the problem. Otherwise this will be closed due to lack of info.
I am using MDNS myself, and it works for me.

aatronics, 2017-09-11:
Ok I will send code

devyte, 2017-09-11:
@aatronics also, please fill out the fields in the issue template. Providing that info is necessary.
And please try a fresh install (delete the old one, install from scratch), and try with latest git.

closed 2018-01-02 by igrr