esp8266/Arduino #3407: Bug with client.print(F("USER ")); and F memory macro in FTP routines

WheresWally, 2017-07-08:
It seems that the F("") memory saver is causing an issue.
I ported my working FTP routine from code running on a MEGA to ESP8266 12E NodeMCU.
It is the standard FTP upload routines that are often used on the Arduinos.
The code segment below  DOESNT work. 
The USER name is passed to the FTP server but the PASS command are not sent. Then it times out.

However if I remove the F("") then it does work as expected.
```
  client.print(F("USER "));
  client.println(userName);
  if(!eRcv()) return 0;
  
  client.print(F("PASS "));
  client.println(password);
  if(!eRcv()) return 0;
```

ikbelkirasan, 2017-07-08:
Are you sure `eRcv()` doesn't return 0 ?

WheresWally, 2017-07-09:
The aRec() times out as no data is returned as nothing seems to be sent.
The FTP server times out as no PASS sentence is received.
It all works after removing the F("") macro.

Makuna, 2017-07-09:
something to try...
store the string helper and debug print it also, like...
```
const __FlashStringHelper* pstrUser = F(""USER ");
Serial.print(pstrUser);
client.print(pstrUser);
```

igrr, 2017-12-28:
I'm not able to reproduce this with the latest git version.

https://gist.github.com/igrr/34a5421bc829ecc2e1fe939ce351f420 includes my test case (a slightly modified telnet server example) and the output.

@WheresWally please feel free to reopen if you can reproduce this with the latest git, and are able to provide the complete code to reproduce (which would not depend on FTP).

closed 2017-12-28 by igrr