esp8266/Arduino #3408: ESP.deepSleep(xxxx, WAKE_NO_RFCAL) not working since 2.0.0

wb8wka, 2017-07-09:
### Summary

RF Calibration cannot be turned off after 2.0.0 in wake from deepSleep  causing current spike and longer boot

#### Hardware
Hardware:			Wemos D1 mini and ESP-12S, 470uf cap between Vcc and Ground
Core Version:      	2.0.0  (works -  No RF Calibration current spike), 2.3.0  (Not working, current spike)
Test equipment: Rigol Scope, 1ohm current shunt, "fuzz" is because I didn't do calibration on scope but traces are accurate. 

### Description

Using identical code function ESP.deepSleep(6000000, WAKE_NO_RFCAL);  in 2.0.0 ESP can wakeup without large current spike from RF_CAL, however after 2.0.0 RF_CAL executes causing large current spike and longer run time.   Please see attached current traces. Correct data is sent in both instances, only different is current spike, which appears to be RF_CAL executing. 

Sketch sends simple beacon using wifi_send_pkt_freedom. GPIO16 connected to reset.  

WAKE_RF_DISABLED does behave as expected in both 2.0.0 and 2.3.0

Also 

    system_deep_sleep_set_option(2);  // Option 2 is WAKE_NO_RFCAL
    //system_deep_sleep(6000000);
    system_deep_sleep_instant(6000000);  //  Trims 98ms off go to sleep

has same characteristics in that it cannot turn off RF_CAL 

### Settings in IDE

Module:  			WeMos D1 Mini & ESP-12S
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Upload Using:		SERIAL

### Images

[image]
### Version 2.0.0 without RF_CAL current spike at head

[image]
### Version 2.3.0 with RF_CAL current spike at head

### Sketch

```

#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
extern "C" {
  #include "user_interface.h"
// (Commented out code below for 2.0.0 to expose send_pkt_freedom, not needed in 2,3.0
//  typedef void (*freedom_outside_cb_t)(uint8 status);
//  int wifi_register_send_pkt_freedom_cb(freedom_outside_cb_t cb);
//  void wifi_unregister_send_pkt_freedom_cb(void);
//  int wifi_send_pkt_freedom(uint8 *buf, int len, bool sys_seq);
  void system_deep_sleep_instant(uint32 time_in_us);
}

byte channel;
byte RandomMac = 1; 
byte SequenceLSB = 0;
byte SequenceMSB = 0;
                 

// Packet I am sending
uint8_t packet[128] = { ......   68 bytes,,,,  };      // Obscured for example                                          

void setup() {
  delay(2);   // in this group was all 50
  wifi_set_opmode(STATION_MODE);
  delay(2);
  wifi_promiscuous_enable(1);     // need
  delay(2);
  system_phy_set_max_tpw(20);    //the maximum value of RF Tx Power, unit : 0.25dBm, range [0, 82] 19dBm = 76, 10dBm = 40, 14 dBm = 56
  delay(2);
}

void loop() {

    wifi_set_channel(1);
    delay(2);     // Delay in MS, was 20
    wifi_send_pkt_freedom(packet, 68, 0);
    //Increment fragment
    //Set fragment
    packet[22] = (SequenceLSB + 1);
    delay(10);     // Delay in MS
    wifi_send_pkt_freedom(packet, 68, 0);
    delay(10);     // Delay in MS
    SequenceLSB = (SequenceLSB & 0xF0);
    packet[22] = SequenceLSB;

    wifi_set_channel(6); 
    delay(2);     // Delay in MS
    wifi_send_pkt_freedom(packet, 68, 0);
    delay(10);     // Delay in MS
    //Increment fragment
    //Set fragment
    packet[22] = (SequenceLSB + 1);
    wifi_send_pkt_freedom(packet, 68, 0);
    delay(10);     // Delay in MS
    SequenceLSB = (SequenceLSB & 0xF0); 
    packet[22] = SequenceLSB;

    wifi_set_channel(11); 
    delay(2);     // Delay in MS
    wifi_send_pkt_freedom(packet, 68, 0);
    delay(10);     // Delay in MS
    //Increment fragment
    packet[22] = (SequenceLSB + 1);
    wifi_send_pkt_freedom(packet, 68, 0);
    delay(2);
   
/// ESP.deepSleep(microseconds, mode) will put the chip into deep sleep. mode is one of WAKE_RF_DEFAULT, WAKE_RFCAL, WAKE_NO_RFCAL, WAKE_RF_DISABLED.

    ESP.deepSleep(6000000, WAKE_NO_RFCAL); // Sleep for 6 seconds,

    //system_deep_sleep_set_option(2);
    //system_deep_sleep(6000000);
    //system_deep_sleep_instant(6000000);
 
}

```

wb8wka, 2017-07-11:
Also tried 2.4.0-rc1, same issue.

5chufti, 2017-08-02:
why didn't you post your results on [espressif ESP8266 NON_OS_SDK github](https://github.com/espressif/ESP8266_NONOS_SDK/issues)? Or make a thread on their [BBS](http://bbs.espressif.com/viewforum.php?f=64) 
It's their tree to bark up ...

wb8wka, 2017-08-02:
Because it works fine on the NON_OS_SDK.   It's an Arduino lib issue.  Their tree but it appears abandoned at this point. We've now moved onto NON_OS_SDK

thanks for your interest

5chufti, 2017-08-02:
but using the system_ ... calls accesses the sdk libs directly. Especially when using _system_deep_sleep_instant();_ I can't think of what arduino lib should interfere with the correct execution of deepsleep with wake_no_rfcal ...

5chufti, 2017-08-03:
according to the starting post

> Also
> 
> system_deep_sleep_set_option(2);  // Option 2 is WAKE_NO_RFCAL
> //system_deep_sleep(6000000);
> system_deep_sleep_instant(6000000);  //  Trims 98ms off go to sleep
> 
> has same characteristics in that it cannot turn off RF_CAL

you say that the direct use of system_... has the same problem, this is why I suggested to report to espressif directly.

wb8wka, 2017-08-03:
I'm not quite sure why you don't understand me

This is the Github site for reporting issues with the Arduino port. The issue is with the Arduino command

ESP.deepSleep(6000000, WAKE_NO_RFCAL)

NOT the system calls even though I had issues calling them from the arduino code. That could just as easily be operator error on my part and I wasn't reporting those. 

Howver I have confirmed the system call's  (i.e. system_deep_sleep_instant(6000000);)  work fine directly under the NON_OS_SDK.    There is nothing to report to espressif because the SDK is doing what it is supposed to.

Sometime after 2.0.0 ESP.deepSleep(xxxxx, WAKE_NO_RFCAL) stop turning off the RF calibration on wake.

If you like I can go back and remove all the commented code out of my example. I had just been trying to test various things.

5chufti, 2017-08-03:
with the parts I quoted from your OP you say that the same problem exists calling the system_ ... functions (not having problems calling them) leading to the assumption that this might be the root cause.
Since looking at code for ESP.deep_sleep() you'll see that it does practically the same... and [hasn't changed](https://github.com/esp8266/Arduino/blame/master/cores/esp8266/Esp.cpp) for quite some time.
but believe me, I couldn't care less .....

wb8wka, 2017-08-03:
For not "couldn't caring less" you sure seemed to want to argue this

None-the-less it's a real issue and my testing is solid. That it persisted through a few versions before being reported is irrelevant. 

Hook your scope up and confirm it yourself if you doubt me

universam1, 2017-11-07:
I can confirm that `ESP.deepSleep(x, WAKE_RFCAL)` is broken in that sense that it does not restart the mcu at all.

5chufti, 2017-11-07:
hmm, working like a clockwork on latest master.
is it only WAKE_RFCAL? also yia direct sdk call?
which core/sdk version are you using?
gpio16 - rst are connected?

universam1, 2017-11-08:
I have to revert my statement a bit. So with older SDK a mere reboot did work `ESP.deepSleep(1, WAKE_RFCAL)` with a 1us delay, this is not working any more, I have to specify >1000us for a reliable restart.

wb8wka, 2018-01-05:
Good grief 5chufti....

5chufti, 2018-01-05:
> Good grief 5chufti, this is what I reported back in July....

very helpfull remark ...

you didn't report this
**It works with core 2.0.0 & sdk 1.3.0_15_08_10_p1
it stops working on core 2.0.0 & sdk 1.5.1_16_01_08**
and I think it is relevant

wb8wka, 2018-01-05:
Except it wasn’t until core 2.1.0 that sdk

   - Update SDK to 1.5.1_16_01_08

Was integrated.

No worries, just a lot of wasted time here. I’m glad you confirmed it.

devyte, 2018-01-05:
@wb8wka I assume this is still relevant after 2.4.0 release, and I also assume you erased the whole flash before upload. You said there is no issue with using only the NONOS SDK system calls. Is this true when using pure system calls from within a simple sketch, or did you mean from a NONOS SDK standalone build without the Arduino core? Could you please pist an example?

wb8wka, 2018-01-06:
@devyte That would be my guess. 2.4.0rc1 I tried and it the issue was still there, Reviewing the change log for 2.4.0  I see no mention of a fix.

I meant it worked with the NONOS SDK standalone build   Here is a little clip
```
#define CAL_OPT 2

// This function executes before WiFi is set up
void user_rf_pre_init (void)
{
  // Set no RF cal
  system_deep_sleep_set_option (CAL_OPT);
  system_phy_set_powerup_option (CAL_OPT);
  system_phy_set_rfoption (CAL_OPT);
}

    wifi_send_pkt_freedom(packet, PACKET_SIZE, 0);
```

igrr, 2018-01-06:
I think I have a fix for this, but need some time to set up a current measurement jig...

5chufti, 2018-01-28:
@igrr did you get somewhere with your investigations?

oleg-umnik, 2018-04-05:
@igrr any new on the issue? Maybe some pointers for workaround?

oleg-umnik, 2018-04-05:
@wb8wka have you tried something like this?

```
extern "C" void __run_user_rf_pre_init(void) {
    system_phy_set_rfoption(0);
}
```

torntrousers, 2018-05-04:
or even any hints at what the possible fix mentioned here - https://github.com/esp8266/Arduino/issues/3408#issuecomment-355725812 - could be so we could try to implement the fix ourselves?

torntrousers, 2018-05-10:
I'd so much like to find a fix for this. Could anyone give any hints on where abouts in the code I could be looking for the problem?

ludiazv, 2018-05-23:
Any news on this issue? This is critical for battery power applications.

Regards,

torntrousers, 2018-05-27:
@igrr sorry to keep pestering I know you must be busy on things but you did say earlier you thought you might know the cause of this so could you give any hints so we could try to find some fix?

ludiazv, 2018-05-31:
A hint on the root cause will be very usefull.

On the other hand a possible work arround would be to modify the 108th byte of the init data (defined in core_esp8266_phy.c) and set it to 255. In principle this will make a call woth WAKE_RF_DEFAULT to calibrate the radio only once every 255 reboots from sleep.

Is there any way to change this byte without patching the core? If not it could included with a similar macro to ADC_MODE/RF_MODE.

Christophe91, 2018-07-18:
Any fix date on this issue?
Regards

artua, 2018-07-27:
Need this too...

rcarmo, 2018-08-10:
I'm looking for a fix for this. Any news?

Ivoatme, 2018-08-13:
Hello, I am new to this discussion, so perhaps I missed something…
If the focus is on the broken RF_NO_CAL then this is not relevant but if it relates to costing more power please consider this:

The power cost per cycle in this example is ~ 50 mA/s. The Cal spike “cost” is ~ 5 mA/s.
However, the “cost” of the ESP to set up for the ESP.deepSleep "cost" is ~ 20 mA/s.

If power savings is what you are after, then you may want to consider gating the power to the ESP instead of the deep sleep.
In this scenario, you can bring the power consumption down from the 50 mA/s down to under 20 mA/s

igrowing, 2018-08-14:
@igrr, please hint us what's your solution. I believe community will be able to add pull request with the fix.

Ivoatme, 2018-08-14:
I will provide some captures from my scope. But really quickly here is better explanation...

The "power cost" is defined by integrating the current drawn by the ESP during each sleep cycle. In my example, I am using the ESPNOW code. From the powerup to transmit 250byte payload and receive CB and go to deep sleep... the shortest I can ever make it is ~ 350 ms and so each deep sleep cycle costs ~ 50 mA/s.

However, the payload is transmitted <150 ms and the remaining ~200 ms before you hit deep sleep can be cut out by simply turning off the power. Doing so the power cost to transmit this payload reliably is < 20mA/s. (the ESP literally sits there for 200 ms consuming 70 mA doing nothing except for getting ready to enter the deep sleep.}

In my opinion, there are two separate issues. Yes, the power spikes due to the CAL are detrimental to the very small battery like a coin cell but that can be dealt with by a capacitor. My point is that the spikes you are referencing are so short that their power "cost" is <10 % of the power consumed during each cycle. I do not have the older SDK to compare thing with but I doubt that you will be able to get rid of the first large spike @ ~ 100 ms. I think that this is the ESP (in my case) connecting to the peer. What is interesting is that in that 10ms spike the ESP also transmits the payload.

ludiazv, 2018-08-14:
Did you tried `bool system_deep_sleep_instant(uint64 time_in_us);` sending the esp to sleep? If I remember well it takes only a few ms to do the job. This , in principle, should remove the 200ms where the esp is preparing the sleep.

wb8wka, 2018-08-15:
As a quick update, I got a consultant to help me and we ported my code over to the native SDK.   Here is the current plot (total time on is about 200ms, I send some management frames) along with the code clip.

[image]

```
#include "osapi.h"
#include "user_interface.h"

// This function executes before WiFi is set up
void user_rf_pre_init (void)
{
  system_deep_sleep_set_option (2);   // No RF cal on wake-up
  // 1: Vdd and TX power, 18 ms
  // 2: Vdd only, 2 ms
  // 3: Full cal, 200ms
  system_phy_set_powerup_option (2);
  system_phy_set_rfoption (2);

  wifi_set_opmode (NULL);
}

// MAIN CODE
void user_init(void)
{
      system_deep_sleep_instant (1000000);    // Sleep for 10s
}

******************************************************************************
*******************************************************************************/
uint32 ICACHE_FLASH_ATTR
user_rf_cal_sector_set(void)
{
    enum flash_size_map size_map = system_get_flash_size_map();
    uint32 rf_cal_sec = 0;

    switch (size_map) {
        case FLASH_SIZE_4M_MAP_256_256:
            rf_cal_sec = 128 - 5;
            break;

        case FLASH_SIZE_8M_MAP_512_512:
            rf_cal_sec = 256 - 5;
            break;

        case FLASH_SIZE_16M_MAP_512_512:
        case FLASH_SIZE_16M_MAP_1024_1024:
            rf_cal_sec = 512 - 5;
            break;

        case FLASH_SIZE_32M_MAP_512_512:
        case FLASH_SIZE_32M_MAP_1024_1024:
            rf_cal_sec = 1024 - 5;
            break;

        default:
            rf_cal_sec = 0;
            break;
    }

    return rf_cal_sec;
}
```

devyte, 2018-08-15:
@wb8wka am I reading this right: no cal works with the native sdk, but not with our core?

wb8wka, 2018-08-15:
@devyte That is correct and I reported this behavior first on August 2nd, 2017 and have been watching this thread flop around like a dead fish ever since. I moved onto the native SDK as I needed to get this done but do wish it would work under Ardinio.

wb8wka, 2018-08-15:
@devyte here is the very first trace I posted in this thread, but with the code ported over to the native SDK and both no rf cal and deep_sleep_instant working. I cleaned up my scope noised a little since then as well. Wakes up, boots, sends 6 management frames, and then goes back to sleep. 200ms total time. 1 ohm resistor and 50mv per division

[image]

Ivoatme, 2018-08-15:
[image]

This is running on ESP WROOM 2 @ 160 MHz. and gated power. Note that the purple is the integral of the current and the scale is 0.5 so that the real values must be multiplied 2X.
Total power cost to reliably transmit 248 byte payload is ~ 7.5*2=15 mA/s. 

I was unable to make the system_deep_sleep_instant work.
Below for comparison is identical code except using deepSleep. (same scales power cost=~ 60 mA/s !)
[image]

Ivoatme, 2018-08-15:
@devyte  did you verify that you are receiving data on the other end?
I ask because I was able to generate something similar but I was unable to push any data across.

devyte, 2018-08-15:
I don't use in my app, so I haven't tried any of this. I asked to clarify and to triage this issue.
Thank you for the nonos examples, they help tremendously in these cases.
Something to keep in mind: I understand that deep sleep waits for the wifi to safely shut down before entering deepsleep. In contrast, deep sleep instant just sleeps right away and screw it all. Our deepSleep is using the former. I mention this mostly in case you can't get data across: it's possible that the data doesn't make it out of the radio if you're using instant.
@igrr did you ever get a current measuring rig set up? If not, how about proposing your fix via PR and having @wb8wka or @Ivoatme help out with testing and measuring?

Ivoatme, 2018-08-15:
@wb8wka just a quick comment. The 200 ms current plot you shared may be possibly incorrect.
The current drawn during the Tx is way too small in proportion to the rest of the pre-Tx. 
I suspect three issues worth checking.
1. Scope bandwidth
2. Shunt resistor is too large or otherwise soft PS or something is current limiting
3. The sharp down rail after Tx suggests the possibility the the your ESP simply shuts down due to power starvation and not necessarily due to the code going to sleep.
The ~ 150mA current you see during the Tx should be 300-350 mA under normal operating conditions.
There is another possibility for the low current peaks and that is due to reduced Tx power setting,  but I do not have any data on that and if that is the case we are not comparing apples to apples.

wb8wka, 2018-08-15:
@lvoatme

No, it's correct or at least close. 

1. Scope bandwidth is 50mhz

2. Shunt is 1 ohm with a 470uf cap next to the ESP (downstream of shunt). WiFi frames are being decoded that it is sending.

3.  What you are seeing as the "sharp down rail" is system_deep_sleep_instant() function.  All 6 packets are decoded and seen by the access point. 

 I'm running about 14dBm, according to the data sheet of the ESP8266 that's about 140ma of current consumption. I'm showing spikes of about 160-170ma in my scope diagram.   The downstream 470uf cap does round things, but not to any degree and doesn't impact power calculations.   I'm also NOT gating power here, which would involve a big inrush current to your filter cap. DeepSleep is triggering the device from sleep  OR I assert CE.  In either case I don't need to recharge the input filter caps, which would lead to larger current spikes. 

I should note this is a working application, using a coin cell and I have 100's of pieces now testing.  I'm just commenting to confirm (again) that this is a arduino issue and not a SDK issue as was earlier suggested.  I covered this in most of my earlier posts.

Regards

torntrousers, 2018-08-15:
@devyte ,  slight tangent - "_deep sleep instant just sleeps right away and screw it all. Our deepSleep is using the former_ " - would you accept a pull request that adds a boolean option to ESP.deepSleep for using deep_sleep_instant?

ludiazv, 2018-08-15:
a boolean would be perfect. Nodemcu Lua Interpreter has this option for deepsleep.

5chufti, 2018-08-15:
it is nice that wb8wka posted *some* code that *somehow* shows the wanted behaviour, but this it is no proof for 
>system_deep_sleep_set_option(); 
system_deep_sleep();

working, as he is using total different mechanism.
He is using 
>system_phy_set_powerup_option ();
system_phy_set_rfoption ();

in 
>user_rf_cal_sector_set(void)

what means actively supressing rf_cal after wake_up from deepsleep
instead of setting flag to not do rf_cal at wake_up before going into deepsleep.

this code would be more like it ...
```
#include "osapi.h"
#include "user_interface.h"

// MAIN CODE
void user_init(void)
{
   system_deep_sleep_set_option (2);   // No RF cal on wake-up
   system_deep_sleep_instant (1000000);    // Sleep for 10s
}

******************************************************************************
*******************************************************************************/
uint32 ICACHE_FLASH_ATTR
user_rf_cal_sector_set(void)
{
    enum flash_size_map size_map = system_get_flash_size_map();
    uint32 rf_cal_sec = 0;

    switch (size_map) {
        case FLASH_SIZE_4M_MAP_256_256:
            rf_cal_sec = 128 - 5;
            break;

        case FLASH_SIZE_8M_MAP_512_512:
            rf_cal_sec = 256 - 5;
            break;

        case FLASH_SIZE_16M_MAP_512_512:
        case FLASH_SIZE_16M_MAP_1024_1024:
            rf_cal_sec = 512 - 5;
            break;

        case FLASH_SIZE_32M_MAP_512_512:
        case FLASH_SIZE_32M_MAP_1024_1024:
            rf_cal_sec = 1024 - 5;
            break;

        default:
            rf_cal_sec = 0;
            break;
    }

    return rf_cal_sec;
}
```

5chufti, 2018-08-15:
p.s.: it is symptomatic for espressif that the current esp8266_non_os_sdk_api_reference_en does not even give the values for the different modes in system_deep_sleep_set_option(), nor on system_phy_set_rfoption(). Only for system_phy_set_powerup_option() but they can't be the same ...
(can't find them in .h of sdk either). Who knows, maybe they changed after 1.3.0 SDK?

Ivoatme, 2018-08-16:
I would like to share few thoughts regarding low power designs. My personal experience with the ESP's is that they are power hungry and there are some background processes that take up power that are unpredictable from loop to loop.  To illustrate some of the challenges and to be aware of these things, I am attaching a screen capture of the shortest 248 byte transmission packet I am able to push across.
The purple plot shows the integral of the current. the real scale is 4 mA/s and the total packet is 6.6 *2= ~ 13 mA/s. the timescale is 10 ms. Note that having very hard PS the peaks are off the scale and they are as much as 500 mA. So, as an example... 470 uF capacitor stored energy @ 3.3V rail is ~ 2.5 mJ and so when using coin cell battery as a PS you will never see these peeks because of the inherent high internal resistance. The lessons I learned are that if you are experiencing an unpredictable behavior, first thing to check is the power source. The below example would require at least 2,200 uF to keep up with this packet.  For stable designs consider using super-caps.
[image]

wb8wka, 2018-08-16:
@lvoatme

My measurements with the 470uf cap where with a lab power supply. The application does use a coin cell and I have a one farad cap across it.  My design is quite stable and any inference it isn't is not correct.  It's actually going into a commercial product. 

As to your experiences with the ESP8266, mine were similar with the Arduino ESDP8266 IDE, when I went to the native SDK/NONOS all of a sudden, for the most part things matched the spec sheet.. including TX currents.  Now that could just as easily be because I gained more experience but I had a job to do and the Arduino release and support didn't appear to be being maintained. That this issue has been confirmed and still open well more then a year later tends to support that feeling.

That being said, I'm happy to confirm current measurements as @devyte suggested should this issue ever need confirmation.

wb8wka, 2018-08-16:
@5chufti 

The reason I am actively suppressing rf_cal after wakeup from deepsleep instead of setting flag to not do "rf_cal at wakeup" before going into deepsleep is because I don't use deepsleep to wake up, hence no states are saved.

Instead, I am using chip enabled controlled by a external microcontroller drawing less the a 1 ua on average.   The ~17ua of sleep current the ESP requires in deep sleep would negatively impact my battery life as I am running about one year off a coin cell. My average wake cycle is about 10 minutes on the ESP. 

I hope that clears it up.

Ivoatme, 2018-08-16:
@wb8wka 
My comments had nothing to do with or critiquing your design / application. I am simply sharing some experiences and to help others with design challenges. 
I do realize that this is bit off the main focus which is the issues with limiting the power draw by software. The changes in behavior from SDK to SDK. 
Having seen the poor quality screen captures in this forum, I wanted to illuminate that in order to troubleshoot the "spikes" you have to be able to see them. That of course creates a bit of a challenge because @ 500 mA spike 1 ohm shunt resistor is a considerable regulation problem. And for that reason I wanted to show an example of the current plot using a contact-less current probe. 

I really do not want to get in-to a pissing match here but for sake of anyone that wants to learn something... a bench PS with a 470 uF cap may not be an adequate power source for debugging strange behavior in ESP. And if you throw in to the mix 1 ohm shunt for your scope, you may be in for some challenges.
It may save someone countless hours of frustration trying to catch something that may look like a bug that is actually caused by a soft PS. Once your design is stable and running scaling back to soft PS is the way to go. Again, for development with ESP's I recommend using Super cap's.
 So, if I ruffle anyone's feathers, it certainly was not the intention. My apologizes ;-)

wb8wka, 2018-08-16:
@lvoatme

The fact you pointed out "coin cell" and "470 uf cap", which I earlier stated I used, yes lead me to believe you were referring to me. I am using a 1 farad supercap.  The 470uf cap is on the input pin of the ESP8266 and is more the enough with a stiff (low esr) supply. My design is up and running and what I posted was accurate. I've used lower values for the shunt resistor and other then scaling, the waveform is the same.  

No interest either in getting in a pissing match but it was these sort of exchanges that caused me to leave the project a year ago. I was making a bug report, not looking for a debate. 

What is the contactless probe you are using?

Ivoatme, 2018-08-16:
I am using Tektronix A 6302 current probe with the 503A amplifier.
The supecap for my development if 3000F. (the supercap is a new proprietary part that is in a late stages of development. When it becomes available I will definitely share more info about it) 
In fact... you may be interested in your commercial application in this. This cap unlike "normal" supecaps has NO leakage so it behaves like a battery but with characteristic low esr of a capacitor. 
With that configuration I was able to identify several issues that I thought were a software and they magically went away with the stiff PS. 
Some of the capture issues are related to the equipment bandwidth also, because the ESP "spikes" have such a fast rise time.  I can even see a difference between 100 MHz  and 500 MHz current probes.
What makes things so elusive sometimes if the build in transmission error correction when using TCP. A packet will eventually get across and absent of some fancy triggering it is a challenge to catch.
With a soft PS and using UDP I finally came to realization that if I wanted to keep my sanity eliminating any possible issues by having a low esr PS was a must.

5chufti, 2018-08-16:
@wb8wka 
nope, that does not clear anything. 
You stated that **ESP.deepSleep(xxxx, WAKE_NO_RFCAL)** is not working since 2.0.0 but is working in plain SDK, but you did not proof that the correct equivalent to **ESP.deepSleep(xxxx, WAKE_NO_RFCAL)** is working in SDK but only that you can get similar effect in plain SDK.
We all here are technicians and should very carefully read such statements and tell facts from fiction.

wb8wka, 2018-08-16:
@5chufti 

If you'd be so kind as to look at your post from November 7th, 2017 you'll see you stated:

> @igrr
> **I can now confirm that WAKE_RF_NOCAL is broken**, via
> ESP.deepSleep();
> as well as
> system_deep_sleep_set_option();
> system_deep_sleep();

Which of course was my bug report. 

As we've now come full circle I'll wish you well.

5chufti, 2018-08-16:
@wb8wka 
the bug report per se was correct.
Your following statement that the equivalent to **ESP.deepSleep(xxxx, WAKE_NO_RFCAL)** is working in SDK was not!
why is it allways so hard to discuss with HAMs on pure technical grounds?
fare well to you and good luck to @igrr, @devyte chasing down the error in the wrong direction...

wb8wka, 2018-08-16:
@5chufti 

Before your latest @igrr indicated he had a fix for this on Jan 7th 2018 so I doubt he got lead in the wrong direction by my moving onto the SDK and using a different method to achive my goal. 

As to your question, I do not know other then to point you back to your last exchange on August 3rd, 2017 which might give you some insight.  This is a bug reporting system,  not a debate board.

mailgpa, 2018-08-16:
It looks like [this change](https://github.com/esp8266/Arduino/commit/e02932fcddc24740852ec0e39e494e57a8906df2) is causing problems. At least, if I disable this block, I can see that wake up time is dropping to ~40ms. But I do not yet understand what it is doing.

torntrousers, 2018-08-16:
Dude! @mailgpa you're a star. That fixes it for me too, wake up to setup() goes from about 200ms down to 39ms.

mailgpa, 2018-08-16:
Thanks to @wb8wka who gave a hint that it works in native SDK. That change seems to be a workaround for some old bug from 2015 which must have been fixed in SDK already.

ludiazv, 2018-08-16:
I confirm that the solution also works fine also for me. Also I tested instant version of deep sleep in my scope, it put the device to sleep in about 5ms. Both PRs submitted by @torntrousers will be able to reduce the wake time more than 350ms. For espnow this will potentially double your battery life.
In mqtt scenarios, this could represent a 20%-30% of improvement.

thanks @mailgpa  for the insight! Finally you have solved this old issue!

devyte, 2018-08-17:
@mailgpa how in the holy source code did you arrive at that? It's like you reached into fhe haystack and pulled out the needle on the first try!

[33 more replies]

closed 2020-01-05 by devyte