esp8266/Arduino #3410: Add button or redirect to /

SadovovAlex, 2017-07-09:
Hi

may be you add a button after successfuly update to redirect to the main page?

Arduino/libraries/ESP8266HTTPUpdateServer/src/ESP8266HTTPUpdateServer.cpp
 "<META http-equiv=\"refresh\" content=\"15;URL=\">Update Success! Rebooting...\n";

add this button to redirect / or autoredirec ?

thx

devyte, 2017-09-05:
I suspect this is probably a good idea, so I invite you to make a PR of what you mean, so that it can be evaluated and tested.

devyte, 2017-10-24:
@SadovovAlex will you be making a pr of your idea?

SadovovAlex, 2017-11-04:
const char* ESP8266HTTPUpdateServer::_successResponse = "<META http-equiv=\"refresh\" content=\"15;URL=/\">Update Success! Rebooting...<input type=\"button\" onclick=\"window.location='/'\" value=\"Go back\"/>";

[ESP8266HTTPUpdateServer.zip](https://github.com/esp8266/Arduino/files/1443010/ESP8266HTTPUpdateServer.zip)

closed 2017-11-04 by SadovovAlex