esp8266/Arduino #3412: ESP8266httpUpdate: How to ensure correct/valid update-file?

tuxedo0801, 2017-07-11:
Hi there,
how do I prevent the http-ota-update from downloading a wrong/invalid firmware file?

Of course there is a MD5 check available. But what about a file-download which get's intercepted/redirected to a kind of malware source? The esp will download the "wrong" file, checks the file integrity with help of MD5 sum and will finally update with the malicious firmware...

Is there a https-example for http-ota-update somewhere?
Has anyone thought about signing the firmware-file? --> would be independant from https/ssl/tls/... 

br,
Alex

tuxedo0801, 2017-07-11:
Just found this: #3105

WheresWally, 2017-07-13:
Ok, I have been wondering about this for a while. I would not like to see a take over of ESP/Arduino cores via updater

earlephilhower, 2019-01-16:
#5213 closed this with full crypto-signed update support, if you enable it.

closed 2019-01-16 by earlephilhower