esp8266/Arduino #3413: AP mode and Telnet problem

OscarGLorenz, 2017-07-11:
Hi,

I tried to put the ESP8266 in AP mode and later to recieve and send data vía telnet, I merged the examples WifiAccessPoint and WiFiTelnetToSerial. 
When a device connects to the ESP8266 via telnet appears on Serial New client: 0, but if the device sends data it will not show on serial in the ESP, and if the ESP sends something the device will not show it.
The two bare examples works for me.

What I am doing wrong?

Thanks

### Basic Infos

#### Hardware
Hardware:			ESP-8266
Core Version:      	2.2.0

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Reset Method:		ck 

### Sketch

```cpp

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <WiFiClient.h> 

//how many clients should be able to telnet to this ESP8266
#define MAX_SRV_CLIENTS 
const char *ssid = "ESPap";
const char *password = "thereisnospoon";

WiFiServer server(23);
WiFiClient serverClients[MAX_SRV_CLIENTS];

void setup() {
  delay(1000);
  Serial.begin(115200);

  Serial.println();
  Serial.print("Configuring access point...");
  WiFi.softAP(ssid);
  
  server.begin();
  server.setNoDelay(true);
  
  Serial.print("Ready! Use 'telnet ");
  Serial.print(WiFi.softAPIP());
  Serial.println(" 23' to connect");
}

void loop() {
  uint8_t i;
  //check if there are any new clients
  if (server.hasClient()){
    for(i = 0; i < MAX_SRV_CLIENTS; i++){
      //find free/disconnected spot
      if (!serverClients[i] || !serverClients[i].connected()){
        if(serverClients[i]) serverClients[i].stop();
        serverClients[i] = server.available();
        Serial.print("New client: "); Serial.println(i);
        continue;
      }
    }
    //no free/disconnected spot so reject
    WiFiClient serverClient = server.available();
    serverClient.stop();
  }
  //check clients for data
  for(i = 0; i < MAX_SRV_CLIENTS; i++){
    if (serverClients[i] && serverClients[i].connected()){
      if(serverClients[i].available()){
        //get data from the telnet client and push it to the UART
        while(serverClients[i].available()) Serial.write(serverClients[i].read());
      }
    }
  }
  //check UART for data
  if(Serial.available()){
    size_t len = Serial.available();
    uint8_t sbuf[len];
    Serial.readBytes(sbuf, len);
    //push UART data to all connected telnet clients
    for(i = 0; i < MAX_SRV_CLIENTS; i++){
      if (serverClients[i] && serverClients[i].connected()){
        serverClients[i].write(sbuf, len);
        delay(1);
      }
    }
  }
}

devyte, 2017-09-05:
I suggest trying with a single server client first, before trying with multiple.
In any case, this is not the right place to discuss what is wrong with your particular project. This is an issue tracker, meant to track issues in the core libs. As you said, the individual examples work, so I don't think there is an issue in the core libs.
Please refer to esp8266.com or stackoverflow for assistance.
Closing.

closed 2017-09-05 by devyte

ancich, 2019-03-28:
Oscar, for what it is worth, your code worked for me!
Meanwhile the telenet-wifi example code provided by this site, did not.

d-a-v, 2019-03-28:
Please elaborate on what doesn't work with examples.
They are supposed to be documentation...

trentbrown13, 2019-03-28:
d-a-v and devyte, for what its worth I have also been trying the wifi-telnet example with the same results Oscar is noting - I can telnet to the esp but I seem to get no serial data in either direction.  I've tried several nodemcu's and a few routers running openwrt with no success. 
I'm running core 2.3.0.
I figured it was operator error even though  it would seem to be a simple application. The setup is torn down now but I can put it back together quickly if needed. I'll probably try Oscars code above as it seems to work for ancich.

d-a-v, 2019-03-28:
core-2.3.0 are you sure ?

trentbrown13, 2019-03-28:
Yes, i just checked. I went back to 2.3.0 a while back after an issue in
2.4 (probably fixed now) and have not upgraded since. I can try the
telnet-wifi example with a more current core version tomorrow if that will
help.

ancich, 2019-03-28:
I have no idea what core version I have.  How do I check?
Anyhow, I get the same non-functioning mentioned by trentbrown13. I can connect a terminal through telnet, and the esp-01 end shows client connected, but no key-strokes at either terminal does anything.

Oscar's code on the other hand, worked first try. And believe me, that seems to be a rarity with anything to do with the esp8266.

trentbrown13, 2019-03-28:
if you are using the IDE, open the boards manager under tools->board->board
manager and look for esp8266 by the esp8266 community. It will show the
current installed version.