esp8266/Arduino #3416: Conflict between interrupt and Wifi scan or Station mode

Banzai00, 2017-07-12:
### Basic Infos

#### Hardware
Hardware:			WROOM-02
Core Version:      	2.4.0-rc1

### Description
I am using interrupt to calculate pulses length but it seems that when interrupt is used while a WiFi scan is running or the esp is in the Station mode, the esp crashes showing Exception (0) that you can see in the debug messages.

My Wificonfig fonction start by a wifi scan : in case she don't find a specific network "SSID" she creates the AP  but in the case she find it she must connect to it.

The esp was crashing when the scan started so i disabled the interrupt during the scan. It solved the problem partially. As the esp still crashes when configured in the Station mode (Still the same error).
But when in the AP mode it works fine.

I couldn't find the cause of the problem. So i need your help guys. I assume that i need to disable interrupt when some wifi function is executed but i don't know exactly where. 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB

### Sketch

```cpp

IPAddress staticIP(203,114,88,224);
IPAddress gateway(203,114,88,1);
IPAddress subnet(255,255,255,0);

int oldvalue = 240;
boolean net;
boolean sscan;

AsyncWebServer server(80);

volatile unsigned long duration,pwm_value,prev_time;

void handleRoot(AsyncWebServerRequest *request) {
  request->send(200, "text/plain", "hello");
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(200);  
  pinMode (12,INPUT);
 
  server.on("/", HTTP_GET, handleRoot);    
   server.onNotFound([](AsyncWebServerRequest *request){
        request->send(200, "text/plain", "not found");   
  });         

  server.begin();
  Serial.println("HTTP server started");              
  

  WiFi.mode(WIFI_STA);
  WiFi.disconnect(); 
  delay(100); 
}

void loop() {
  // put your main code here, to run repeatedly:
  Wificonfig ();  
  delay(40);
}

void falling() {
  attachInterrupt(12, rising, RISING);
  prev_time = micros();
}
 
void rising() {
  attachInterrupt(12, falling, FALLING);
  pwm_value = micros()-prev_time;
}

void Wificonfig (){
  if (WiFi.status() != oldvalue) {    
    oldvalue = WiFi.status();    
    if (WiFi.status() == WL_NO_SSID_AVAIL ||WiFi.status() == WL_IDLE_STATUS)sscan = true;  
  }  
  if (sscan){
          sscan = false;
          detachInterrupt(12);          
          WiFi.scanNetworks(true);
   }
  int n = WiFi.scanComplete();
  if(n >= 0)  {   
          if (n > 0){
            for (int i = 0; i < n; ++i){             
              Serial.print(WiFi.SSID(i));
                  if (WiFi.SSID(i) == "SSID"){
                    net = true;
                    break;
                  }
                  
                  delay(10);
        
             }
            }                  
            delay(10);
            if (net){                  
                                WiFi.mode(WIFI_STA);
                                WiFi.begin("SSID", "PASSWORD");
                                WiFi.config(staticIP, gateway, subnet);
                                while (WiFi.status() != WL_CONNECTED)
                                {
                                  delay(500);
                                }                                
                                net = false; 
         
              }else{
                      if (WiFi.softAPIP() != gateway){
                             WiFi.mode(WIFI_AP_STA);
                             WiFi.softAP("SSID", "PASSWORD");
                             WiFi.softAPConfig(gateway, gateway, subnet);               
                      }
                                
              } 
              WiFi.scanDelete();

              attachInterrupt(12, falling, FALLING);

  }
  delay(50);
}
```

### Debug Messages

```
Exception (0):
epc1=0x4020682c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc20 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdc0:  401062dc 0000000a 3ffef824 4020b1a9  
3ffffdd0:  ffffffff 00000020 00000020 00000a0d  
3ffffde0:  3ffeef48 3ffef680 3ffffe0d 00000022  
3ffffdf0:  3fffc200 401062a4 3fffc258 4000050c  
3ffffe00:  40004320 00000030 00000012 ffffffff  
3ffffe10:  60000200 00000000 00000005 80000000  
3ffffe20:  20000000 80000000 00000005 00000000  
3ffffe30:  00000006 00000000 3ffef680 3fff22ec  
3ffffe40:  00000014 0007c260 60000600 00000030  
3ffffe50:  4021781c 00000000 00000000 4020b2c4  
3ffffe60:  ffffffff 0000000c 00000000 4020680e  
3ffffe70:  40106340 3fff0dfc 00000000 4010632a  
3ffffe80:  ffffffff 00000020 3fff121c 00000000  
3ffffe90:  00000000 00000000 0000001f 401049ed  
3ffffea0:  4000050c 401062a4 3fffc258 4000050c  
3ffffeb0:  40000f68 00000030 00000012 ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  feefeffe 3ffef680 3ffef680 fffffffe  
3ffffee0:  ffffffff 3ffe9b34 3ffef680 3fffdab0  
3ffffef0:  00000000 400042db 3ffef8e4 00000030  
3fffff00:  40004b31 3fff208c 00000274 0007c000  
3fffff10:  401055b0 00000010 00000010 401065f4  
3fffff20:  401061f5 3fff208c 3ffeef70 00f96187  
3fffff30:  3fff0dfc 3fff0df8 3ffedfa0 4021f5c6  
3fffff40:  4021fa82 00000008 60000600 40217915  
3fffff50:  00000000 00000000 4021fb47 3ffee054  
3fffff60:  3ffeef70 004dc599 3ffeef70 402012d8  
3fffff70:  40231cfd 3ffee054 3ffeef70 00f92f1a  
3fffff80:  40231d42 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffeef80 3fffdad0 3ffef8e4 4020ba47  
3fffffa0:  40000f49 3ffe9dd0 3fffdab0 40000f49  
<<<stack<<<

Exception 0: Illegal instruction
Decoding 32 results
0x4020682c: rising() at C:\Users\user\Documents\Arduino\sketch_jul11d/sketch_jul11d.ino line 73
0x401062dc: interrupt_handler at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_wiring_digital.c line 125
0x4020b1a9: Print::write(char const*) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/Print.cpp line 107
0x401062a4: interrupt_handler at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_wiring_digital.c line 119
0x4021781c: igmp_timer at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/timers.c line 372
0x4020b2c4: Print::println(unsigned long, int) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/Print.cpp line 107
0x4020680e: falling() at C:\Users\user\Documents\Arduino\sketch_jul11d/sketch_jul11d.ino line 69
0x40106340: interrupt_handler at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_wiring_digital.c line 141
0x4010632a: interrupt_handler at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_wiring_digital.c line 137
0x401049ed: ets_timer_disarm at ?? line ?
0x401062a4: interrupt_handler at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_wiring_digital.c line 119
0x401055b0: spi_flash_read at ?? line ?
0x401065f4: pvPortMalloc at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/heap.c line 40
0x401061f5: __wrap_spi_flash_read at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_phy.c line 267
0x4021f5c6: pm_set_sleep_time at ?? line ?
0x4021fa82: pm_get_sleep_type at ?? line ?
0x40217915: sys_check_timeouts at /Users/ivan/e/arduino-esp8266/tools/sdk/lwip/src/core/timers.c line 420
0x4021fb47: pm_get_sleep_type at ?? line ?
0x402012d8: delay_end at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_wiring.c line 41
0x40231cfd: ets_timer_handler_isr at ?? line ?
0x40231d42: ets_timer_handler_isr at ?? line ?
0x4020ba47: loop_task at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266/core_esp8266_main.cpp line 57

```

rolpal, 2017-08-06:
First, rewrite your sketch to use CHANGE and only one ISR.
In the ISR read the pin to detect if it is low or high and based on that do your calculations and set a flag when you have a result.
In main loop remove the delay and If you want to print PWM value look for the flag...

Redefining the ISR in the ISR seems to me to be bad practise.

Banzai00, 2017-08-18:
I made the change you proposed but it's still crashes when i use delay() in the main loop. And delay is necessary for me without calling her the Wifi dont work. So do you have other ideas? or Interrupt and delays in main loop can't work together?

Makuna, 2017-08-18:
Please provide your new updated sketch.

Having no delay in the loop I doubt is your problem; it is not required as the code that calls your loop is already doing everything to allow WiFi to work behind the scenes as long as you don't "stall" in the loop.

You can't call the serial routines within an ISR.  You have to store a flag/value and call them within your loop.

devyte, 2017-09-05:
@Banzai00 any progress with this? What @Makuna says is essentially correct.

rolpal, 2017-09-06:
Well, there will be no progress if you do not provide the latest version of the non working code...

igrr, 2017-09-06:
@Banzai00:
The digit '2' in the ISR address (`0x40201ea8: rising() `) indicates that 'rising' is not actually placed into IRAM, it is still in flash.

Arduino IDE (before certain version, don't remember which) does not correctly generate forward declarations for functions which have `__attribute__`s in their definitions.
You need to declare ISR function manually, without relying on Arduino to generate forward declarations for you. And use ICACHE_RAM_ATTR in the definition. That should solve the issue.

Banzai00, 2017-09-13:
Sorry it took me so long to respond.
Thanks Ivan, it's working now (i have arduino IDE version 1.6.8, it's
really strange to do the declarations manually).

devyte, 2017-10-24:
Closing as resolved.

closed 2017-10-24 by devyte