esp8266/Arduino #3418: How to use 2.4.0-rc1 with patform.io?

tuxedo0801, 2017-07-13:
Hey,
how can I use the latest 2.4.0-rc1 with platform.io?

I initially thought the platform-io stuff for esp8266 comes from same repository? But the versions do not match.

In platform.io I have "espressif8266" platform in version 1.3.1 installed, but this does not contain a feature which I require which comes with 2.4.0-rc1.....

br,
Alex

porkyneal, 2017-07-15:
There is an open pull request which will directly use this repository within platformio. See here #3321.
We just need to wait for this to be finalised and merged by  @igrr  and @ivankravets.

It would certainly be a welcome addition.

ivankravets, 2017-07-15:
See http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

ivankravets, 2017-08-11:
Recently has been resolved. Please follow docs steps:

- http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

closed 2017-08-11 by ivankravets