esp8266/Arduino #3422: SMTP program not running in my main program errors in below: I used here Wemos D1 ESP8266.. What i do plz send me in details.

abhijitsd55, 2017-07-15:
Please fill the info fields, it helps to get you faster support ;)

ERROR:

IP address: 192.168.1.144
Connect to  http://192.168.1.144

Panic C:\Users\ABHID\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src\WiFiClientSecure.cpp:399 ax_port_realloc

ctx: cont 
sp: 3fff3670 end: 3fff3d40 offset: 01b0

>>>stack>>>
3fff3820:  40106cc6 3fffb7e8 00000082 3fffb580  
3fff3830:  00000478 00000001 00004840 40209ec9  
3fff3840:  3fffb6c0 40106c9a 3fffb6b0 40223cbb  
3fff3850:  3fffb580 3fffb7e8 00000081 40224921  
3fff3860:  00000001 00000000 00000000 00000000  
3fff3870:  3fffbb70 3fffba38 00000040 00000000  
3fff3880:  40101389 3fffabf0 00000041 00000041  
3fff3890:  00000000 00000001 00000081 3fffb7e8  
3fff38a0:  3fffb6b0 3fffabf0 00000000 40209ebb  
3fff38b0:  3fffb580 3fffabf0 3fffac48 0000007f  
3fff38c0:  3fffb7e8 3fffabf0 3fffabf0 40224be9  
3fff38d0:  00000002 00000001 00000000 40223d20  
3fff38e0:  00000100 00000100 3fffb438 3fffb548  
3fff38f0:  3fff8a74 00000100 3fffabf0 40225da4  
3fff3900:  3fffb438 3fffabd8 00000003 40226598  
3fff3910:  3fffab98 40106c9a 00000003 00000000  
3fff3920:  00000100 3fff9406 3fff3a48 40226a86  
3fff3930:  3fffabd8 3fffb438 3fff3a48 4022691c  
3fff3940:  3fff8a48 00000013 3fff3a48 402267bc  
3fff3950:  0000024f 00000004 3fff9406 00000000  
3fff3960:  00000004 3fff8a48 3fff9406 40222ae6  
3fff3970:  00000263 401034ab 00000002 3ffec380  
3fff3980:  00000001 00000000 00000020 40103b33  
3fff3990:  401033a4 3ffefe60 3ffec380 7fffffff  
3fff39a0:  126e4e4b 16128630 06b28108 cff008c7  
3fff39b0:  c7717e66 eec2edd4 3b1ffff0 f0c84ed6  
3fff39c0:  4338b0b9 307d18d0 5583a26a cb36119c  
3fff39d0:  424adbd1 c22ab5c0 38647811 6c186210  
3fff39e0:  af211b68 00000000 00000000 00000000  
3fff39f0:  00000000 00000000 00000000 00000000  
3fff3a00:  00000000 00000000 00000000 00000000  
3fff3a10:  00000000 00000000 00000000 00000000  
3fff3a20:  00000000 00000000 00000000 ef200000  
3fff3a30:  66469e1f ff71f7eb 67e9dd4a b7b7495e  
3fff3a40:  058ac940 0000042c 00000203 00000318  
3fff3a50:  3fff7540 00000263 00000001 00000000  
3fff3a60:  3fff7a10 00000000 50e23f2f d5d89814  
3fff3a70:  fd0db69f ef3d0e62 177c65fb 3fff8f75  
3fff3a80:  3fff7540 0000042c 00000491 40221392  
3fff3a90:  00000c3a 3fff75a0 00000c41 00020011  
3fff3aa0:  f31ec567 4cb8fdfd 47303529 764c0b80  
3fff3ab0:  206108fa a8a66205 3920b7a1 00000c41  
3fff3ac0:  3fff8f75 00000000 3fff75a0 402223d9  
3fff3ad0:  848d61c7 0e7ce849 b7efbb28 4507e417  
3fff3ae0:  620dd71d 1771e8ad d200f354 5545f5ce  
3fff3af0:  c06e4e09 3fff76c0 3fff7680 00000bc0  
3fff3b00:  00000c41 3fff9b77 0000003e 3fff76c0  
3fff3b10:  3fff75a0 3fff8f75 00000c41 00000c41  
3fff3b20:  3fff8f75 00000000 3fff75a0 40221f2e  
3fff3b30:  00000038 00000001 00000033 00000000  
3fff3b40:  3fff75a0 00000000 00000038 4022164c  
3fff3b50:  00020010 0000000b 00000010 00000c41  
3fff3b60:  00000033 00000016 3fff75a0 00000000  
3fff3b70:  00000000 3fff75a0 3fff75a0 40222062  
3fff3b80:  00000000 3fff8f75 3fff75a0 40222229  
3fff3b90:  00000000 3fff74a8 3fff75a0 402222b0  
3fff3ba0:  3fff51a0 4020c732 3fff3c5c 3fff2bac  
[... 545 lines cut ...]
    loopCount++;
    // if nothing received for 10 seconds, timeout
    if (loopCount > 10000) {
      client.stop();
      Serial.println(F("\r\nTimeout"));
      return;
    }
  }

  while (client.available())
  {
    thisByte = client.read();
    Serial.write(thisByte);
  }
  client.stop();
  Serial.println(F("disconnected"));
}

void loop(void)
{
  
  server.handleClient();
  

}

igrr, 2017-12-28:
WiFiClientSecure has received a number of fixes since this issue has been opened. Please try again with git version and report back.

devyte, 2018-07-12:
Closing due to lack if feedback and bearssl integration.

closed 2018-07-12 by devyte