esp8266/Arduino #3429: Compile error when trying the CardInfo.ini examplo from SD Library

lrossel, 2017-07-16:
In order to learn how to work with sd card with ESP8266 dev. kit and the SD Card, I'm trying to compile the CardInfo.ino example included with the sd library.

When I compile the sketch, this is the error message from Arduino IDE:

Arduino:1.8.3 (Windows 10), Tarjeta:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\lrossel\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\lrossel\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\lrossel\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10803 -build-path C:\Users\lrossel\AppData\Local\Temp\arduino_build_35037 -warnings=none -build-cache C:\Users\lrossel\AppData\Local\Temp\arduino_cache_382529 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -verbose C:\Program Files (x86)\Arduino\libraries\SD\examples\CardInfo\CardInfo.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\lrossel\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\lrossel\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\lrossel\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10803 -build-path C:\Users\lrossel\AppData\Local\Temp\arduino_build_35037 -warnings=none -build-cache C:\Users\lrossel\AppData\Local\Temp\arduino_cache_382529 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -verbose C:\Program Files (x86)\Arduino\libraries\SD\examples\CardInfo\CardInfo.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1
Using core 'esp8266' from platform in folder: C:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1
Detecting libraries used...
"C:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/include" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/lwip/include" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\lrossel\AppData\Local\Temp\arduino_build_35037/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10803 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\variants\nodemcu" "C:\Users\lrossel\AppData\Local\Temp\arduino_build_35037\sketch\CardInfo.ino.cpp" -o "nul"
"C:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/include" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/lwip/include" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\lrossel\AppData\Local\Temp\arduino_build_35037/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10803 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\variants\nodemcu" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\SPI" "C:\Users\lrossel\AppData\Local\Temp\arduino_build_35037\sketch\CardInfo.ino.cpp" -o "nul"
"C:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/include" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/lwip/include" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\lrossel\AppData\Local\Temp\arduino_build_35037/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10803 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\variants\nodemcu" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\SPI" "-IC:\Program Files (x86)\Arduino\libraries\SD\src" "C:\Users\lrossel\AppData\Local\Temp\arduino_build_35037\sketch\CardInfo.ino.cpp" -o "nul"
"C:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/include" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/lwip/include" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\lrossel\AppData\Local\Temp\arduino_build_35037/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10803 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\cores\esp8266" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\variants\nodemcu" "-IC:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\SPI" "-IC:\Program Files (x86)\Arduino\libraries\SD\src" "C:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc1\libraries\SPI\SPI.cpp" -o "nul"
"C:\Users\lrossel\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\lrossel\AppData\Local\Ardu [... cut ...]
Usando librería SD con versión 1.1.1 en la carpeta: C:\Program Files (x86)\Arduino\libraries\SD 
exit status 1
Error compilando para la tarjeta NodeMCU 1.0 (ESP-12E Module).

At this point I have not connected yet the ESP8266 to my computer.
Because of this message:
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:510:2: error: #error Architecture or board not supported.
I believe it's some problem of compatibility of the Library with the NodeMCU 1.0 (ESP-12E Module).

Any idea?

lrossel, 2017-07-16:
This is the comprised compile error message:

Arduino:1.8.3 (Windows 10), Tarjeta:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2Card.h:26:0,

                 from C:\Program Files (x86)\Arduino\libraries\SD\src/utility/SdFat.h:29,

                 from C:\Program Files (x86)\Arduino\libraries\SD\src/SD.h:20,

                 from C:\Program Files (x86)\Arduino\libraries\SD\examples\CardInfo\CardInfo.ino:24:

C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:510:2: error: #error Architecture or board not supported.

 #error Architecture or board not supported.

  ^

exit status 1
Error compilando para la tarjeta NodeMCU 1.0 (ESP-12E Module).

lrossel, 2017-07-18:
Update:

Problem solved.

I was using a not stable board. I downgraded ESP8266 board to last stable version and the problem was solved.

ashot11091950, 2018-06-20:
i have same problem, what can i do?

earlephilhower, 2018-11-08:
This error it caused by having an incompatible Arduino SD library in the sketch directory, so the compiler doesn't use the one for the ESP8266 in the 8266 tree.  Remove the offending library from the sketches/libraries folder and compiles will succeed.

Closing.

closed 2018-11-08 by earlephilhower