esp8266/Arduino #3431: ESP crashing upon https update

ismavg, 2017-07-18:
###Info
Trying to do OTA update from HTTP/HTTPS server. HTTP is working but https is not. the debug output is below. please help in any way you can.
#### Hardware
Hardware:			ESP 12E
Core Version:      	GIT version even tried 2.3

### Description

esp crashing upon https update

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		OTA
Reset Method:		ck

### Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  WiFi.begin("Awalk", "creatinno");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  Serial.println("V12 Connected, updating");
  delay(1000);

  t_httpUpdate_return ret = ESPhttpUpdate.update("https://rondu-b0089.firebaseapp.com/arduino.ino.bin","","80 83 ee c4 1b 1d 1c 8c 4c 77 c0 fa fa 08 91 04 39 c2 9c a4");
  //t_httpUpdate_return ret = ESPhttpUpdate.update("http://awalk.co.in/okati.ino.bin");
  // If OTA update is successful ESP will restart HERE

  Serial.print("Update failed with error number: ");
  Serial.println((int) ret);
  switch (ret) {
    case HTTP_UPDATE_FAILED:
      Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
      Serial.println();
      Serial.println();
      Serial.println();
      break;

    case HTTP_UPDATE_NO_UPDATES:
      Serial.println("HTTP_UPDATE_NO_UPDATES");
      Serial.println();
      Serial.println();
      break;
  }
  delay(500);
}

void loop() {
}
```

### Serial Output debug level: all

[... 304 lines cut ...]
3fff28b0:  0000000f 0000000e 3fff4724 0000000f  
3fff28c0:  00000005 3fff4acc 0000011f 0000011e  
3fff28d0:  3fff4764 0000001f 00000013 3fff499c  
3fff28e0:  0000000f 00000000 3fff49c0 00000001  
3fff28f0:  000000c8 0004ac60 3fff2900 00000000 <
3fff2900:  00000000 00000000 00000000 00000000  
3fff2910:  00000000 00000000 3fff2940 40206a66  
3fff2920:  3ffe89d0 000003e8 000003e8 3fff1968  
3fff2930:  3fffdad0 3fff16b8 3fff18a8 40202261  
3fff2940:  3fff46b4 0000003f 0000003b 3fff3db4  
3fff2950:  0000000f 00000000 3fff4664 0000003f  
3fff2960:  00000031 feefeffe feefeffe feefeffe  
3fff2970:  3fffdad0 00000000 3fff1960 40207527  
3fff2980:  feefeffe feefeffe 3fff1970 40100978  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```

igrr, 2017-12-28:
WiFiClientSecure memory footprint has been improved since this issue was opened. Please try latest git version.

ismavg, 2017-12-28:
Don't have the hardware right now will get back to you after few days. Thank you :)

ismavg, 2018-01-16:
Thank you very much. it is working. Closing the issue

closed 2018-01-16 by ismavg

onionhammer, 2019-01-19:
@igrr how much heap should be available prior to running this?