esp8266/Arduino #3432: The module stops after the config

mcmega, 2017-07-18:
Hi all!
Use ESP8266-12 (D1 mini)
I write the firmware to the module. I configure the connection parameters via the Web interface.
I reboot the module with the command ESP.restart ();
Reboot starts, but wdt/
The module is normally started only after pressing the Reset or power off button.

mcmega, 2017-07-21:
Did so:
1. Connected Reset through a 100n capacitor to GND
2. Pullup Reset through the resistor to the VCC
3. GPIO15 connected with capacitor 10uF and Reset
When the module is started, I set "1" on GPIO 15. When you need to reset - set "0" and Wifi reboots.
Is this the correct solution?

devyte, 2017-10-24:
@mcmega this is a known issue, and there is currently no known workaround. After uploading a sketch over serial, the module must be physically reset before using, or you'll get the described symptoms.
Closing.

closed 2017-10-24 by devyte