esp8266/Arduino #3434: Esp8266 error

Abdul2988, 2017-07-19:
----------------------------- Remove above -----------------------------

### Basic Infos Not working

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description it keeps saying this: Arduino: 1.8.3 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

Archiving built core (caching) in: C:\Users\sosa\AppData\Local\Temp\arduino_cache_284288\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadTool_esptool,UploadSpeed_115200,FlashSize_4M3M_4beafef8a25368dc356c168078b08d42.a
Sketch uses 198612 bytes (19%) of program storage space. Maximum is 1044464 bytes.
Global variables use 33014 bytes (40%) of dynamic memory, leaving 48906 bytes for local variables. Maximum is 81920 bytes.
error: Failed to open COM4
error: espcomm_open failed
error: espcomm_open failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

Abdul2988, 2017-07-19:
and this Arduino: 1.8.3 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

Archiving built core (caching) in: C:\Users\sosa\AppData\Local\Temp\arduino_cache_284288\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadTool_esptool,UploadSpeed_115200,FlashSize_4M3M_4beafef8a25368dc356c168078b08d42.a
Sketch uses 198612 bytes (19%) of program storage space. Maximum is 1044464 bytes.
Global variables use 33014 bytes (40%) of dynamic memory, leaving 48906 bytes for local variables. Maximum is 81920 bytes.
error: Failed to open COM4
error: espcomm_open failed
error: espcomm_open failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

JamesGKent, 2017-08-04:
this line: `error: Failed to open COM4` tells you everything you need to know. you do not have a COM4 on this pc, that is either because you are on the wrong port or you may have some driver issues.

devyte, 2017-09-05:
If you unplug/replug the module, sometimes the COM can change.
In any case, this is not the right place to ask for usage assistance, as this is an issue tracker for the core libs. Please refer to esp8266.com or stackoverflow for discussion, or look at examples online.
Closing.

closed 2017-09-05 by devyte

yadakrishna21, 2017-10-16:
Archiving built core (caching) in: C:\Users\CHINTU\AppData\Local\Temp\arduino_cache_595012\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadTool_esptool,UploadSpeed_115200,FlashSize_4M3M_67e4ff5307eb8c1eec9f1511e35f03cd.a
Sketch uses 554878 bytes (53%) of program storage space. Maximum is 1044464 bytes.
Global variables use 56968 bytes (69%) of dynamic memory, leaving 24952 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed