esp8266/Arduino #3441: Problem with ADC and delay?

mumrah, 2017-07-20:
### Basic Infos

#### Hardware
(not sure how to determine these)
Hardware:		NodeMCU ESP8266
Core Version:      	?2.1.0-rc2?

### Description

If I have any code using `delay`, I get regular crashes. The stack trace always comes from the same method: [adc_single_channel_read(unsigned char)](https://github.com/williamhemmingsen/LibAPRS_ESP/blob/master/AFSK.cpp#L510). The library I'm using is sampling from an external ADC using an interrupt on timer1.

Using the sketch below, I will see one or two "tick" messages printed out followed by the crash.

### Settings in IDE

Module:  			NodeMCU 0.9 (ESP-12 Module)
Flash Size: 		?4MB/1MB?
CPU Frequency:		80MHz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

```cpp
#include <LibAPRSesp.h>

void aprs_msg_callback(struct AX25Msg *incomingPacket) {

}

void setup() {
  Serial.begin(9600);
  APRS_init();
}

void loop() {
  delay(5000);
  Serial.println("tick");
}
```

[LibAPRSesp repo](https://github.com/williamhemmingsen/LibAPRS_ESP) is a ESP specific fork of LibAPRS.

### Debug Messages

```
Exception 0: Illegal instruction
Decoding 27 results
0x40201ca4: adc_single_channel_read(unsigned char) at /Users/mumrah/Documents/Arduino/libraries/LibAPRSesp/AFSK.cpp line 160
0x40106b50: timer1_isr_handler at /Users/mumrah/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_timer.c line 33
0x40106b50: timer1_isr_handler at /Users/mumrah/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_timer.c line 33
0x40106b98: timer1_isr_handler at /Users/mumrah/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_timer.c line 40
0x40106b50: timer1_isr_handler at /Users/mumrah/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_timer.c line 33
0x40106b0d: sample_isr() at /Users/mumrah/Documents/Arduino/libraries/LibAPRSesp/AFSK.cpp line 160
0x4010103d: ets_timer_disarm at ?? line ?
0x4021b8f4: pm_get_sleep_type at ?? line ?
0x40101f6e: spi_flash_read at ?? line ?
0x4010709c: pvPortZalloc at /Users/mumrah/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/heap.c line 33
0x4021b3c1: pm_set_sleep_time at ?? line ?
0x4021b85a: pm_get_sleep_type at ?? line ?
0x40102360: igmp_timer at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/timers.c line 217
0x4021b907: pm_get_sleep_type at ?? line ?
0x40203e79: ets_timer_handler_isr at ?? line ?
0x40203ebe: ets_timer_handler_isr at ?? line ?
```

mumrah, 2017-07-20:
For now, I was able to workaround this issue by using `os_timer_t` from the "user_interface.h" lib.

```c
extern "C" {
  #include "user_interface.h"
}

os_timer_t timer;
boolean tickOccured = false;

void timer_callback(void *pArg) {
  tickOccured = true;
}

void setup() {
  Serial.begin(9600);
  os_timer_setfn(&timer, timer_callback, NULL);
  os_timer_arm(&timer, 1000, true);
}

void loop() {
  if(tickOccured) {
    Serial.println("tick");
    tickOccured = false;
  }
  yield();
}

```

devyte, 2017-09-05:
You can't have something like delay(5000) in the loop. The max time it takes for one loop() call to execute is limited, otherwise all kinds of wrong happen. And that doesn't even take into account the wdt.
In addition, the time you spend inside an ISR is even more limited, and there are certain things that you can't do from within it, such as socket comms or mem allocs.
Your "workaround" is actually the correct approach: you set a flag from the ISR, and do heavier lifting inside the loop. There are other ways, such as calculating the amount of ms elapsed using millis(), and taking action once over a threshold.

Having said the above, this looks like a user error due to lack of understanding of the ESP. If you need further discussion, please ask at esp8266.com or stackoverflow, as this is a tracker for issues in the core libs.
Closing.

closed 2017-09-05 by devyte