esp8266/Arduino #3442: Trying to connect to WPA-PEAP without success

rjorod, 2017-07-20:
### Basic Infos

#### Hardware
Hardware:			?ESP-12E
Core Version:      	Using git commit 7f6daa2459ed231e08ce2048b15ce823bffeae1d

### Description
Trying to connect to WPA-PEAP without success

### Settings in IDE

Module:  			NodeMcu 1.0 (ESP-12E)
Flash Size: 		4MB/
CPU Frequency:        80Mhz
Upload Using:		SERIAL

### Sketch

```cpp
extern "C" {
#include "user_interface.h"
#include "wpa2_enterprise.h"
}
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ArduinoJson.h>

// SSID to connect to
static const char* ssid = "SSID";
// Username for authentification
static const char* username = "username";
// Password for authentification
static const char* password = "password";
const int ledPin = 0;
// --------------------------------------------------------------
void setup() {
  Serial.begin(115200);
  pinMode(ledPin, OUTPUT);

  Serial.print("Tryingonnect to ");
  Serial.println(ssid);

  wifi_station_disconnect();
  struct station_config wifi_config;

  memset(&wifi_config, 0, sizeof(wifi_config));
  strcpy((char*)wifi_config.ssid, ssid);
  strcpy((char*)wifi_config.password, password);
  wifi_station_set_config(&wifi_config);

  wifi_station_set_wpa2_enterprise_auth(1);
  wifi_station_set_enterprise_username((uint8*)username, strlen(username));
  wifi_station_set_enterprise_password((uint8*)password, strlen(password));
  wifi_station_connect();

  Serial.print("Status: ");
  Serial.println(wifi_station_get_connect_status());

  // Wait for connection AND IP address from DHCP
  while (WiFi.status() != WL_CONNECTED) {
    delay(2000);
    Serial.println(WiFi.localIP());

  }

} // setup

// ---------------------------
void loop() {
 
}
```

devyte, 2017-07-20:
You are trying to use wpa2 enterprise, which I believe isn't yet fully
operational in the espressif SDK. Until there is a definitive fix from
them, the core won't be able to formally support it.
There have been some success stories with workarounds and hacks, though,
but also some fails with those same tricks.

If you want to try them, just search the issue tracker for wpa2 enterprise.

victorclaessen, 2017-11-07:
There is a bug in the ESP8266 firmware. We are waiting for ExpressIf to fix it.
https://github.com/esp8266/Arduino/issues/1032#issuecomment-341812973

kapyaar, 2018-05-02:
@duncandrennan HAve you tried it with 2.4.1? Playing with eduroam in my university, as well as another wpa2 network with user and password, I have been trying all sorts of things, including what you mentioned, and it never seems to get an IP, keeps on disconnect, scandone, wifi evt:1, etc and loops

I can try going back to 2.4.0 but just wanted to check if it indeed was the version.

duncandrennan, 2018-05-04:
@kapyaar I had this working at a customer's premises with their WPA enterprise setup. I then tried to replicate the setup in our lab and have the ESP8266 crashing for all the different ways I can set up the RADUIS server (I'm testing with TekRADUIS on Windows).

I think (suspect) there is something with the certificates that it doesn't like in my lab setup. It looks like the server is using EAP-MS-CHAP-V2 and it is exchanging some certificates. Possibly specifying the certificates correctly on the ESP8266 would solve the problem (I'm just not sure exactly what needs to be specified).

I'm waiting for feedback on the customers exact network setup, but I think it may be EAP-PAP, which wouldn't send a certificate for authentication.

I did try with 2.4.1 but had the same issue in our lab setup.

kapyaar, 2018-05-04:
@duncandrennan thanks very much for the info. I have ordered a router that has eap support, and I plan to test with ironwifi Radius as a service provider. Atleast this way, I could hopefully see logs on the router where as all my tests so far were blind shots into the dark as I have no access to the router side.
Also, I read on espressiff manual that ESP needs around 26 or 28k ram while on enterprise, so the crashes you see may have something to do with memory? just a thought. I will keep you posted if I get this going.

duncandrennan, 2018-05-04:
@kapyaar I did a check on the heap memory before connecting (for this exact reason), and it was about 32k.

```
 Serial.print("Free heap: ");
 Serial.println(system_get_free_heap_size());
```

One thing I was confused about was the reset cause that it displays after the crash - it shows reset cause 2 which is a pin reset, which is definitely not the case.

kapyaar, 2018-05-04:
@duncandrennan ok, so that's not the issue then. My Debug looked like below. I will update once I get the EAP router .

SDK:2.2.1(cfd48f3)/Core:2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
Connecting to UB_Secure
WPA2 ENTERPRISE VERSION: [v2.0] enable
scandone
Wifi station connect status:1
6
wifi evt: 8
wifi evt: 2
Not connected
6
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (4)
reconnect
wifi evt: 1
STA disconnect: 4
Not connected
6
Not connected
6
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 
Not connected
6
Not connected
6
Not connected
6
Not connected
6
pm open,type:2 0
Not connected
6
Not connected
6
Not connected
6
Not connected
6
Not connected
6
Not connected
6
Not connected
6
Not connected
6
Not connected
6
Not connected
6
state: 5 -> 0 (2)
rm 0
pm close 7
reconnect
wifi evt: 1
STA disconnect: 204
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (4)
reconnect
wifi evt: 1
STA disconnect: 4
scandone
wifi evt: 1
STA disconnect: 201
Not connected
4
reconnect
Not connected
4
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt

duncandrennan, 2018-05-07:
@kapyaar Could you please send me the code which outputs the extra wifi debugging info from the ESP8266? I haven't looked at the events and I would like to see where the ESP8266 gets to before it crashes.

kapyaar, 2018-05-08:
@duncandrennan https://github.com/esp8266/Arduino/issues/4698 I posted the code few days ago

edyradice, 2018-05-28:
hello guys! nothing new on this issue? i have been trying all sort of magic and mojo i found on the web, but the feather HUZZAH ESP8266 seem to not connect to the WiFi network here (EAP-PEAP)... o have this logs on the serial port... 

```
SDK:2.2.1(cfd48f3)/Core:2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
WPA2 ENTERPRISE VERSION: [v2.0] enable
scandone

Waiting for connection and IP Address from DHCP
..scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 110
cnt 
..........pm open,type:2 0
....................state: 5 -> 0 (2)
rm 0
pm close 7
reconnect
.scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 110
cnt 
..........pm open,type:2 0
...........state: 5 -> 2 (7c0)
rm 0
pm close 7
.reconnect
```
Did someone have an advice? i'm stuck on this since two days, trying to figure out what's wrong and reading the non os SDK2 API reference ...

Any help will be reeeally appreciated, thanks!

kapyaar, 2018-05-28:
@edyradice Not much inputs coming from the forum as far as I can tell. Have been beating my head to the wall far too long on this. One thing you can do is to see if you can access the radius server logs and find out what is the reason for the reject

victorclaessen, 2018-05-28:
It's probably never going to work until they fix this: https://bbs.espressif.com/viewtopic.php?f=7&t=8802

edyradice, 2018-05-28:
so frustrating... it's probably some sdk issue, i have been following all their informations and examples but nothing works.... BAH! anyway, thanks for the advices and replies @kapyaar and @victorclaessen !! that was my first post in here!

kapyaar, 2018-05-28:
@edyradice May be those of us interested parties can hold a candle vigil infront of espressif HQ :P

victorclaessen, 2018-05-28:
@edyradice: Every voice we can add to the plea for ExpressIf to finally fix WPA2-Enterprise is very welcome. Thanks for your post.

bospre, 2018-12-17:
Hello together,
good news.
I found a working sketch for wpa2 enterprise with username and password (PAP)
Look here:
https://github.com/jtuttas/ESP8266-WPA2-Enterprise/blob/master/ino/webserver/webserver.ino
I think the difference is resetting (or initialise) the certificate.
2 years of waiting are now ending....

jmccoy555, 2019-01-01:
Works with 2.5.0-beta2 for me, but not 2.4.1, at last I can get my IoT devices onto their own VLAN. 

Thanks everyone!

Simplified sketch;

```
#include <ESP8266WiFi.h>
#include "user_interface.h"
#include "wpa2_enterprise.h"

// SSID to connect to
static const char* ssid = "WiFi";
// Username for authentification
static const char* username = "test";
// Password for authentication
static const char* password = "letmein";

void setup() {

  // put your setup code here, to run once:
  Serial.begin(115200);

  delay(500);

  // WPA2 Connection starts here
  // Setting ESP into STATION mode only (no AP mode or dual mode)
    wifi_set_opmode(STATION_MODE);
    struct station_config wifi_config;
    memset(&wifi_config, 0, sizeof(wifi_config));
    strcpy((char*)wifi_config.ssid, ssid);
    wifi_station_set_config(&wifi_config);
    wifi_station_clear_cert_key();
    wifi_station_clear_enterprise_ca_cert();
    wifi_station_set_wpa2_enterprise_auth(1);
    wifi_station_set_enterprise_identity((uint8*)username, strlen(username));
    wifi_station_set_enterprise_username((uint8*)username, strlen(username));
    wifi_station_set_enterprise_password((uint8*)password, strlen(password));
    wifi_station_connect();
  // WPA2 Connection ends here

  // Normal Connection starts here
  /*
  WiFi.mode(WIFI_STA);
  Serial.write("\r\nConnect to WLAN");
  WiFi.begin(ssid, password);
  // Normal Connection ends here
  */

  // Wait for connection AND IP address from DHCP
  Serial.println();
  Serial.println("Waiting for connection and IP Address from DHCP");
  while (WiFi.status() != WL_CONNECTED) {
    delay(2000);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  IPAddress myAddr = WiFi.localIP();
}

void loop() {

}
```

beta-tank, 2019-01-16:
Create PIO sample project based on https://github.com/jtuttas/ESP8266-WPA2-Enterprise and @jmccoy555 comment.  You can find it here https://github.com/beta-tank/ESP8266-WPA2-Enterprise-PEAP-demo

kapyaar, 2019-01-18:
@beta-tank  @jmccoy555 what kind of wpa2 enterprise network did you test with?  I  am trying same code with a tplink router and ironwifi radius server. but cannot connect.

jmccoy555, 2019-01-19:
@kapyaar  UniFi and freeradius; works with both users entered in the user file and authenticating against Active Directory.

beta-tank, 2019-01-20:
@kapyaar all I know about the tested network, that it is using Active Directory for user authentication.

victorclaessen, 2019-01-21:
Just to be clear; it still doesn't work on eduroam, because we can't control the authentication mechanism and expressif won't fix their bug.

d-a-v, 2019-01-21:
We will have to switch SDK eventually

https://github.com/espressif/ESP8266_RTOS_SDK/issues/442

zhampu, 2019-02-28:
`#include <ESP8266WiFi.h>

extern "C" {
#include "user_interface.h"
#include "wpa2_enterprise.h"
}

// SSID to connect to
static const char* ssid = "wifi";
// Username for authentification
static const char* username = "username";
// Password for authentication
static const char* password = "password";

void setup() {

  // put your setup code here, to run once:
  Serial.begin(115200);

  delay(500);

  // WPA2 Connection starts here
  // Setting ESP into STATION mode only (no AP mode or dual mode)
    wifi_set_opmode(STATION_MODE);
    struct station_config wifi_config;
    memset(&wifi_config, 0, sizeof(wifi_config));
    strcpy((char*)wifi_config.ssid, ssid);
    wifi_station_set_config(&wifi_config);
    wifi_station_clear_cert_key();
    wifi_station_clear_enterprise_ca_cert();
    wifi_station_set_wpa2_enterprise_auth(1);
    wifi_station_set_enterprise_identity((uint8*)username, strlen(username));
    wifi_station_set_enterprise_username((uint8*)username, strlen(username));
    wifi_station_set_enterprise_password((uint8*)password, strlen(password));
    wifi_station_connect();
  // WPA2 Connection ends here

  // Normal Connection starts here
  /*
  WiFi.mode(WIFI_STA);
  Serial.write("\r\nConnect to WLAN");
  WiFi.begin(ssid, password);
  // Normal Connection ends here
  */

  // Wait for connection AND IP address from DHCP
  Serial.println();
  Serial.println("Waiting for connection and IP Address from DHCP");
  while (WiFi.status() != WL_CONNECTED) {
    delay(2000);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  IPAddress myAddr = WiFi.localIP();
}

void loop() {

}`

This works for me.

eveningdemons, 2019-03-16:
> This works for me with the "wpa2_enterprise.h" file in the same forlder of teh sketch
> 
> ```
> #include <ESP8266WiFi.h>
> 
> extern "C" {
> #include "user_interface.h"
> #include "wpa2_enterprise.h"
> }
> 
> 
> // SSID to connect to
> static const char* ssid = "wifi";
> // Username for authentification
> static const char* username = "username";
> // Password for authentication
> static const char* password = "password";
> 
> void setup() {
> 
>   // put your setup code here, to run once:
>   Serial.begin(115200);
> 
>   delay(500);
> 
>   // WPA2 Connection starts here
>   // Setting ESP into STATION mode only (no AP mode or dual mode)
>     wifi_set_opmode(STATION_MODE);
>     struct station_config wifi_config;
>     memset(&wifi_config, 0, sizeof(wifi_config));
>     strcpy((char*)wifi_config.ssid, ssid);
>     wifi_station_set_config(&wifi_config);
>     wifi_station_clear_cert_key();
>     wifi_station_clear_enterprise_ca_cert();
>     wifi_station_set_wpa2_enterprise_auth(1);
>     wifi_station_set_enterprise_identity((uint8*)username, strlen(username));
>     wifi_station_set_enterprise_username((uint8*)username, strlen(username));
>     wifi_station_set_enterprise_password((uint8*)password, strlen(password));
>     wifi_station_connect();
>   // WPA2 Connection ends here
> 
>   // Normal Connection starts here
>   /*
>   WiFi.mode(WIFI_STA);
>   Serial.write("\r\nConnect to WLAN");
>   WiFi.begin(ssid, password);
>   // Normal Connection ends here
>   */
> 
>   // Wait for connection AND IP address from DHCP
>   Serial.println();
>   Serial.println("Waiting for connection and IP Address from DHCP");
>   while (WiFi.status() != WL_CONNECTED) {
>     delay(2000);
>     Serial.print(".");
>   }
>   Serial.println("");
>   Serial.println("WiFi connected");
>   Serial.println("IP address: ");
>   Serial.println(WiFi.localIP());
>   IPAddress myAddr = WiFi.localIP();
> }
> 
> void loop() {
> 
> }
> ```

Hello, everyone! I'm fairly new to the whole microcontroller hobbyist scene. And am trying to use PEAP on a movement senzor connected to a NodeMCU v3 with ESP8266 (ESP-12E) that sends email notifications to IFTTT.

@zhampu, could you please tell me the exact setup that you have?
Are you using a recent firmware for your NodeMcu?
What else is in your folder except the sketch and "wpa2_enterprise.h" and "user_interface.h" files?

I'm using your exact sketch and the two h files in the same folder as the sketch. Plus the other h files that the "user_interface.h" refers to with "include". Yet my device keeps reseting in a loop. You can see what I get in the serial monitor below.

12:47:38.484 -> Waiting for connection and IP Address from DHCP
12:47:40.459 -> .
12:47:49.505 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
12:47:49.505 -> 
12:47:49.505 -> wdt reset
12:47:49.505 -> load 0x4010f000, len 1384, room 16 
12:47:49.505 -> tail 8
12:47:49.505 -> chksum 0x2d
12:47:49.505 -> csum 0x2d
12:47:49.505 -> v951aeffa
12:47:49.505 -> ~ld
12:47:50.140 -> 
12:47:50.140 -> Waiting for connection and IP Address from DHCP
12:47:52.117 -> .................

kapyaar, 2019-03-16:
Just something for you to try. Can you update your code between wpa2 start and end to.

```
// WPA2 Connection starts here
  // Setting ESP into STATION mode only (no AP mode or dual mode)
  // wifi_set_opmode(STATION_MODE);
    WiFi.mode(WIFI_STA);
    struct station_config wifi_config;
    memset(&wifi_config, 0, sizeof(wifi_config));
    strcpy((char*)wifi_config.ssid, ssid);
    wifi_station_set_config(&wifi_config);
    wifi_station_clear_cert_key();
    wifi_station_clear_enterprise_ca_cert();
    wifi_station_set_wpa2_enterprise_auth(1);
    wifi_station_set_enterprise_identity((uint8*)username, strlen(username));
    wifi_station_set_enterprise_username((uint8*)username, strlen(username));
    wifi_station_set_enterprise_password((uint8*)password, strlen(password));
    wifi_station_connect();
  // WPA2 Connection ends here
```

 And see if it makes any difference? This is only aimed at the crash issue. Wpa2 success is an entirely different nightmare :)

eveningdemons, 2019-03-16:
Thanks! Unfortunately it doesn't seem to work. Still not connecting. Keeps
waiting for IP. But no more crashes.

Meanwhile, I've gone back to my own sketch, where I integrated the WPA2
code.

Could it be that I have an older version of the firmware on my NodeMcu?
Should I post my entire sketch here?

kapyaar, 2019-03-16:
Like I said, the above change was only meant to stop the crash. Connection could be a myriad of things, starting with teh fact that wpa2 enterprise library is not an all inclusive one in the sense that it only works what espressif has chosen. Here is my 2 cents. https://github.com/esp8266/Arduino/pull/4853#issuecomment-418454607

You can refer to that full discussion for more info :)

christian-0s, 2019-05-24:
Hi, I also have trouble using PEAP authentication.
The code I'm using:
```
#include <ESP8266HTTPClient.h>
#include <ESP8266WiFi.h>

extern "C" {
#include "user_interface.h"
#include "wpa2_enterprise.h"
}

static const char* ssid = "Flex-Prod";
static const char* username = "-removed-";
static const char* password = "-removed-";

void setup() {
  Serial.begin(115200);
  if (WiFi.status() != WL_CONNECTED)
  {    
    wifi_set_opmode(STATION_MODE);
    struct station_config wifi_config;
    memset(&wifi_config, 0, sizeof(wifi_config));
    strcpy((char*)wifi_config.ssid, ssid);
    wifi_station_set_config(&wifi_config);
    wifi_station_clear_cert_key();
    wifi_station_clear_enterprise_ca_cert();
    wifi_station_set_wpa2_enterprise_auth(1);
    wifi_station_set_enterprise_identity((uint8*)username, strlen(username));
    wifi_station_set_enterprise_username((uint8*)username, strlen(username));
    wifi_station_set_enterprise_password((uint8*)password, strlen(password));
    wifi_station_connect(); 
    
    unsigned long startTime = millis();
    while (WiFi.status() != WL_CONNECTED && millis() - startTime < 10000) {
      ESP.wdtFeed();
      Serial.write('.');
      delay(500);
    }
    if (WiFi.status() != WL_CONNECTED) {
      delay(5000);
      ESP.restart();
    }
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  } 
}
void loop() {
  // put your main code here, to run repeatedly:

}
```

Serial output:

```
SDK:2.2.1(cfd48f3)/Core:2.5.2-3-g6191fbbd=20502003/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
bcn 0
del if1
usl
mode : sta(cc:50:e3:55:03:68)
add if0
WPA2 ENTERPRISE VERSION: [v2.0] enable
.wifi evt: 8
wifi evt: 2
.....scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
............pm open,type:2 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

In attachment some wireshark capture

[wireshark.zip](https://github.com/esp8266/Arduino/files/3218335/wireshark.zip)

nayanashettyft, 2019-05-29:
I am using a ESP8266 NodeMCU ESP -12E Module.
Our enterprise network setting is a PEAP Version1, MSCHAPv2 with no cacerts and AES encryption.
I am on Arduino 2.5.0 version of the SDK. 
My code is the same as @christian-0s mentioned above.
I have tried a few suggestions so far. Updated the libwpa2.a with my username, this dint make any difference. 
Tried replacing wpa2_enterprise.h with the latest version on git@github.com:espressif/ESP8266_NONOS_SDK.git but got the following error:
```
/Users/nayanashetty/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/simple_webserver.ino.cpp.o:(.text.setup+0x28): undefined reference to `wifi_station_clear_enterprise_ca_cert()'
/Users/nayanashetty/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/simple_webserver.ino.cpp.o:(.text.setup+0x2c): undefined reference to `wifi_station_set_wpa2_enterprise_auth(int)'
/Users/nayanashetty/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/simple_webserver.ino.cpp.o:(.text.setup+0x30): undefined reference to `wifi_station_set_enterprise_identity(unsigned char*, int)'
/Users/nayanashetty/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/simple_webserver.ino.cpp.o:(.text.setup+0x34): undefined reference to `wifi_station_set_enterprise_username(unsigned char*, int)'
/Users/nayanashetty/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/simple_webserver.ino.cpp.o:(.text.setup+0x38): undefined reference to `wifi_station_set_enterprise_password(unsigned char*, int)'
/Users/nayanashetty/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/simple_webserver.ino.cpp.o: in function `setup':
/Users/nayanashetty/workspace/simple_webserver/simple_webserver.ino:14: undefined reference to `wifi_station_clear_enterprise_ca_cert()'
/Users/nayanashetty/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/simple_webserver.ino.cpp.o: in function `setup':
/Users/nayanashetty/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h:76: undefined reference to `wifi_station_set_wpa2_enterprise_auth(int)'
/Users/nayanashetty/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/simple_webserver.ino.cpp.o: in function `setup':
/Users/nayanashetty/workspace/simple_webserver/simple_webserver.ino:14: undefined reference to `wifi_station_set_enterprise_identity(unsigned char*, int)'
/Users/nayanashetty/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/simple_webserver.ino.cpp.o: in function `setup':
/Users/nayanashetty/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/HardwareSerial.h:76: undefined reference to `wifi_station_set_enterprise_username(unsigned char*, int)'
/Users/nayanashetty/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/simple_webserver.ino.cpp.o: in function `setup':
/Users/nayanashetty/workspace/simple_webserver/simple_webserver.ino:16: undefined reference to `wifi_station_set_enterprise_password(unsigned char*, int)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```
Any pointers would be helpful.

d-a-v, 2019-06-23:
[https://github.com/espressif/ESP8266_RTOS_SDK/issues/635](http://comment/if/you/can)

nayanashettyft, 2019-06-24:
This link doesnt work for me

christian-0s, 2019-06-24:
Copy and paste visualized text

d-a-v, 2019-06-24:
> This link doesnt work for me

It's on purpose otherwise a link to here would have appeared there. And I don't want to stress them out, given the time and the number of times we've asked that to them. The link in there states that it was planned for release 3.2, which just happened to come out.

If there are some comments or :+1: there to show them I'm not the only one waiting for this - and I know it's true - they may put more efforts on what was initially planned for v3.2. If they do so, it would be a good reason to try to move from nearly EOL nonos-sdk to this supported firmware.

d-a-v, 2019-06-24:
@christian-0s @nayanashettyft 

Can you add in anywhere in `setup()` this call: `disable_extra4k_at_link_time();`

because (and I'm not saying it will work)
- nonos-sdk is supposed to work with MSCHAPv2 (I have never tried myself, my eduroam is PAP)
- we use WPS ram for our stack, maybe it's also WPA2-Enterprise ram: this call leaves this space alone as it is supposed to be, for fw internals.

christian-0s, 2019-07-08:
Hi @d-a-v 
Adding `disable_extra4k_at_link_time();` it does not crash any more, but still not getting a connection:
```
SDK:2.2.2-dev(c0eb301)/Core:2.5.2-3-g6191fbbd=20502003/lwIP:1.4.0rc2/BearSSL:a143020
bcn 0
del if1
usl
mode : sta(cc:50:e3:55:03:90)
add if0
WPA2 ENTERPRISE VERSION: [v2.0] enable

Waiting for connection and IP Address from DHCP
wifi evt: 8
wifi evt: 2
.scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 
.....pm open,type:2 0
..........state: 5 -> 0 (2)
rm 0
pm close 7
reconnect
wifi evt: 1
STA disconnect: 204
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 
.....pm open,type:2 0
..........state: 5 -> 0 (2)
rm 0
pm close 7
reconnect
wifi evt: 1
STA disconnect: 204
.scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 
.....pm open,type:2 0
state: 5 -> 2 (7c0)
rm 0
pm close 7
wifi evt: 1
STA disconnect: 7
```

d-a-v, 2019-07-08:
@christian-0s 
 > Adding `disable_extra4k_at_link_time();` it does not reset any more

Thanks alot for testing this.

How is configured your AP ?
I have had no success with TTLS/PAP but (and I did not test it myself) it is supposed to work with TTLS/MSCHAPv2.

@nayanashettyft Can you give `disable_extra4k_at_link_time();` a try ?

christian-0s, 2019-07-09:
> How is configured your AP ?
> I have had no success with TTLS/PAP but (and I did not test it myself) it is supposed to work with TTLS/MSCHAPv2.
> 

Security Mode: WAP2-Enterprise/PEAP
tunnel authentication type: MS CHAP v2
encryption type: AES

BTW using an ESP32 based board I managed to connect to the same network (code is a bit different):
```
  WiFi.mode(WIFI_STA); //init wifi mode
  esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY)); //provide identity
  esp_wifi_sta_wpa2_ent_set_username((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY)); //provide username --> identity and username are the same
  esp_wifi_sta_wpa2_ent_set_password((uint8_t *)EAP_PASSWORD, strlen(EAP_PASSWORD)); //provide password
  esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT(); //set config settings to default
  esp_wifi_sta_wpa2_ent_enable(&config); //set config settings to enable function
  WiFi.begin(ssid); //connect to wifi
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    counter++;
    if(counter>=60){ //after 30 seconds timeout - reset board
      ESP.restart();
    }
  }
```

ketankr9, 2019-12-06:
> Create PIO sample project based on https://github.com/jtuttas/ESP8266-WPA2-Enterprise and @jmccoy555 comment. You can find it here https://github.com/beta-tank/ESP8266-WPA2-Enterprise-PEAP-demo  

I tried this on my college enterprise-PEAP wifi network and **it is working**.  
https://github.com/beta-tank/ESP8266-WPA2-Enterprise-PEAP-demo