esp8266/Arduino #3443: ESP8266 resetting periodically during Udp.write (seemingly)

Plamena24, 2017-07-20:
### Basic Infos

#### Hardware
Hardware:			ESP-12-E
Core Version:      	not sure how to find out

### Description

Problem description
The board resets every so often, not in a regular pattern while executing the sendButtonState() function. The WatchDogTimer triggers each time. The packets I am writing are 196 bytes which doesn't seem all that big but I don't know enough to determine what is actually going on. [](url)

### Settings in IDE

Module:  			NodeMCU 1.0(ESP 12E module)
Flash Size: 		4M
CPU Frequency:		80MHz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu (but not really sure)

### Sketch

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>

#include <Wire.h>
#include "Adafruit_MCP23017.h"
#include <Adafruit_NeoPixel.h>
#ifdef __AVR__
  #include <avr/power.h>
#endif

#include <OSCBoards.h>
#include <OSCBundle.h>
#include <OSCData.h>
#include <OSCMatch.h>
#include <OSCMessage.h>
#include <OSCTiming.h>

#define PIN 14

char ssid[] = "Balloon Network";          //  your network SSID (name) 
char pass[] = "balloons";   // your network password

//char ssid[] = "botcave";          //  your network SSID (name) 
//char pass[] = "12porcupines"; 

char* hostString = "buttonNode";
const unsigned int localPort = 5005;        // local port to listen for UDP packets (here's where we send the packets)

char messageBuffer[196]= {};  
bool buttonsPressed[98] = {};

unsigned long lightTimers_ms[98] = {};

unsigned long next_tx_ms = 0;
unsigned long next_rx_ms = 0;
int tx_interval_ms = 100;
int rx_interval_ms = 90;
int lightChange_interval_ms = 5000;

Adafruit_MCP23017 MCPArray[7] = {};
Adafruit_NeoPixel strip = Adafruit_NeoPixel(98, PIN, NEO_RGBW + NEO_KHZ800);

typedef struct
{
  char chipId;
  char pinNumber;
[... 240 lines cut ...]
 
  delay(1);
}

### Debug Messages
This is the most common stack:

0x40203980: UdpContext::send(ip_addr*, unsigned short) at /Users/plamena/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/WiFiUdp.cpp line 67
:  (inlined by) WiFiUDP::endPacket() at /Users/plamena/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/WiFiUdp.cpp line 191
0x402055dd: Adafruit_NeoPixel::show() at /Users/plamena/Documents/Arduino/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp line 1971
0x4021d92e: wifi_get_opmode at ?? line ?
0x402027d1: sendButtonState() at /Users/plamena/Documents/Projects/MIRR files/MIRR/buttonController_MIRR/buttonController_MIRR.ino line 162
0x40202849: loop at /Users/plamena/Documents/Projects/MIRR files/MIRR/buttonController_MIRR/buttonController_MIRR.ino line 291
0x40205f28: loop_wrapper at /Users/plamena/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at /Users/plamena/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/cont.S line 109

This one is the other one that shows up:
0x40203980: UdpContext::send(ip_addr*, unsigned short) at /Users/plamena/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/WiFiUdp.cpp line 67
:  (inlined by) WiFiUDP::endPacket() at /Users/plamena/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/WiFiUdp.cpp line 191
0x4021d92e: wifi_get_opmode at ?? line ?
0x402027d1: sendButtonState() at /Users/plamena/Documents/Projects/MIRR files/MIRR/buttonController_MIRR/buttonController_MIRR.ino line 162
0x4020152f: delay at /Users/plamena/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_wiring.c line 50
0x40202849: loop at /Users/plamena/Documents/Projects/MIRR files/MIRR/buttonController_MIRR/buttonController_MIRR.ino line 291
0x40205f28: loop_wrapper at /Users/plamena/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at /Users/plamena/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/cont.S line 109

devyte, 2017-07-20:
How long does it take to execute one loop()?
What does the heap usage look like over time?

Plamena24, 2017-07-20:
Looks like about 50 milliseconds for the loop but that was done crudely by printing millis() at start and end of loop. If there is a better way to figure that out please let me know. As far as heap usage, I actually have no idea what that question means so if you can elaborate I might be able to check this. (As you can see from my code, I am definitely a novice but happy to learn.)

devyte, 2018-12-05:
A lot of things have changed since this was opened, including lwip changes (including udp), mdns rewrite, wiring rewrites. Closing due to age. If the issue is still present, please open a new issue and fill in the requested info.

closed 2018-12-05 by devyte