esp8266/Arduino #3445: WPS Fails for 802.11ac access points

dalbert2, 2017-07-20:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### WPS works for 802.11n access points but not for 802.11ac access points

#### Hardware
Hardware:			ESP-12F
Core Version:      	2.3.0

### Description

The WPS function seems to work reliably with older 802.11n access points (Trendnet TEW-652, Belkin N300), but fails reliably with 802.11ac access points (AT&T/Pace 5268AC and Asus TM-AC1900).  
 I am using Arduino/EPS8266 version 2.3.0 (uses SDK version: 1.5.3(aec24ac9)).

The failure is always the same; it finds the access point, showing it's SSID correctly but fails to successfully process the public key provided by the access point and acquire the PSK (see trace below).  Has anyone been able to connect to an 802.11ac access point using WPS and if so, can you share details (model #, anything special you had to do in the code)?  Thanks!

### Settings in IDE

Module:  			WifInfo
Flash Size: 		4MB
CPU Frequency:	80MHz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp

void ICACHE_FLASH_ATTR startWPSPBC() {

	Serial.printf("WPS start...");

	// must be in station mode for WPS
	WiFi.mode(WIFI_STA);
	delay(1000);

	// Start WPS connect process - takes a long time to timeout
	if (WiFi.beginWPSConfig()) {
		// Well this means not always success :-/ in case of a timeout we have an empty ssid
		String newSSID = WiFi.SSID();
		String newPSK = WiFi.psk();
		if ((newSSID.length() > 0) && (newPSK.length() > 0)) {
			// WPSConfig has already connected in STA mode successfully to the new station. 
			Serial.printf("Connected to '%s'\n", newSSID.c_str());
		}
		else {
			Serial.printf("WPS failed\n");
			wifi_wps_disable();
		}
	}

	// return HotSpot to operation
	Serial.printf("Returning to AP_STATION mode...\n");
	WiFi.mode(WIFI_AP_STA);
}

```
***DEBUG OUTPUT IN FAILURE CASE***
wifi_wps_disable
wifi_wps_enable
wps scan
build public key start
build public key finish
f r0, wps discover [ATT4PcDe8QR]
scandone
WPS: neg start
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt
process pubkey start
wps: ignore overlap identifier
wps: ignore overlap identifier
wps: ignore overlap identifier
wps: ignore overlap identifier
wps: ignore overlap identifier
wps: ignore overlap identifier
state: 5 -> 2 (17c0)
rm

```

devyte, 2017-07-22:
@dalbert2 [This is the link to the espressif forum,](http://bbs.espressif.com/search.php) where ESP discussions are held and issues are tracked. In there, you can find a[ forum specific to the ESP8266 SDK](http://bbs.espressif.com/viewforum.php?f=7), and there you can search for WPS.
The Arduino core SDK developed in this repository is just wrappers around the espressif SDK. In an ideal world, when faced with an issue, users would investigate the core libs and discriminate, so that bugs specific to these wrappers are reported here, and issues that are lower level within the espressif SDK are reported there. Of course, this doesn't happen often, but that's understandable.
[Here is a link](http://espressif.com/en/support/download/documents) to the ESP8266 docs, which explains the usage and expected behavior of their lower level API, on which the Arduino libs are based. If you test with their lower level api and it works, then test the Arduino libs, and it fails, then it's very likely the issue is in the Arduino libs, of course. That gives a starting point to look at the Arduino lib code.
Here is a [link to how to use the git](http://arduino-esp8266.readthedocs.io/en/latest/installing.html#), and here is a [link to the readthedocs](https://arduino-esp8266.readthedocs.io/en/latest/index.html) latest toplelvel.
Finally, if you want to look at the Arduino code, it's generally divided into three parts: 

- the core itself: core code like uart, FS, and so on. Generally something/cores/esp8266
- the core libs: libs that are included in the base install, like ESP8266WiFi. Generally something/libraries
- the libs you install, like NTPClient. Generally under .arduino/libraries

The above dirs are for Linux, they're different for Windows. I just realized I have a bit of a mess and should do a clean reinstall, so my directory pointers above may be inaccurate, but in any case, you can find the answers online, there's no mystery.

Hope this helps, good luck!

dalbert2, 2017-09-23:
The problem turns out to be free RAM.  The 802.11ac WPS process requires considerably more free RAM than for 802.11a/b/g/n.  My application had some large statically allocated tables.  When I changed these to be dynamically allocated after WPS completed, it worked properly.

closed 2017-09-23 by dalbert2