esp8266/Arduino #3447: WiFiClientSecure can connect to SSL MQTT broker without verfication

mnsk, 2017-07-21:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.3.0

### Description

I was trying to connect my esp with username and password authenticated SSL MQTT broker using root CA certificate as verification and was having issue with setCACert() and WiFiManger. So I decided to test with fingerprint verification and it worked. But when I removed all kind of verification I was still able to connect with my broker on port 8883. I tried with different clients like android, linux etc. and none was able to connect without root CA certificate. So how come esp could connect without verification?

### Settings in IDE

Module:  			NodeMcu v3
Flash Size: 		4MB
CPU Frequency:		1600Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		nodemcu

copercini, 2017-07-26:
It's your option verify if you are talking with the real server or a fake one.
I highly recommend verify it with CA certificate instead of bypass.

devyte, 2018-07-13:
Closing due to merged bearssl (axtls will be deprecated).

closed 2018-07-13 by devyte