esp8266/Arduino #3450: Exception (3)

minhan74, 2017-07-23:
### Description

I am running into an issue, the details are below. The issue happen when i tried to add a new variable to the function SendCommand(), more specific is the variable "room". In the previous version, I use the directly function without new variable, so the function is ROOM1.print(); , everything works fine, but then because I want to control more room, so I need to create a variable and point it to the room that i want, so it became room.print(); and i will point "room" to ROOM1 ROOM2 ROOM3, but when i applied it, the error happened. I have checked everything but still cannot find the bug, the error only appear after the second time the board get message from the server, the first time it get the mesage smoothly. Sorry for my bad english. What did i do wrong? How can I fix it? Thanks for reading.

### Settings in IDE

Module:  			Node MCU 1.0 (ESP 12E module)
Flash Size: 		4MB
CPU Frequency:       80Mhz
Reset Method:		nodemcu

### Sketch

```
#include <ESP8266WiFi.h>
#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"
#include <SoftwareSerial.h>
#include <ArduinoJson.h>

//include thu vien de kiem tra free RAM tren esp8266
extern "C" {
  #include "user_interface.h"
}

// the on off button feed turns this LED on/off
#define RX1 D1  
#define TX1 D2
#define RX2 D3
#define TX2 D5
#define RX3 D6
#define TX3 D7
SoftwareSerial ROOM1 = SoftwareSerial(RX1,TX1,false,256);
SoftwareSerial ROOM2 = SoftwareSerial(RX2,TX2,false,256);
SoftwareSerial ROOM3 = SoftwareSerial(RX3,TX3,false,256);
/************************* WiFi Access Point *********************************/

#define WLAN_SSID       "xxx"
#define WLAN_PASS       "xxx"

/************************* Adafruit.io Setup *********************************/

#define AIO_SERVER      "xxx"
#define AIO_SERVERPORT  1883                   // use 8883 for SSL
#define AIO_USERNAME    "xxx"
#define AIO_KEY         "xxx"

/************ Global State (you don't need to change this!) ******************/

// Create an ESP8266 WiFiClient class to connect to the MQTT server.
WiFiClient client;
// or... use WiFiFlientSecure for SSL
//WiFiClientSecure client;

// Setup the MQTT client class by passing in the WiFi client and MQTT server and login details.
Adafruit_MQTT_Client mqtt(&client, AIO_SERVER, AIO_SERVERPORT, AIO_USERNAME, AIO_USERNAME, AIO_KEY);

/****************************** Feeds ***************************************/

// Notice MQTT paths for AIO follow the form: <username>/feeds/<feedname>

Adafruit_MQTT_Subscribe r1message = Adafruit_MQTT_Subscribe(&mqtt,AIO_USERNAME "/feeds/r1-message",MQTT_QOS_0);
Adafruit_MQTT_Subscribe r2message = Adafruit_MQTT_Subscribe(&mqtt,AIO_USERNAME "/feeds/r2-message",MQTT_QOS_0);
Adafruit_MQTT_Subscribe r3message = Adafruit_MQTT_Subscribe(&mqtt,AIO_USERNAME "/feeds/r3-message",MQTT_QOS_0);

/*************************** Sketch Code ************************************/

// Bug workaround for Arduino 1.6.6, it seems to need a function declaration
// for some reason (only affects ESP8266, likely an arduino-builder bug).
void MQTT_connect();

void setup() {
  Serial.begin(57600);
  ROOM1.begin(57600); //open software serial to communicate arduino
  ROOM2.begin(57600); //open software serial to communicate arduino
  ROOM3.begin(57600); //open software serial to communicate arduino

  WiFi.mode(WIFI_STA);
  
  delay(10);

  Serial.println(); Serial.println();

  // Connect to WiFi access point.
  Serial.println();
  Serial.print(F("Connecting to "));
  Serial.println(WLAN_SSID);

  WiFi.begin(WLAN_SSID, WLAN_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();

  Serial.println("WiFi connected");
  Serial.println("IP address: "); Serial.println(WiFi.localIP());

  // Setup MQTT subscription for onoff & start time end time.
  mqtt.subscribe(&r1message);
  mqtt.subscribe(&r2message);
  mqtt.subscribe(&r3message);
 
}

void loop() {
  // Ensure the connection to the MQTT server is alive (this will make the first
  // connection and automatically reconnect when disconnected).  See the MQTT_connect
  // function definition further below.
  MQTT_connect();

  // this is our 'wait for incoming subscription packets' busy subloop

  Adafruit_MQTT_Subscribe *subscription;
  while ((subscription = mqtt.readSubscription(5000))) {

       //ROOM1
       if (subscription == &r1message) {
          SendCommand(r1message,ROOM1,"ROOM1");
          
        }                    
       //ROOM2
       if (subscription == &r2message) {
          SendCommand(r2message,ROOM2,"ROOM2");
        }             
       //ROOM3
       if (subscription == &r3message) {
          SendCommand(r3message,ROOM3,"ROOM3");
        }             
       
}

uint32_t free = system_get_free_heap_size(); //KIEM TRA XEM CON BAO NHIEU RAM DE DEBUG
        Serial.println();
        Serial.print(F("luong RAM con lai:    "));
        Serial.println(free);

// ping the server to keep the mqtt connection alive
  if(! mqtt.ping()) {
    mqtt.disconnect();
    }      

  
  }

// Function to connect and reconnect as necessary to the MQTT server.
// Should be called in the loop function and it will take care if connecting.
void MQTT_connect() {
  int8_t ret;

  // Stop if already connected.
  if (mqtt.connected()) {
    return;
  }

  Serial.print("Connecting to MQTT... ");

  uint8_t retries = 3;
  while ((ret = mqtt.connect()) != 0) { // connect will return 0 for connected
       Serial.println(mqtt.connectErrorString(ret));
       Serial.println("Retrying MQTT connection in 5 seconds...");
       mqtt.disconnect();
       delay(5000);  // wait 5 seconds
       retries--;
       if (retries == 0) {
         // basically die and wait for WDT to reset me
         while (1);
       }
  }
  Serial.println("MQTT Connected!");
}

String parsestring(String json,String names)
{
   StaticJsonBuffer<200> jsonBuffer;  
   JsonObject& root = jsonBuffer.parseObject(json); 
   if (!root.success()) {
    Serial.println("parseObject() failed");
    exit;
  }
  return root[names];
}

void SendCommand(Adafruit_MQTT_Subscribe message,SoftwareSerial room, const char *r) {
          char *json=(char *)message.lastread;
          String action = parsestring(json,"action");  // get action
       
          if (action == "ON") {
            Serial.println(r);
            room.print("ON");
            room.print('\r');
            room.print(json);
            room.print('\r');
            Serial.println(json);
          }
          if (action == "OFF") {
            Serial.println(r);
            room.print("OFF");
            room.print('\r');
            room.print("");
            room.print('\r');
            Serial.println(json);
          }
}

```

### Debug Messages

```
ROOM1
{"action":"ON","id":"84760CC5"}
ROOM1
{"action":"OFF","id":""}

Exception (3):
epc1=0x4010011d epc2=0x00000000 epc3=0x00000000 excvaddr=0x40030be8 depc=0x00000000

ctx: cont 
sp: 3ffef9b0 end: 3ffefc90 offset: 01a0

>>>stack>>>
3ffefb50:  00000000 00000000 00000000 00000000  
3ffefb60:  3ffeeb88 000000c6 000000c6 4010020c  
3ffefb70:  3ffee8c8 3ffee840 3ffefc30 4010068c  
3ffefb80:  00000003 00000015 4023127e 40203e4c  
3ffefb90:  3ffee8c8 3ffee840 3ffefc30 40202970  
3ffefba0:  3ffe8560 61227b00 6f697463 223a226e  
3ffefbb0:  2246464f 6469222c 22223a22 0000007d  
3ffefbc0:  00000000 00000000 00000000 00000000  
3ffefbd0:  00000000 00000000 00000000 00000000  
3ffefbe0:  00000000 00000000 00000000 00000000  
3ffefbf0:  00000000 00000000 00000000 00000000  
3ffefc00:  00000000 00000000 00180000 00000000  
3ffefc10:  00000000 00000000 00000000 00000000  
3ffefc20:  00000000 3ffee8c8 3ffefc30 3ffe8514 <
3ffefc30:  3ffe86e0 00000000 000003e8 00000000  
3ffefc40:  00000005 00000004 00000000 00000101  
3ffefc50:  0000056c 00000000 00000000 00000100  
3ffefc60:  3fff0ae4 00000000 00000001 4020465d  
3ffefc70:  3fffdad0 00000000 3ffeec60 40204688  
3ffefc80:  feefeffe feefeffe 3ffeec70 40100718  
<<<stack<<<

```

devyte, 2017-07-23:
I don't know what your json looks like, but your staticjsonbuffer looks too
small at 200. Try with dynamicjsonbuffer, this could be stack corruption.

minhan74, 2017-07-24:
my jsons are: {"action":"ON","id":"84760CC5"} and {"action":"OFF","id":""}

I have tried DynamicJsonBuffer, but the same error still occured...

devyte, 2017-07-24:
Isn't pin D3 used for the flash on Nodemcu module? Double check the pins
you are using for your softwareserial objects.

minhan74, 2017-07-24:
I think D0 and D4 are the pins used for internal flashs. Anyway I change it to D8, and still, first message being delivered from the server sucessfully recieved, regardless what is that message, but the second time i try to send a new message from the server to the esp, the error occured

devyte, 2017-07-24:
Don't confuse GPIO0 with D0. I just looked at a pinout, and on the nodemcu,
GPIO0 is the same as D3, which is used for flash.

D8 is the same as GPIO15, which says TXD2/HSPICS.

minhan74, 2017-07-25:
This time I tried to run the old version of the file, and it run smoothly, This is the old sketch:
`
#include <ESP8266WiFi.h>
#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"
#include <SoftwareSerial.h>
#include <ArduinoJson.h>

//include thu vien de kiem tra free RAM tren esp8266
extern "C" {
  #include "user_interface.h"
}

// the on off button feed turns this LED on/off
#define RX D1  
#define TX D2
SoftwareSerial ROOM1 = SoftwareSerial(RX,TX,false,256);
/************************* WiFi Access Point *********************************/

#define WLAN_SSID       "Home"
#define WLAN_PASS       "minhan74"

/************************* Adafruit.io Setup *********************************/

#define AIO_SERVER      "io.adafruit.com"
#define AIO_SERVERPORT  1883                   // use 8883 for SSL
#define AIO_USERNAME    "minhan74"
#define AIO_KEY         "da2d77feb2d34079826149d563623b4d"

/************ Global State (you don't need to change this!) ******************/

// Create an ESP8266 WiFiClient class to connect to the MQTT server.
WiFiClient client;
// or... use WiFiFlientSecure for SSL
//WiFiClientSecure client;

// Setup the MQTT client class by passing in the WiFi client and MQTT server and login details.
Adafruit_MQTT_Client mqtt(&client, AIO_SERVER, AIO_SERVERPORT, AIO_USERNAME, AIO_USERNAME, AIO_KEY);

/****************************** Feeds ***************************************/

// Notice MQTT paths for AIO follow the form: <username>/feeds/<feedname>

Adafruit_MQTT_Subscribe r1message = Adafruit_MQTT_Subscribe(&mqtt,AIO_USERNAME "/feeds/r1-message",MQTT_QOS_0);
/*************************** Sketch Code ************************************/

// Bug workaround for Arduino 1.6.6, it seems to need a function declaration
// for some reason (only affects ESP8266, likely an arduino-builder bug).
void MQTT_connect();

void setup() {
  Serial.begin(57600);
  ROOM1.begin(57600); 

  WiFi.mode(WIFI_STA);
  
  delay(10);

  Serial.println(); Serial.println();
  Serial.println(F("TAO NHAT DINH SE LAM DC!!!!!!"));

  // Connect to WiFi access point.
  Serial.println();
  Serial.print(F("Connecting to "));
  Serial.println(WLAN_SSID);

  WiFi.begin(WLAN_SSID, WLAN_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
[... 107 lines cut ...]
          SendCommand(r1message,"ROOM1",ROOM1);
          
        }                    `

`void SendCommand(Adafruit_MQTT_Subscribe message,const char* room,SoftwareSerial ROOM) {
          char *json=(char *)message.lastread;
          String action = parsestring(json,"action");  // get action
       
          if (action == "ON") {
            Serial.println(room);
            ROOM.print("ON");
            ROOM.print('\r');
            ROOM.print(json);
            ROOM.print('\r');
            Serial.println(json);
          }
          if (action == "OFF") {
            Serial.println(room);
            ROOM.print("OFF");
            ROOM.print('\r');
            ROOM.print("");
            ROOM.print('\r');
            Serial.println(json);
          }
}`

minhan74, 2017-07-28:
This time I tried to run the old version of the file, and it run smoothly, This is the old sketch:
```
#include <ESP8266WiFi.h>
#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"
#include <SoftwareSerial.h>
#include <ArduinoJson.h>

//include thu vien de kiem tra free RAM tren esp8266
extern "C" {
  #include "user_interface.h"
}

// the on off button feed turns this LED on/off
#define RX D1  
#define TX D2
SoftwareSerial ROOM1 = SoftwareSerial(RX,TX,false,256);
/************************* WiFi Access Point *********************************/

#define WLAN_SSID       "xxx"
#define WLAN_PASS       "xxx"

/************************* Adafruit.io Setup *********************************/

#define AIO_SERVER      "io.adafruit.com"
#define AIO_SERVERPORT  1883                   // use 8883 for SSL
#define AIO_USERNAME    "xxx"
#define AIO_KEY         "xxx"

/************ Global State (you don't need to change this!) ******************/

// Create an ESP8266 WiFiClient class to connect to the MQTT server.
WiFiClient client;
// or... use WiFiFlientSecure for SSL
//WiFiClientSecure client;

// Setup the MQTT client class by passing in the WiFi client and MQTT server and login details.
Adafruit_MQTT_Client mqtt(&client, AIO_SERVER, AIO_SERVERPORT, AIO_USERNAME, AIO_USERNAME, AIO_KEY);

/****************************** Feeds ***************************************/

// Notice MQTT paths for AIO follow the form: <username>/feeds/<feedname>

Adafruit_MQTT_Subscribe r1message = Adafruit_MQTT_Subscribe(&mqtt,AIO_USERNAME "/feeds/r1-message",MQTT_QOS_0);
/*************************** Sketch Code ************************************/

// Bug workaround for Arduino 1.6.6, it seems to need a function declaration
// for some reason (only affects ESP8266, likely an arduino-builder bug).
void MQTT_connect();

void setup() {
  Serial.begin(57600);
  ROOM1.begin(57600); 

  WiFi.mode(WIFI_STA);
  
  delay(10);

  Serial.println(); Serial.println();
  Serial.println(F("TAO NHAT DINH SE LAM DC!!!!!!"));

  // Connect to WiFi access point.
  Serial.println();
  Serial.print(F("Connecting to "));
  Serial.println(WLAN_SSID);

  WiFi.begin(WLAN_SSID, WLAN_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
[... 110 lines cut ...]

```
void SendCommand(Adafruit_MQTT_Subscribe message,const char* room,SoftwareSerial ROOM) {
          char *json=(char *)message.lastread;
          String action = parsestring(json,"action");  // get action
       
          if (action == "ON") {
            Serial.println(room);
            ROOM.print("ON");
            ROOM.print('\r');
            ROOM.print(json);
            ROOM.print('\r');
            Serial.println(json);
          }
          if (action == "OFF") {
            Serial.println(room);
            ROOM.print("OFF");
            ROOM.print('\r');
            ROOM.print("");
            ROOM.print('\r');
            Serial.println(json);
          }
}
```
Nothing else was change, but it won't work like it should be  T.T

devyte, 2017-09-05:
Assuming that you are in fact using the correct pins for softwareserial (which I'm not certain that you are), I suspect you have timing issues. You have things like delay(5000) inside the loop, which is far too much.
Also, I see way too many Send this and connect that, with no yields in between.
One pass through the loop can't take very long, or the wifi stack will blow up. 

Having said that, this is not the right place to discuss what is wrong with your particular project. This is an issue tracker, meant for tracking issues in the core. Please ask for assistance/discussion at esp8266.com or stackoverflow. If you can reduce the crash to something that shows an issue in the core, then please open a new issue here, and reference this one.
Closing for now.

closed 2017-09-05 by devyte